t6x/reaver-wps-fork-t6x #88: memory corruption resulting in segfaults

rofl0r, 2016-11-27:
there's some kind of malloc-related memory corruption bug in reaver.
something like use-after-free, double-free or the like.
the bug hits me all the time because i use a linux distribution (for example alpine linux) with musl libc, which has a built-in memory corruption detection in its malloc implementation. the bug is of course also present in glibc-based distributions, just that it doesnt get noticed and silently continues corrupting memory.

it would be nice if someone could run reaver under valgrind and see where the bug comes from.

note: this bug is also present in the original reaver 1.4 by craig heffner.
i tried libpcap version 1.6.1, 1.8.0, and 1.8.1, so that is probably not the culprit.

here are some backtraces i got
```
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x02), re-trying last pin

Program received signal SIGSEGV, Segmentation fault.
free (p=0x676760) at src/malloc/malloc.c:464
464             if (next->psize != self->csize) a_crash();
(gdb) bt
#0  free (p=0x676760) at src/malloc/malloc.c:464
#1  0x00007ffff7afa4ba in ?? () from /lib/libpcap.so.1
#2  0x00007ffff7ffc5a0 in buf () from /lib/ld-musl-x86_64.so.1
#3  0x000000000069da20 in ?? ()
#4  0x00007fffffffea70 in ?? ()
#5  0x00007ffff7b011bb in pcap_close () from /lib/libpcap.so.1
#6  0x0000000000000000 in ?? ()

[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x02), re-trying last pin

Program received signal SIGSEGV, Segmentation fault.
free (p=0x6f4bc0) at src/malloc/malloc.c:464
464             if (next->psize != self->csize) a_crash();
(gdb) bt
#0  free (p=0x6f4bc0) at src/malloc/malloc.c:464
#1  0x00007ffff7ae5708 in pcap_cleanup_linux_mmap (handle=0x6f4440)
    at ./pcap-linux.c:4449
#2  0x00007ffff7aee1e1 in pcap_close (p=0x6f4440) at ./pcap.c:2104
#3  0x0000000000409390 in pcap_sleep (seconds=1) at misc.c:130
#4  0x0000000000406613 in crack () at cracker.c:131
#5  0x000000000040303e in main (argc=11, argv=0x7fffffffeb88) at wpscrack.c:115
(gdb) 

[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7d405e2 in unbin (i=3, c=0x6f4850) at src/malloc/malloc.c:218
218             c->prev->next = c->next;
(gdb) bt
#0  0x00007ffff7d405e2 in unbin (i=3, c=0x6f4850) at src/malloc/malloc.c:218
#1  malloc (n=<optimized out>) at src/malloc/malloc.c:346
#2  0x00000000004051a1 in build_radio_tap_header (len=0x7fffffffe960)
    at builder.c:41
#3  0x00000000004056b5 in build_snap_packet (len=0x7fffffffe9b8)
    at builder.c:199
#4  0x00000000004059f1 in build_eapol_start_packet (len=0x7fffffffe9f8)
    at builder.c:296
#5  0x00000000004097f7 in send_eapol_start () at send.c:43
#6  0x000000000040743b in do_wps_exchange () at exchange.c:208
#7  0x000000000040679c in crack () at cracker.c:195
#8  0x000000000040303e in main (argc=11, argv=0x7fffffffeb88) at wpscrack.c:115
(gdb) 

[+] Sending WSC NACK
[+] p1_index set to 4
[+] Pin count advanced: 4. Max pin attempts: 11000

Program received signal SIGSEGV, Segmentation fault.
free (p=0x6f4440) at src/malloc/malloc.c:464
464             if (next->psize != self->csize) a_crash();
(gdb) bt
#0  free (p=0x6f4440) at src/malloc/malloc.c:464
#1  0x00007ffff7ae5708 in pcap_cleanup_linux_mmap (handle=0x69da00)
    at ./pcap-linux.c:4449
#2  0x00007ffff7aee1e1 in pcap_close (p=0x69da00) at ./pcap.c:2104
#3  0x0000000000409390 in pcap_sleep (seconds=1) at misc.c:130
#4  0x0000000000406613 in crack () at cracker.c:131
#5  0x000000000040303e in main (argc=11, argv=0x7fffffffeb88) at wpscrack.c:115
(gdb) 

[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7d405e2 in unbin (i=2, c=0x71abf0) at src/malloc/malloc.c:218
218             c->prev->next = c->next;
(gdb) bt
#0  0x00007ffff7d405e2 in unbin (i=2, c=0x71abf0) at src/malloc/malloc.c:218
#1  malloc (n=<optimized out>) at src/malloc/malloc.c:346
#2  0x0000000000405721 in build_snap_packet (len=0x7fffffffe9b8)
    at builder.c:206
#3  0x00000000004059f1 in build_eapol_start_packet (len=0x7fffffffe9f8)
    at builder.c:296
#4  0x00000000004097f7 in send_eapol_start () at send.c:43
#5  0x000000000040743b in do_wps_exchange () at exchange.c:208
#6  0x000000000040679c in crack () at cracker.c:195
#7  0x000000000040303e in main (argc=11, argv=0x7fffffffeb88) at wpscrack.c:115
(gdb) 

[+] Received M1 message

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7d405ea in unbin (i=4, c=0x6f4870) at src/malloc/malloc.c:219
219             c->next->prev = c->prev;
(gdb) bt
#0  0x00007ffff7d405ea in unbin (i=4, c=0x6f4870) at src/malloc/malloc.c:219
#1  malloc (n=<optimized out>, n@entry=24) at src/malloc/malloc.c:346
#2  0x00007ffff7d40a49 in __malloc0 (n=n@entry=24) at src/malloc/malloc.c:361
#3  0x00007ffff7d3f70f in calloc (m=m@entry=1, n=n@entry=24)
    at src/malloc/calloc.c:12
#4  0x0000000000421cdd in os_zalloc (size=size@entry=24) at os_unix.c:318
#5  0x000000000041f5db in bignum_init () at bignum.c:38
#6  0x00000000004258f0 in crypto_mod_exp (
    base=0x43cfa0 <dh_group5_generator> "\002", base_len=<optimized out>, 
    power=0x74c4b8 "\001", power_len=1, 
    modulus=0x43cfc0 <dh_group5_prime> "\377\377\377\377\377\377\377\377\311\017\332\242!h\302\064\304\306b\213\200\334\034\321)\002N\b\212g\314t\002\v\276\246;\023\233\"QJ\by\216\064\004\335\357\225\031\263\315:C\033\060+\nm\362_\024\067O\341\065mmQ\302E\344\205\265vb^~\306\364LB\351\246\067\355k\v\377\\\266\364\006\267\355\356\070k\373Z\211\237\245\256\237$\021|K\037\346I(fQ\354\344[=\302", 
    modulus_len=192, result=0x74cc58 "", 
    result_len=result_len@entry=0x7fffffffe8a8) at crypto_internal-modexp.c:31
#7  0x000000000042399e in dh_init (dh=0x671840 <dh_groups>, 
    priv=priv@entry=0x74c2d0) at dh_groups.c:597
#8  0x00000000004237e9 in dh5_init (priv=0x74c2d0, publ=0x7fffffffe928)
    at dh_group5.c:24
#9  0x000000000040cfa8 in wps_build_public_key ()
#10 0x000000000041805e in wps_build_m2 ()
---Type <return> to continue, or q <return> to quit---
#11 0x0000000000418a96 in wps_registrar_get_msg ()
#12 0x000000000040994a in send_msg (type=14) at send.c:104
#13 0x00000000004073d2 in do_wps_exchange () at exchange.c:181
#14 0x000000000040679c in crack () at cracker.c:195
#15 0x000000000040303e in main (argc=11, argv=0x7fffffffeb88) at wpscrack.c:115
(gdb) 

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7d405e2 in unbin (i=39, c=0x6f4610) at src/malloc/malloc.c:218
218             c->prev->next = c->next;
(gdb) bt
#0  0x00007ffff7d405e2 in unbin (i=39, c=0x6f4610) at src/malloc/malloc.c:218
#1  malloc (n=<optimized out>, n@entry=413) at src/malloc/malloc.c:346
#2  0x00007ffff7d40a49 in __malloc0 (n=n@entry=413) at src/malloc/malloc.c:361
#3  0x00007ffff7d3f70f in calloc (m=m@entry=1, n=n@entry=413)
    at src/malloc/calloc.c:12
#4  0x0000000000421cdd in os_zalloc (size=size@entry=413) at os_unix.c:318
#5  0x000000000042128d in wpabuf_alloc (len=389) at wpabuf.c:129
#6  0x00000000004214b4 in wpabuf_dup (src=0x69dd80) at wpabuf.c:174
#7  0x000000000041bbc7 in wps_process_wsc_msg ()
#8  0x000000000041c2e4 in wps_registrar_process_msg ()
#9  0x00000000004078ae in process_wps_message (data=0x6f4084, data_size=389)
    at exchange.c:445
#10 0x000000000040781b in process_packet (packet=0x6f4040 "", 
    header=0x7fffffffea20) at exchange.c:418
#11 0x0000000000407174 in do_wps_exchange () at exchange.c:80
#12 0x000000000040679c in crack () at cracker.c:195
#13 0x000000000040303e in main (argc=11, argv=0x7fffffffeb88) at wpscrack.c:115
(gdb) 

```

kcdtv, 2016-11-27:
Thanks for reporting the issue

> note: this bug is also present in the original reaver 1.4 by craig heffner.

This is very important to know. Thanks for testing that.

rofl0r, 2016-11-28:
since i found a number of bugs on x86_64 just by looking at (after enabling) compiler warnings ( #89 ), such as pointer truncation due to implicit function declarations, it may even be that the code is just almost untested on x86_64. i remember having run reaver 1.4 in the past on a 32bit ARM system and never ran into issues.

rofl0r, 2016-11-29:
probably related: https://github.com/openwrt/packages/issues/3333
since @kcdtv keeps recommending libpcap version 0.8 here, maybe there's a known incompatibility with newer pcap versions? at least it seems that reaver is doing rather unconventional things with libpcap such as the `pcap_sleep` function that closes the pcap handle, sleeps, and then creates a new pcap context just for the sake of not having to process the packets that arrived during the sleeptime. each of those events causes pcap to free and reallocate memory.

xhebox, 2016-12-03:
glib-2.24 with archlinux, i'm going to take a try with -g2.
```
==490== Invalid read of size 8
==490==    at 0x408782: ??? (in /usr/bin/reaver)
==490==    by 0x406090: ??? (in /usr/bin/reaver)
==490==    by 0x402B6F: ??? (in /usr/bin/reaver)
==490==    by 0x58B5290: (below main) (in /usr/lib/libc-2.24.so)
==490==  Address 0x7472fd0 is 4 bytes after a block of size 12 alloc'd
==490==    at 0x4C2AB8D: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==490==    by 0x5915189: strdup (in /usr/lib/libc-2.24.so)
==490==    by 0x406B4C: ??? (in /usr/bin/reaver)
==490==    by 0x40444B: ??? (in /usr/bin/reaver)
==490==    by 0x402AA5: ??? (in /usr/bin/reaver)
==490==    by 0x58B5290: (below main) (in /usr/lib/libc-2.24.so)
==490== 
```

xhebox, 2016-12-03:
@kcdtv @rofl0r I've made a patch for this. It has worked for 10min, no segfaults when starting, running, exiting(with musl). I'm going to leave it running for one night to see if this is the key. roflor, you could try it with sabotage directly :D, since it won't exit with glibc and you hardly get valgrind working on musl. Need to try it with glibc, either.

The main part is the third patch(I checked the test programs under libpcap's source tree, found `BUFSIZ` suspicious.),
 the second one should solve the error i posted on previous comment, the first one is for musl. To be honest, reaver's code is a bit messy, maybe we should consider cleaning it.
```
--- .src/reaver-wps-fork-t6x-1.5.2/src/libwps/libwps.h	2015-06-07 03:29:28.000000000 +0000
+++ reaver-wps-fork-t6x-1.5.2/src/libwps/libwps.h	2015-06-07 03:29:28.000000000 +0000
@@ -16,6 +16,7 @@
 #include <stdint.h>
 #include <stdio.h>
 #include <string.h>
+#include <sys/types.h>
 #include <arpa/inet.h>
 
 #define LIBWPS_MAX_STR_LEN 256
--- src/session.c	2015-06-07 03:29:28.000000000 +0000
+++ b/src/session.c	2015-06-07 03:29:28.000000000 +0000
@@ -192,7 +191,7 @@
          */
         if(get_session())
         {
-            memcpy((char *) &file_name, get_session(), FILENAME_MAX-1);
+            memcpy(file_name, get_session(), strlen(get_session())+1);
         }
         else
         {	
--- .src/reaver-wps-fork-t6x-1.5.2/src/init.c	2015-06-07 03:29:28.000000000 +0000
+++ reaver-wps-fork-t6x-1.5.2/src/init.c	2015-06-07 03:29:28.000000000 +0000
@@ -121,7 +121,7 @@
     pcap_t *handle = NULL;
     char errbuf[PCAP_ERRBUF_SIZE] = { 0 };
 
-    handle = pcap_open_live(capture_source, BUFSIZ, 1, 0, errbuf);
+    handle = pcap_open_live(capture_source,  65535, 1, 0, errbuf);
     if(!handle)
     {
         handle = pcap_open_offline(capture_source, errbuf);
```

rofl0r, 2016-12-03:
@xhebox that's a most welcome addition! i'm curious to hear the results of your stresstest!

xhebox, 2016-12-04:
@rofl0r About 9 hours and 30 mins, no segfaults, surfing internet without any problems after closing reaver. 95.5% completed, however i'm in blacklist now XD.  But where is the defination of `BUFSIZ`? It should be deleted.

YI1992, 2016-12-04:
Thank you xhebox.I use your patch, t6x-reaver on the router has been running 30＋ minutes,LEDE r2400,brcm47xx,MIPS 74Kc,USB wireless network card ar9271,libpcap1.7.4.

xhebox, 2016-12-04:
@YI1992 Happy to see that :D! 
@rofl0r BUFSIZ is defined in stdio.h, 1024. It seems that it's too small for reaver. According to the man page of pcap, i replaced it with 65535.
```
A snapshot length of 65535 should be sufficient, on most if not all networks, to capture all the data available from the packet.
The snapshot length is set with pcap_set_snaplen().
```
I'm not faimilar with reaver, i guess reaver does not need the whole packets, so it choose 1024. Though 65535 wastes more space, but we get more compatibility. It's suitable in my opinion.

Here the final patch is(the first part could be improved, we don't need to copy as much as the size of buffer):
```
--- src/session.c	2015-06-07 03:29:28.000000000 +0000
+++ b/src/session.c	2015-06-07 03:29:28.000000000 +0000
@@ -192,7 +191,7 @@
          */
         if(get_session())
         {
-            memcpy((char *) &file_name, get_session(), FILENAME_MAX-1);
+            memcpy(file_name, get_session(), strlen(get_session())+1);
         }
         else
         {	
--- .src/reaver-wps-fork-t6x-1.5.2/src/init.c	2015-06-07 03:29:28.000000000 +0000
+++ reaver-wps-fork-t6x-1.5.2/src/init.c	2015-06-07 03:29:28.000000000 +0000
@@ -134,7 +134,7 @@
     
         handle = pcap_create(capture_source,errbuf);
         if (handle) {
-                pcap_set_snaplen(handle, BUFSIZ);
+                pcap_set_snaplen(handle, 65535);
                 pcap_set_timeout(handle, 50);
                 pcap_set_rfmon(handle, 1);
                 pcap_set_promisc(handle, 1);
@@ -143,7 +143,7 @@
                         cprintf(CRITICAL, "pcap_activate status %d\n", status);
             }
     #else
-            handle = pcap_open_live(capture_source, BUFSIZ, 1, 0, errbuf);
+            handle = pcap_open_live(capture_source, 65535, 1, 0, errbuf);
     #endif
     if(!handle)
     {
```

rofl0r, 2016-12-04:
@xhebox: thanks! would you mind to open a PR with your change ? that way it's possible to see in the commit history the author.

xhebox, 2016-12-04:
Add a pr.

rofl0r, 2016-12-04:
thanks, i will now post further improvements. you are correct in that the code is of a very low quality and needs cleaning up :/

rofl0r, 2017-01-05:
after some time with @xhebox 's patch integrated, i can now say with high confidence that this did indeed fix the segfault bug! thanks again!

closed 2017-01-05 by rofl0r

xhebox, 2017-01-06:
you are welcome~