t6x/reaver-wps-fork-t6x #95: Executable for android

chrisk44, 2016-12-02:
Is there a way to build this as an executable for arm that will not require any other files to be present? I have compiled it for arm but when I run it, it says "No such file or directory". Obviously the file is there, but something it needs is not. I am currently using a reaver binary for arm on android but it's not this version, I want this for pixie-dust. Any help would be appreciated, thanks.

kcdtv, 2016-12-02:
Although i don't really get then concept of "executable for ARM" no reason why you could not.
Cannot help you more than that with so little information....

chrisk44, 2016-12-02:
yeah sorry, the key here was "that will not require any other files to be present". I'm writing an android app that provides a GUI for wireless auditing tools and I want to use this because of pixie-dust support. But since it's an android device, I can't just download the source, run ./configure, make and make install to install it. All the other tools I'm using (former version of reaver, aircrack-ng suite, mdk3) are standalone binaries, no other files required, just copy&paste to /system/bin and they work. I'm asking if there is a way to do this, as I have no idea how to make this happen. I can compile it for arm but whenever I run it, I get the message "File or directory not found", when the file is clearly there, which means that it's looking for something (lib, I don't know) that isn't there.

rofl0r, 2016-12-03:
what you want is called "static linking". in short you need to compile a static version of libsqlite3, libpcap and then link that with reaver by using `LDFLAGS=-static ./configure`

rofl0r, 2016-12-03:
also, the lib you're missing is probably the interpreter (dynamic linker). you can do `readelf -a reaver` and there you can see which libraries it needs (do `readelf -a reaver | grep -i needed` to see all required libs and `readelf -a reaver | grep -i interp` to see the interpreter).
also, in doubt `strace` tells you which file it's trying to open.

closed 2016-12-25 by chrisk44