t6x/reaver-wps-fork-t6x #97: Maybe a bug or not with PixieWPS?

Open1029, 2016-12-02:
I'm trying to hack a router maybe vulnerable by pixie dust attack...

So I launch reaver with this parameters:

`reaver  -I wlan0mon -b XX:XX:XX:XX:XX:XX -e ESSID -c 1 -K 1 -l 10 -t 10 -g 1 -vv` 

[image]

After that, when there is reaver cmd attack it stay freeze like in the pic...

So I try to cry and paste the command that reaver is trying to execute, but in another terminal...

`reaver  -I wlan0mon -b XX:XX:XX:XX:XX:XX -c 1 -s y -vv -p 12345670`

But reaver, differently form the reaver cmd line attack, returns me errors like in image: 

`[+] Sending EAPOL START request
  [!] Warning: Receive timeout occurred
` 

[image]

p.s: I noticied sometimes after my attacks router change his channel... 

so is it maybe a bug of reaver with wps? Or what can I do against this router , for example?

Thank you so much!

kcdtv, 2016-12-03:
That's not a bug 
You are not in correct conditions (interferences, weak signal) to do a correct ful WPS exchange.
Cheers

closed 2016-12-03 by kcdtv

kcdtv, 2016-12-04:
Pixiewps attack is done without even fully checking the first half of the PIN, 
You should read the thread of soxrock2212 about pixie dust attack in kali forum. 
Remember that this is a GitHub repository mainly to speak about code issues.
Not a forum about WiFi hacking