t6x/reaver-wps-fork-t6x #100: Error when compiling reaver t6x backbox 4.5 amd64

Boburjon, 2016-12-03:
Hi all, help me please
Error when compiling reaver t6x backbox 4.5 amd64
result ./configure 
`checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for pcap_open_live in -lpcap... yes
checking for sqlite3_open in -lsqlite3... yes
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for stdlib.h... (cached) yes
checking for stdint.h... (cached) yes
checking for string.h... (cached) yes
checking pcap.h usability... yes
checking pcap.h presence... yes
checking for pcap.h... yes
checking sqlite3.h usability... yes
checking sqlite3.h presence... yes
checking for sqlite3.h... yes
configure: creating ./config.status
config.status: creating Makefile`

result make
`(cd lwe && make BUILD_STATIC=y libiw.a)
make[1]: Вход в каталог `/root/reaver-wps-fork-t6x-master/src/lwe'
cp wireless.21.h wireless.h
gcc -Os -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Wpointer-arith -Wcast-qual -Winline -I. -MMD     -fPIC -c -o iwlib.so iwlib.c
iwlib.c: In function ‘iw_enum_devices’:
iwlib.c:254:14: warning: ignoring return value of ‘fgets’, declared with attribute warn_unused_result [-Wunused-result]
         fgets(buff, sizeof(buff), fh);
              ^
iwlib.c:255:14: warning: ignoring return value of ‘fgets’, declared with attribute warn_unused_result [-Wunused-result]
         fgets(buff, sizeof(buff), fh);
              ^
iwlib.c: In function ‘iw_get_kernel_we_version’:
iwlib.c:337:10: warning: ignoring return value of ‘fgets’, declared with attribute warn_unused_result [-Wunused-result]
     fgets(buff, sizeof(buff), fh);
          ^
iwlib.c:353:10: warning: ignoring return value of ‘fgets’, declared with attribute warn_unused_result [-Wunused-result]
     fgets(buff, sizeof(buff), fh);
          ^
rm -f libiw.a
ar cru libiw.a iwlib.so
ranlib libiw.a
make[1]: Выход из каталога `/root/reaver-wps-fork-t6x-master/src/lwe'
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Wno-unused-function globule.c -c 
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Wno-unused-function init.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Wno-unused-function misc.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Wno-unused-function -Ilibwps -I. sql.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Wno-unused-function builder.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Wno-unused-function crc.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Wno-unused-function -Ilibwps -I. 80211.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Wno-unused-function iface.c -c
[... 146 lines cut ...]
  CC  tls_internal.c
  CC  fips_prf_internal.c
ar crT libcrypto.a aes-cbc.o aes-ctr.o aes-eax.o aes-encblock.o aes-internal.o aes-internal-dec.o aes-internal-enc.o aes-omac1.o aes-unwrap.o aes-wrap.o des-internal.o dh_group5.o dh_groups.o md4-internal.o md5.o md5-internal.o md5-non-fips.o milenage.o ms_funcs.o rc4.o sha1.o sha1-internal.o sha1-pbkdf2.o sha1-tlsprf.o sha1-tprf.o sha256.o sha256-internal.o crypto_internal.o crypto_internal-cipher.o crypto_internal-modexp.o crypto_internal-rsa.o tls_internal.o fips_prf_internal.o
make[1]: Выход из каталога `/root/reaver-wps-fork-t6x-master/src/crypto'
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Wno-unused-function -Ilibwps -I. wpsmon.c globule.o init.o misc.o 80211.o iface.o builder.o crc.o libwps/libwps.o wps/wps_attr_build.o wps/wps_attr_parse.o wps/wps_attr_process.o wps/wps.o wps/wps_common.o wps/wps_dev_attr.o wps/wps_enrollee.o wps/wps_registrar.o wps/wps_ufd.o utils/libutils.a crypto/libcrypto.a tls/libtls.a lwe/libiw.a -ldl -lm -lpcap -lsqlite3  -o wash
wpsmon.c: In function ‘parse_wps_settings’:
wpsmon.c:361:26: warning: unused variable ‘encryption’ [-Wunused-variable]
     enum encryption_type encryption = NONE;
                          ^
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Wno-unused-function -Ilibwps -I. argsparser.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Wno-unused-function -Ilibwps -I. sigint.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Wno-unused-function sigalrm.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Wno-unused-function keys.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Wno-unused-function pins.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Wno-unused-function send.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Wno-unused-function -Ilibwps -I. exchange.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Wno-unused-function -Ilibwps -I. session.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Wno-unused-function -Ilibwps -I. cracker.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Wno-unused-function -Ilibwps -I. wpscrack.c argsparser.o globule.o init.o sigint.o iface.o sigalrm.o misc.o cracker.o sql.o session.o send.o pins.o 80211.o exchange.o builder.o keys.o crc.o libwps/libwps.o wps/wps_attr_build.o wps/wps_attr_parse.o wps/wps_attr_process.o wps/wps.o wps/wps_common.o wps/wps_dev_attr.o wps/wps_enrollee.o wps/wps_registrar.o wps/wps_ufd.o utils/libutils.a crypto/libcrypto.a tls/libtls.a lwe/libiw.a -ldl -lm -lpcap -lsqlite3  -o reaver
wpscrack.c: In function ‘main’:
wpscrack.c:134:19: warning: ignoring return value of ‘system’, declared with attribute warn_unused_result [-Wunused-result]
             system(get_exec_string());
`

what could be the cause of the error?

kcdtv, 2016-12-03:
it looks like more warnings than errors... 
Is that the complete stdout?
Can you install reaver (make install)?

rofl0r, 2016-12-03:
@GhostWarriorS : is that current git or do you use an older version ?

GhostWarriorS, 2016-12-03:
Yes, this is the make install result

> rm -f /usr/local/bin/reaver /usr/local/bin/wash /usr/local/bin/walsh
if [ ! -d /usr/local/etc/reaver ]; then mkdir -p /usr/local/etc/reaver; fi
if [ ! -e /usr/local/etc/reaver/reaver.db ]; then cp reaver.db /usr/local/etc/reaver/reaver.db && chmod -R a+rw /usr/local/etc/reaver; fi
if [ -e wash ]; then cp wash /usr/local/bin/wash; fi
if [ -e reaver ]; then cp reaver /usr/local/bin/reaver; fi

GhostWarriorS, 2016-12-03:
All agreed))) after apt-get dist-upgrade working

GhostWarriorS, 2016-12-04:
@rofl0r  current!

GhostWarriorS, 2016-12-04:
Error still relevant! details here: 
https://forum.backbox.org/software-support/error-when-compiling-reaver-t6x-backbox-4-5-amd64/

rofl0r, 2016-12-04:
@GhostWarriorS according to your paste in that forum you're using version 1.4 of reaver and not this fork here. this one here is 1.5.3 so please checkout this version from git and compile this version here before further questions.

GhostWarriorS, 2016-12-04:
@rofl0r The above warnings get when you compile the current version. It would be great if you wrote a script for a particular automated assembly with the right package.
I had to re-install backbox Linux 4.5. A freshly installed OS system comes with version 1.4 Reaver. I need to remove the Reaver 1.4 and compile Reaver 1.5 with the desired packages and worked wlan0mon monitoring mode instead mon0 like Kali Linux 2.0

**Why I deleted kali linux 2016.2?**
Because when I include command monitoring mode:
**airmon-ng check kill & airmon-ng start wlan0**
Everything works well. And in order to restore the operation of the network I have to restart the computer each time, otherwise it is impossible to switch on the wifi points. To put it simply the Kali linux after the network monitoring mode begins to fail until you restart the computer. And that's why I chose Backbox.

GhostWarriorS, 2016-12-04:
By the way I forgot to mention. I do not have a virtual machine, I ultrabook Samsung and Wifi card Ralink RT2870 / 3070 rt2800usb. If this information will somehow correct the error

kcdtv, 2016-12-04:
The issue you are talking about now is not related to reaver. 
If you use **check kill** your have to restart network manger to connect threw network manager once you put your interface back in managed mode.
Check with `sudo rfkill list`
If it is "Soft blocked:" than it is probably this. 
cheers

closed 2016-12-04 by kcdtv