t6x/reaver-wps-fork-t6x #106: keep the list of parameters compatible

blshkv, 2016-12-21:
We are facing an issue with wifite which is calling reaver with "-C" parameter:
https://github.com/pentoo/pentoo-overlay/issues/139

However, the following commit inverted it's meaning:
https://github.com/t6x/reaver-wps-fork-t6x/commit/63c5c02bfc44e732399118d844906b0708e8e28c

This is just wrong. Please introduce a new flag if you do that but keep it compatible with original code

kcdtv, 2016-12-21:
the -C tag in reaver is not "inverted" and it is there to execute a command after reaver

blshkv, 2016-12-21:
Dude, dont start again. I was talking about this project in general and wanted to bring a message to all of your developers to stay compatible and not to reverse meanings of *any* parameter randomly.

wpsmon.c (or wash) is just an example

rofl0r, 2016-12-21:
i agree that in this specific case the inversion of the meaning is very confusing, to say the least. maybe @t6x recalls why this was done ?

yuri-sevatz, 2016-12-21:
Sorry for the confusion everyone.  I found this while trying to fix WPS support in wifite2 during Christmas break.  I wrote a detailed explanation of the '-C inversion' problem here:

https://github.com/pentoo/pentoo-overlay/issues/139

My suspicion is that these lines should probably be reverted, because having different outcomes depending on which version of the wash, or the fork of it, are installed breaks runtime compatibility for scripts that try and communicate with arbitrary versions of wash.

From what I found, it doesn't look like wifite-v1 necessarily needs versions of wash from the forked repositories, but then when you don't have them installed you lose some functionality from the forks (so, maintainers make this choice optional).  That's why I think this was never an issue before for any distros shipping with the non-forked versions of reaver-wps-fork-t6x.

Wifite-v2 looks like it has a hard dependency on leveraging the pixie attack from reaver-wps-fork-t6x, so, when they copied the logic over from v1, it doesn't look like the old passing '-C' works anymore.  And that's precisely because it's meaning was inverted in this fork of wash.

Might be easier for upstream just to revert the '-C' inversion, though I don't know enough about scripts that depend on this new version of wash to make that criticism.

Btw @blshkv very thorough work on that patch, you fixed a lot of backwards-compatibility! 👍

blshkv, 2016-12-21:
My patch is very simple: the current -C flag  is renamed as -F,  while new  -C  does nothing since it is a new default behavior of this fork.  It is a quick hack but it works.
I'm  sure you  guys come up with a better one.

rofl0r, 2016-12-21:
submitted a PR. ^

rofl0r, 2016-12-21:
also note that the flatr0ze patch was so hasty that it did not even change the output of the help text:
```
-C, --announce-fcs                   Ignore frame checksum errors
```
the option was renamed to "announce-fcs", but the explanation text still talked about "Ignore".

kcdtv, 2016-12-21:
> Dude, dont start again.

  I would say exactly the same to you.
And don't call me dude: I am not your friend, your buddy or whatever.
Have a neutral and correct attitude and i will have a neutral and correct attitude.
Deal?
  Problem is not about the change but about the lack of presentation/explanation about it. 
  About wifite, just get read of the -C tag in your wash line, it is a python script so it is a very basic change.... 
  Wifite is a very nice code but inefficient (useless) and just for script kiddies
Wrappers to break the neighbor PIN are not the priority (at all) here. 
  For wifi pentest you need aircrack-ng and reaver-pixiewps, nothing more, nothing else, the rest is just superficial-redundant. 

> This is just wrong. Please introduce a new flag if you do that and keep it compatible with original code

  The focus here is to have a the most efficient minimal syntax with wash and that what this change is about.
  Like any code some option get "deprecated", updating a wrapper is up to you and so easy....   
  I will take the time to explain you why in detail with illustrated example when i am in condition to do so. 
  That means at home with my collection of chipset and my computer, and we will change the help-instuction that were not correctly updated (you can see that even the interface suggested is mon0, that changed years ago, the help-instructions were not updated/documented) 
  You need to get the whole picture and right now you do not
  Relax and have a merry xmas  
   
@ rofl0r

thanks again for your effort, sox and i are on holliday right now and will come back full of energy soon and we will speak about this and a couple of things more if you agree; some details that can be improved.
Take care and have a merry xmas :smile_cat: 

. .

kcdtv, 2017-01-02:
closed because we are dealing about it here: [https://github.com/t6x/reaver-wps-fork-t6x/pull/107](url)

closed 2017-01-02 by kcdtv