t6x/reaver-wps-fork-t6x #109: Reaver start from PIN 2 zte ZXHN H118N ?

slmafiq, 2017-01-02:
Hello !
What is happening here? Does anybody knows why the reaver starts from PIN 2
 p2_index set to 1
[+] Pin count advanced: 10001. Max pin attempts: 11000
[+] Trying pin 12340002.
90%
and after finish 99.99% reaver start from 50.00%

root@kali:~# reaver -i wlan1mon -b 78:96:82:38:xx:xx -S -vv -c 1

[+] Switching wlan1mon to channel 1
[+] Waiting for beacon from 78:96:82:38:xxxxx
[+] Associated with 78:96:82:38:xxxx (ESSID: AAAAAA)
[+] Starting Cracking Session. Pin count: 0, Max pin attempts: 11000
[+] Trying pin 12345670.
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[P] E-Nonce: 28:d6:ef:64:60:fe:9a:0d:52:ab:05:0f:68:67:63:ba
[P] PKE: d0:14:1b:15:65:6e:96:b8:5f:ce:ad:2e:8e:76:33:0d:2b:1a:c1:57:6b:b0:26:e7:a3:28:c0:e1:ba:f8:cf:91:66:43:71:17:4c:08:ee:12:ec:92:b0:51:9c:54:87:9f:21:25:5b:e5:a8:77:0e:1f:a1:88:04:70:ef:42:3c:90:e3:4d:78:47:a6:fc:b4:92:45:63:d1:af:1d:b0:c4:81:ea:d9:85:2c:51:9b:f1:dd:42:9c:16:39:51:cf:69:18:1b:13:2a:ea:2a:36:84:ca:f3:5b:c5:4a:ca:1b:20:c8:8b:b3:b7:33:9f:f7:d5:6e:09:13:9d:77:f0:ac:58:07:90:97:93:82:51:db:be:75:e8:67:15:cc:6b:7c:0c:a9:45:fa:8d:d8:d6:61:be:b7:3b:41:40:32:79:8d:ad:ee:32:b5:dd:61:bf:10:5f:18:d8:92:17:76:0b:75:c5:d9:66:a5:a4:90:47:2c:eb:a9:e3:b4:22:4f:3d:89:fb:2b
[P] WPS Manufacturer: ZTE Corporation
[P] WPS Model Name: ZXHN H118N
[P] WPS Model Number: ZXHN H118N
[P] Access Point Serial Number: 123456789012347
[+] Received M1 message
[P] R-Nonce: 3d:c2:d7:b2:26:17:fc:99:17:65:e4:4e:5e:86:81:18
[P] PKR: 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:02
[P] AuthKey: 12:77:db:6f:ee:3a:a3:cc:27:f6:14:c1:f5:62:9e:f2:85:7e:1c:b4:d8:76:09:12:44:83:39:9b:29:59:4c:bf
[+] Sending M2 message
[P] E-Hash1: 53:ad:3d:29:fa:95:94:f6:b1:da:4e:96:91:88:51:0b:d4:f8:f0:cf:c8:35:dc:3e:ea:b5:f1:e1:d9:e0:86:ae
[P] E-Hash2: 53:ad:3d:29:fa:95:94:f6:b1:da:4e:96:91:88:51:0b:d4:f8:f0:cf:c8:35:dc:3e:ea:b5:f1:e1:d9:e0:86:ae
[+] Received M3 message
[+] Sending M4 message
[+] Received M5 message
[+] Sending M6 message
[+] Received M5 message
[+] Sending WSC NACK
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x03), re-trying last pin
[+] Trying pin 12345670.
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[P] E-Nonce: 2b:0b:1c:73:05:55:22:6c:71:28:85:7c:66:07:15:2f
[P] PKE: d0:14:1b:15:65:6e:96:b8:5f:ce:ad:2e:8e:76:33:0d:2b:1a:c1:57:6b:b0:26:e7:a3:28:c0:e1:ba:f8:cf:91:66:43:71:17:4c:08:ee:12:ec:92:b0:51:9c:54:87:9f:21:25:5b:e5:a8:77:0e:1f:a1:88:04:70:ef:42:3c:90:e3:4d:78:47:a6:fc:b4:92:45:63:d1:af:1d:b0:c4:81:ea:d9:85:2c:51:9b:f1:dd:42:9c:16:39:51:cf:69:18:1b:13:2a:ea:2a:36:84:ca:f3:5b:c5:4a:ca:1b:20:c8:8b:b3:b7:33:9f:f7:d5:6e:09:13:9d:77:f0:ac:58:07:90:97:93:82:51:db:be:75:e8:67:15:cc:6b:7c:0c:a9:45:fa:8d:d8:d6:61:be:b7:3b:41:40:32:79:8d:ad:ee:32:b5:dd:61:bf:10:5f:18:d8:92:17:76:0b:75:c5:d9:66:a5:a4:90:47:2c:eb:a9:e3:b4:22:4f:3d:89:fb:2b
[P] WPS Manufacturer: ZTE Corporation
[P] WPS Model Name: ZXHN H118N
[P] WPS Model Number: ZXHN H118N
[P] Access Point Serial Number: 123456789012347
[+] Received M1 message
[P] R-Nonce: 8f:54:c8:2f:a0:8c:b6:ed:0a:91:6e:df:26:48:31:48
[P] PKR: 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:02
[P] AuthKey: 48:98:cf:9d:ab:5d:c8:23:12:d9:c3:5b:3e:0b:b3:c1:04:0f:cd:f2:5f:95:a6:dc:63:25:63:c8:54:7b:06:57
[+] Sending M2 message
[P] E-Hash1: 0c:7c:b0:04:07:5b:ff:47:8f:26:d8:51:19:3c:f7:10:30:c2:1f:5e:dd:d0:10:0a:56:1d:4e:de:f4:b3:7f:59
[P] E-Hash2: 0c:7c:b0:04:07:5b:ff:47:8f:26:d8:51:19:3c:f7:10:30:c2:1f:5e:dd:d0:10:0a:56:1d:4e:de:f4:b3:7f:59
[+] Received M3 message
[+] Sending M4 message
[+] Received M5 message
[+] Sending M6 message
[!] WARNING: Receive timeout occurred
[+] Sending WSC NACK
[+] p2_index set to 1
[+] Pin count advanced: 10001. Max pin attempts: 11000
[+] Trying pin 12340002.

[+] Trying pin 12343331.

rofl0r, 2017-01-02:
can you try it with this branch https://github.com/t6x/reaver-wps-fork-t6x/pull/107 and report back.
the first commit there undoes some dubious change that could be causing that

kcdtv, 2017-01-02:
The router sends a M5 when it should not.

> [+] Received M5 message
> [+] Sending M6 message
> [+] Received M5 message 

This routers is used by jazztel in Spain and I could observe this: Bruteforce is not possible because the routeur always send an M5 but does not respond to M6.
It looks like first half is correct 
  So reaver jumps to 50% because he received a M5 
  Try with your default PIN and you will get the WPA key. (-p option)
  You might need to use the **-n tag** too.

slmafiq, 2017-01-03:
root@kali:~# reaver -i wlan1 -b 78:96:82:38XXXXX -c 1 -n -S -vv

Reaver v1.5.2 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
mod by t6_x <t6_x@hotmail.com> & DataHead & Soxrok2212

[+] Switching wlan1 to channel 1
[+] Waiting for beacon from 78:96:82:38:XXXX
[+] Associated with 78:96:82:38:2XXX (ESSID: aaaaa)
[+] Starting Cracking Session. Pin count: 0, Max pin attempts: 11000
[+] Trying pin 12345670.
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[P] E-Nonce: 12:fb:42:94:50:c6:b4:76:31:72:0c:ad:37:c6:56:b8
[P] PKE: d0:14:1b:15:65:6e:96:b8:5f:ce:ad:2e:8e:76:33:0d:2b:1a:c1:57:6b:b0:26:e7:a3:28:c0:e1:ba:f8:cf:91:66:43:71:17:4c:08:ee:12:ec:92:b0:51:9c:54:87:9f:21:25:5b:e5:a8:77:0e:1f:a1:88:04:70:ef:42:3c:90:e3:4d:78:47:a6:fc:b4:92:45:63:d1:af:1d:b0:c4:81:ea:d9:85:2c:51:9b:f1:dd:42:9c:16:39:51:cf:69:18:1b:13:2a:ea:2a:36:84:ca:f3:5b:c5:4a:ca:1b:20:c8:8b:b3:b7:33:9f:f7:d5:6e:09:13:9d:77:f0:ac:58:07:90:97:93:82:51:db:be:75:e8:67:15:cc:6b:7c:0c:a9:45:fa:8d:d8:d6:61:be:b7:3b:41:40:32:79:8d:ad:ee:32:b5:dd:61:bf:10:5f:18:d8:92:17:76:0b:75:c5:d9:66:a5:a4:90:47:2c:eb:a9:e3:b4:22:4f:3d:89:fb:2b
[P] WPS Manufacturer: ZTE Corporation
[P] WPS Model Name: ZXHN H118N
[P] WPS Model Number: ZXHN H118N
[P] Access Point Serial Number: 123456789012347
[+] Received M1 message
[P] R-Nonce: fc:e3:83:d3:77:3c:0d:5c:74:49:91:c4:26:13:90:e2
[P] PKR: 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:02
[P] AuthKey: 4a:bb:ab:03:e6:3d:4d:e4:d3:51:a2:c0:be:b2:7a:ef:9a:06:0e:d3:9d:dd:74:bc:fa:38:25:b3:e5:ed:9f:a4
[+] Sending M2 message
[P] E-Hash1: 68:ff:da:43:7d:e0:1e:f6:67:46:9f:42:cb:dc:00:5c:ef:fa:7b:86:ec:c3:36:ad:a2:5f:5e:64:b1:7a:96:52
[P] E-Hash2: 68:ff:da:43:7d:e0:1e:f6:67:46:9f:42:cb:dc:00:5c:ef:fa:7b:86:ec:c3:36:ad:a2:5f:5e:64:b1:7a:96:52
[+] Received M3 message
[+] Sending M4 message
[+] Received M5 message
[+] Sending M6 message
[+] Received M5 message
[+] Sending WSC NACK
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x03), re-trying last pin
[+] Trying pin 12345670.
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[P] E-Nonce: 65:c6:10:bc:4d:79:f6:30:19:5a:49:ad:6d:17:97:82
[P] PKE:........................
..................................
with -n options

--------------------------------------------------------------------------------------------------------
----------------------------------------------------------------------------------------------------------
root@kali:~# reaver -i wlan1 -b 78:96:82:38:2E:AC -c 1 -S -vv
See this : 
start from 
[+] Pin count advanced: 10005. Max pin attempts: 11000

[+] 90.95% complete. Elapsed time: 0d0h0m13s.
and 
after  finish 999 pin attempts
[+] Checksum mode was not successful. Starting exhaustive attack

[+] Pin count advanced: 10001. Max pin attempts: 20000

[+] p2_index set to 0
and start from 50 %
wtf ? :D 
root@kali:~# reaver -i wlan1 -b 78:96:82:38:2E:AC -c 1 -S -vv

Reaver v1.5.2 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
mod by t6_x <t6_x@hotmail.com> & DataHead & Soxrok2212

[+] Switching wlan1 to channel 1
[... 387 lines cut ...]
[P] PKR: 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:02
[P] AuthKey: 5e:1f:75:5f:ab:02:39:2f:a4:43:1d:98:d3:3a:e3:26:fb:7f:1e:d8:80:6d:1a:52:ff:92:51:16:ab:45:55:ca

[+] Sending M2 message
[P] E-Hash1: cd:cd:2e:27:65:61:0d:3c:97:27:7d:b0:11:63:41:39:8f:e1:8e:4f:62:07:b2:3c:46:85:1a:dd:de:9e:29:54

[P] E-Hash2: cd:cd:2e:27:65:61:0d:3c:97:27:7d:b0:11:63:41:39:8f:e1:8e:4f:62:07:b2:3c:46:85:1a:dd:de:9e:29:54

[+] Received M3 message

[+] Sending M4 message

[+] Received M5 message

[+] Sending M6 message

[!] WARNING: Receive timeout occurred

[+] Sending WSC NACK

[+] p2_index set to 1

[+] Pin count advanced: 10001. Max pin attempts: 20000

50.01 complete. Elapsed time 31 min

rofl0r, 2017-01-03:
that's a wall of text. can you point out/summarize the, according to your opinion, relevant bits ?

btw just a thought that came after reading @kcdtv 's analyzation: some routers re-send packets to fight packet loss (can be seen by pinging them which returns DUPs), so if a second M5 comes this could be such a case, and it would be better to just ignore the re-submitted paket instead of rewinding the internal state.

slmafiq, 2017-01-03:
Is there any way to find my password :D ? or it is  impossible?
I just dont understand why it counts till 90% 30 mn in the beginning and then after 999
 attempts 
 Pin count advanced: 11000. Max pin attempts: 11000
Checksum mode was not successful. Starting exhaustive attack
 p2_index set to 0
 p2_index set to 1

[+] Pin count advanced: 10001. Max pin attempts: 20000

50.01 complete. Elapsed time 31 min

kcdtv, 2017-01-04:
I scrolled the whole stdout and it is the same:  There is a "fake" M5 that make impossible the brute force in two phases.  

> Is there any way to find my password :D ? or it is impossible?

Send your PIN with -p option
`reaver -i <interace> -b <bssid> -c <channel> -p <pin> -n`
And put the stdout here
by the way put the stdout with any PIN (that doesn't start with 1234) you will see that you'll get an M5 eventhough your first half is different (you shouldn't)

> btw just a thought that came after reading @kcdtv 's analysis: some routers re-send packets to fight packet loss (can be seen by pinging them which returns DUPs), so if a second M5 comes this could be such a case, and it would be better to just ignore the re-submitted paket instead of rewinding the internal state.

Acknowledging the first M5 is enough to create the fake positive for the first half. Problem here is that this M5 should not exist and totally disable the concept of two stages brute force.

slmafiq, 2017-01-04:
@kcdtv  Thanks for all! This is the answer :)
if i had known the pin code i would not search it :D 

is it possible to in the future to find another  way?
Or Router TL-WR740N 
if you have program from test write me personal webhosting [kordokolenkov@abv.bg]

slmafiq, 2017-01-11:
Is it possible to be made script, which could use PIN LIST for first half or for the whole pin, with 11 000 possibilities for imitation normal brute force?
ex: reaver -i wlan1mon -b X:X:X:X:X:X:X -vv -c 11 -p 12345670
next -p 11111111
next -p 11112345
next -p ........................

soxrok2212, 2017-03-10:
inactive

closed 2017-03-10 by soxrok2212