t6x/reaver-wps-fork-t6x #111: p2_index gets increased even when the pin try failed.

rofl0r, 2017-01-05:
that's quite a showstopper, and basically breaks bruteforcing the pin entirely.
this is a bug specific to this fork, it works well in the original reaver 1.4.

for example:
```
[+] Received M5 message
[!] WARNING: Receive timeout occurred
[+] Sending WSC NACK
[+] p2_index set to 10
[+] Pin count advanced: 10010. Max pin attempts: 11000
[+] Trying pin 03057957.

...

[+] Received M3 message
[!] WARNING: Receive timeout occurred
[+] Sending WSC NACK
[+] p2_index set to 9

...

[+] Received M5 message
[!] WARNING: Receive timeout occurred
[+] Sending WSC NACK
[+] p2_index set to 8
[+] Pin count advanced: 10008. Max pin attempts: 11000
[+] Trying pin 03057957.

```
(note i started reaver with -p 03057957)

if we dont get that bug fixed, reaver is broken for anything else as a targeted pin try (i.e. if one knows the default pin algorithm).

rofl0r, 2017-01-05:
i think someone that has good testing opportunities (i.e. you see an AP with WPS on, optimally one that doesnt lock you out, that is sufficiently far away that its unlikely you'll ever get to a successful try) should use `git bisect` to find the commit that introduced the bug.
here's a git bisect tutorial: https://www.metaltoad.com/blog/beginners-guide-git-bisect-process-elimination

kcdtv, 2017-01-06:
That the same issue than here:  [https://github.com/t6x/reaver-wps-fork-t6x/issues/50](url)

sheepdestroyer, 2017-01-06:
I get an error on this last link

rofl0r, 2017-01-06:
@kcdtv your link above is broken. it *shows* the right link, but the link itself is bad.
just fyi you can make a link to an issue by using just `#50`

amazer-bil, 2017-01-07:
Hi guys....a help if that possible :)
 
a way to crack latest rooters such as ' technicolor or broadcom or dlink because when i run reaver and pixiewps it shows in the end ' wps pin not found ' and thanks :)

soxrok2212, 2017-01-07:
No. stop asking

rofl0r, 2017-06-22:
should be fixed now with 1.6b.

closed 2017-06-22 by rofl0r