t6x/reaver-wps-fork-t6x #117: Wash not returning any result

evilkid, 2017-01-17:
Running under Ubuntu 16.04, 
When using the old reaver, i have to pass the --ignore-fcs argument to get a result using wash.
However, this version does not have the --ignore-fcs argument, so when running: sudo wash -i mon0 i get not result, i even tried to redirect it to file using -o, but still no results.

kcdtv, 2017-01-17:
Could you tell us the exact model of your chipset?
Paste here the results of please
`sudo airmon-ng`
Thanks for reporting

evilkid, 2017-01-17:
Interface	Chipset		Driver

wlp3s0		Atheros AR9285	ath9k - [phy0]
mon0		Atheros AR9285	ath9k - [phy0]

kcdtv, 2017-01-17:
My ath9k device is working fine under kali... 
The  _--ignore-fcs argument_ has been removed because it is now the default behavior.
No writing nothing is now like wirtting **-C** before.
  Actually your issue has nothing to do with the problem solved by the option. 
  I guess it is more a "system configuration issue"
You are using a very old version of aircrack-ng, you should start by updating your system... 
I am gonna check in ubuntu 16.10 now to see what happens.

evilkid, 2017-01-17:
Thank you for your reply,
Well using the old it worked with the ignore-fcs, and I have installed the aircrack-ng straight from the repo, its "Aircrack-ng 1.2 beta3".
But couldnt be because it needed the "libpcap0.8-dev"? Because according to this answer, http://askubuntu.com/a/714807 he said that it needed the libpcap0.8_1.4.0-2_i386 version.

kcdtv, 2017-01-17:
I've tried right now in a freshly installed ubuntu 16.10 that i have for testing. 
It is now enough with ubuntu 16.04 and 16.10 with **libcap-dev**.
 There is no more need for a downgrade to old package, that was true with some previous release, it is now working good with the general package for **libcap**  development libraries.
Just follow what is saied here in the instruction and do 
`sudo apt install libpcap-dev`
  My chipset also use ath9k driver and it works correctly with **libcap-dev** only (that will automatically install libpcap0.8-dev in the latest version for amd64 architecture)
  It is better to don't use a downgraded package to avoid conflict with other packages. 
Everything up to dated... 
  For aircrack-ng you have to follow the indication from aircrack-ng wiki. Ubuntu is not a pentest system and the version in its repositories is totally out of date. [https://www.aircrack-ng.org/doku.php?id=install_aircrack](url)

closed 2017-01-17 by kcdtv