t6x/reaver-wps-fork-t6x #118: Wash

FightingForFun, 2017-01-28:
Good day.

T6X can you modify the wash that he displayed when scanning the vendors name of the router by MAC / IEEE OUI. http://standards-oui.ieee.org/oui.txt

Example:
```
BSSID                       Channel       RSSI       WPS Version       WPS Locked        ESSID           VENDOR
------------------------------------------------------------------------------------------------------------------------------------------
C8:6C:87:00:00:00          1              00                 1.0                       No                FNET               ZyXEL
```

rofl0r, 2017-01-28:
not a bad idea actually.
note that wash is not a person, so using `he` is not adequate. it's an `it`.
btw i noticed that almost all programs that use that mac address database (wireshark, nmap, kismet, ...) ship their own copies which are put into some app-specific dir, thereby duplicating a lot of info. sometimes they even use incompatible format.
i would advise making some kind of standardized oui package that stores its stuff in /share/oui; and other apps can make use of that.
@kcdtv when you're reading this could you check by the chatroom? i've some things to discuss.

kcdtv, 2017-01-30:
Not a bad one but the problem is the list... 
We cannot rely on external file (not everybody have a OUI table it in it system)
aircrack-ng is a mandatory dependencies and have a OUI file... but not by default (you have to do download it with `airodump-ng oui` update comand) WE canot be sure it will be installed

> i would advise making some kind of standardized oui package that stores its stuff in /share/oui; and other apps can make use of that.

I don't think it is a good idea to put a nother list in an other directory ... at least we could use the list of aircrack-ng which is not in /usr/share. If there is a chance that one iee list being used by other ap, it is this one, not a new one sepcific to reaver. 
Idea is not bad but itself but it has some annoying second side effects.. 

> @kcdtv when you're reading this could you check by the chatroom? i've some things to discuss.

A bit late lol .... I'll try to be around the chat today.... if you wake up before 2 afternoon we can see each other.... :smile_cat:

ZeroChaos-, 2017-01-30:
If you want to do it like aircrack-ng here it is:

https://github.com/aircrack-ng/aircrack-ng/blob/master/src/airodump-ng.h#L145-L155

Basically we check a bunch of standard paths.  Also of note, most distros have the oui database as a dependency for aircrack-ng so you can almost just assume it's there.

closed 2017-09-13 by soxrok2212