t6x/reaver-wps-fork-t6x #121: Extra point after PIN in stdout

kcdtv, 2017-02-06:
I don't know when it appeared but this point was not always here before...
[image]
Does anyone have one idea?

horrorho, 2017-02-06:
Looking at the corresponding code line [here](https://github.com/t6x/reaver-wps-fork-t6x/blob/b5b4b78a71c6fca5db1aed4791a3cecfd232ee24/src/cracker.c#L165):
```c
 cprintf(WARNING, "[+] Trying pin %s.\n", pin);
```
It seems it's been like that since the initial commit [here](https://github.com/t6x/reaver-wps-fork-t6x/blob/f52c553bbff05e8fa03d64732d772f04e099c90c/src/cracker.c#L169) two years ago. It's a subtle visual change so is it possible it's just not been noticed?

kcdtv, 2017-02-07:
It could be...
I noticed it at beginning, I thought it went away at some point, but I am not 100% sure... It could have been always here too. :smiley_cat:

closed 2017-02-07 by kcdtv