t6x/reaver-wps-fork-t6x #122: Improved behaviour regarding -n auto

horrorho, 2017-02-06:
Hi! Follow up of the conversation [here](https://github.com/t6x/reaver-wps-fork-t6x/pull/119#issuecomment-276361160). Technically it doesn't add new functionality but improves the user experience for novices.

Scenario: AP that is correctly sending NACKs operating in an environment where frequent time outs are occuring (e.g. long distance to AP, RF congestion).  Reaver operating _without_ the `-n` switch, i.e. auto mode.

As it stands Reaver will default to using [time outs as NACKs](https://github.com/t6x/reaver-wps-fork-t6x/blob/b5b4b78a71c6fca5db1aed4791a3cecfd232ee24/src/exchange.c#L268) until it receives a NACK, at which point it will switch to `-n` [automatically](https://github.com/t6x/reaver-wps-fork-t6x/blob/b5b4b78a71c6fca5db1aed4791a3cecfd232ee24/src/exchange.c#L248). From this point on Reaver will correctly treat time outs as failures and retry.

The problem arises where we have cycles of time outs which are incorrectly treated as NACKs before `-n` auto kicks in, as we can potentially miss a valid pin. As default pins are considered first, the probability of this happening is higher than we might otherwise expect. This possibly explains the situation [here](https://github.com/t6x/reaver-wps-fork-t6x/pull/119#issuecomment-276223457).

These false negatives are recognised as a [problem](https://github.com/t6x/reaver-wps-fork-t6x/blob/b5b4b78a71c6fca5db1aed4791a3cecfd232ee24/src/exchange.c#L265) and the user can specify `-n` to correct this. However, it relies on the user recognising the issue which may not be the case.

As an alternative, we can consider changing Reaver's behaviour to respectively retry pins once the `-n` [auto](https://github.com/t6x/reaver-wps-fork-t6x/blob/b5b4b78a71c6fca5db1aed4791a3cecfd232ee24/src/exchange.c#L248) kicks in. It will then have double checked the time outs it had encountered before were not false negatives.

Illustrative pseudo sessions. Let's say our imaginary AP correctly sends NACKs and has a WPS pin of 00000001.

Pseudo session 1. Reaver's current behaviour _without_ the `-n` switch, i.e. auto mode.
```
Trying pin: 00000001
Time-out received (treated as NACK). False negative.
Trying pin: 00000002
NACK received. -n auto activated.
Trying pin: 00000003
Time-out received (now correctly not treated as a NACK).
Trying pin: 00000003
NACK received. False.
Trying pin: 00000004
NACK received. False.
...
Eventually.. WPS pin not found.
```

Pseudo session 2. Proposed behaviour of retrospectively checking time outs for false negatives.
```
Trying pin: 00000001
Time-out received (treated as NACK). False negative.
Trying pin: 00000002
NACK received. -n auto activated.
Retrospectively check.
Trying pin: 00000001
Time-out received (now correctly not treated as a NACK).
Trying pin: 00000001
Exchange completes. Pin found! 00000001
```

Normally, we would only expect a few false negatives before Reaver encounters a NACK and auto kicks in `-n`. If we are trying a single pin, then we might need to consider multiple attempts to exclude false negatives.

Implementation should be relatively simple. Retrospectively checking basically means Reaver restarts the session with `-n` enabled. We could persistently save the `-n` state between user sessions.

It's probably not as complicated as I've made it sound. Thank you!

**TLDR:** Reaver should restart the session when `-n` [auto](https://github.com/t6x/reaver-wps-fork-t6x/blob/b5b4b78a71c6fca5db1aed4791a3cecfd232ee24/src/exchange.c#L248) kicks in.

kcdtv, 2017-02-07:
Very interesting and "seducing"
  Using the -n argument and understanding why and when is indeed very "technical" even for somebody that really did an effort of reading and searching about WPS cracking. 
  There were (is) a lot of misunderstanding around the so called "99,99% loop-bug" 
  Pseudo session 1 is one case that would lead at the end to this bug: If the session goes on, the user will get to the point were the PIN 9999XXXX is sent continuously with progress at 99,99% once all the first half of PIN have been tried. 
  The "walk around" I used was to keep a duplicated stdout and then goes back to the suspicious PIN if that happens. 
  It would be so better to do it "live" and after a couple of "suspicious" PIN instead of realizing it after sending 10 000 PIN.... (like in pseudo session 2) 
  I think that it is something that was missing to this "[auto]" feature...  It is strange to have to explicitly use it  when the help says "auto".  
  It is really hard to guess for a random user that using doing so could change something, isn't it?  

> Retrospectively checking basically means Reaver restarts the session with -n enabled. We could persistently save the -n state between user sessions.

  I don't see how to do that without adding some new parameter in the *.wpc file. (saved session) 
   But don't pay too much attention, i am not a programmer, if you have an idea, for me it is fine. :smile_cat: 
   I really like your idea

horrorho, 2017-02-08:
> I don't see how to do that without adding some new parameter in the *.wpc file. (saved session)
But don't pay too much attention, i am not a programmer, if you have something in mind, for me it is fine. :smile_cat:

Yes, that's a good point. I'll try and figure out a solution over the weekend.

commit 40d3cb66 (2017-12-15): reaver: activate -n behaviour by default and obsolete the switch

most APs are conforming to the standard and send a NACK as expected
when the pin is wrong, but reaver defaulted to assuming the opposite.
this could cause reaver to wrongly assume a correct pin is incorrect
in non-optimal network conditions.

the old default behaviour to treat a timeout as NACK can be accessed
using the new -J / --timeout-is-nack switch.

closes #122
closed 2017-12-15 by rofl0r