t6x/reaver-wps-fork-t6x #124: Crash when using -g and -f

crass, 2017-02-19:
This crash is at wpsmon.c line 479 because iface is NULL (we're using -f and giving a pcap file).  Currently the implementation for getting the chipset uses a reaver instance listening on an interface to get chipset parameters from M1, so an interface is needed.  The option parsing code should disallow -g in conjunction with -f.

kcdtv, 2017-02-20:
> The option parsing code should disallow -g in conjunction with -f.

  Or when -g is used with -f it should requires a bssid to stdout the "chipset" information from a precise target and not look for an interface... That is another possibility 
Thank you very much for reporting this, we'll have it in mind.  
The point is that "anyway" **-g option** is not optimal and I think that we should "revisit it" 
 I think that it would be better to parse a PROBE answer and to not use a PIN attempt (AP rate limiting problem) to get this information.
  I think also that it would be better to simply display the raw content of the WPS tag parameters founds in the PROBE. 
  It is easiest and more complete. that is maximum 20-30 lines, nothing to be sacred about.

soxrok2212, 2017-03-10:
we are aware of this and have a better solution in mind

closed 2017-03-10 by soxrok2212