t6x/reaver-wps-fork-t6x #127: did anyone try with macos ?

jonasman, 2017-02-23:
Im trying this tool with mac os after tried the original reaver 1.4 but it is the same.

(the makefile wont install but it compiles fine)

I get timeouts no matter what options i try, also tried many wifi networks:

```
$ sudo ./reaver -i en0 -b 92:EF:68:B7:D5:6C -K 1 -vv

Reaver v1.5.3 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
mod by t6_x <t6_x@hotmail.com> & DataHead & Soxrok2212 & Wiire & AAnarchYY & KokoSoft

[+] Waiting for beacon from 92:EF:68:B7:D5:6C
[+] Switching en0 to channel 1
[+] Associated with 92:EF:68:B7:D5:6C (ESSID: Southfork_wlan)
[+] Starting Cracking Session. Pin count: 0, Max pin attempts: 11000
[+] Trying pin 12345670
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request

```

kcdtv, 2017-02-24:
You first need to find and install drivers that fully support  mode monitor and packet injection.

jonasman, 2017-02-24:
The mac supports monitor mode. Maybe the packet inkection is the issue. Have you tried using linux with a virtual machine ? Could that work?

kcdtv, 2017-02-24:
> Maybe the packet inkection is the issue.

For sure it is. 

> Have you tried using linux with a virtual machine ? Could that work?

For sure it could as long as your wifi device fully support monitor mode and packet injection.

closed 2017-02-24 by kcdtv

jonasman, 2017-02-24:
Thanks for the info. 
Maybe we can add to reaver a message saying "card not supported" when a packate cannot be injected?

jonasman, 2017-02-24:
Do you mean that my mac card might be able to do packet inkection in linux (virtual machine) due to different drivers?

kcdtv, 2017-02-25:
> Do you mean that my mac card might be able to do packet inkection in linux (virtual machine) due to different drivers?

Yes it could be possible. Although i am not 100% sure that it can be done with an internal card. But if you can configure your Virtual Machine to use the internal wifi card, if the chipset is compatible in linux, it will inject. 

> Thanks for the info.
Maybe we can add to reaver a message saying "card not supported" when a packate cannot be injected?

Usr is supposed to use arimon-ng and check with aireplay-ng command 9 if injection is working before using aircrack-ng, reaver, bully or whichever tool that requires packet injection. I am not really for adding more code for a function that is really not easy to set... so many cases, so many kernel versions-drivers, so many distributions...

jonasman, 2017-02-26:
I tried with VirtualBox - Not possible, VB does not emulate or pass-trough the internal wifi card
VMWare fusion - Not possible, same as VB.