t6x/reaver-wps-fork-t6x #128: Wash stops with no results

philipperemy, 2017-03-01:
Hey guys,

I have a small problem here. `wash` exits one second after it was started. `airodump` works like a charm and all the compilation stuffs went good.

I'm running on ubuntu 14.04.

Can you guys help?

Thanks!

```
wlan1		Unknown 	rt2800usb - [phy0]
				(monitor mode enabled on mon0)
```

```
sudo wash -i mon0

Wash v1.5.3 WiFi Protected Setup Scan Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner
mod by t6_x<t6_x@hotmail.com>, DataHead, Soxrok2212, Wiire, AAnarchYY & rofl0r

BSSID              Ch  dBm  WPS  Lck  ESSID
--------------------------------------------------------------------------------

```

kcdtv, 2017-03-01:
Using the outdated version of aircrack-ng available in ubuntu repositories is not recommended. 
I recommend you to remove the version you have
`sudo apt-get autoremove aircrack-ng ` 
Than download and install aircrack-ng as it is recommended in aircrack-ng website. One way (i would say the best) is to use svn repositories to get the very lest revision including the "experimental" tools (aircrack-ng GUI, airoscript, tkipun-ng etc...) . 
So that would be the dependencies for Ubuntu for a full instalation with experimental tools: 
`sudo apt-get install libnl-3-dev libssl-dev sqlite3 libsqlite3-dev libnl-genl-3-dev g++ subversion`
Than download the source code from subversion
`svn co http://svn.aircrack-ng.org/trunk/ aircrack-ng`
And install
`cd aircrack-ng && make sqlite=true experimental=true ext_scripts=true && sudo make sqlite=true experimental=true ext_scripts=true install`
It should go better now with wash.

philipperemy, 2017-03-07:
Hey!

I did what you said and unfortunately I'm still experiencing issues:

Aircrack-ng seems to be up to date:

```
aircrack-ng --help

  Aircrack-ng 1.2 rc4 r2895 - (C) 2006-2015 Thomas d'Otreppe
  http://www.aircrack-ng.org
```

```
airmon-ng start wlan2

Found 4 processes that could cause trouble.
If airodump-ng, aireplay-ng or airtun-ng stops working after
a short period of time, you may want to run 'airmon-ng check kill'

  PID Name
  837 avahi-daemon
  846 avahi-daemon
  868 NetworkManager
  992 wpa_supplicant

PHY	Interface	Driver		Chipset

phy0	wlan2		rt2800usb	Ralink Technology, Corp. RT2870/RT3070

		(mac80211 monitor mode vif enabled for [phy0]wlan2 on [phy0]wlan2mon)
		(mac80211 station mode vif disabled for [phy0]wlan2)
```

```
wash -i wlan2mon

Wash v1.5.3 WiFi Protected Setup Scan Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner
mod by t6_x<t6_x@hotmail.com>, DataHead, Soxrok2212, Wiire, AAnarchYY & rofl0r

BSSID              Ch  dBm  WPS  Lck  ESSID
--------------------------------------------------------------------------------
```

rofl0r, 2017-03-07:
can you start wash with gdb: `gdb --args wash -i wlan2mon ...`
then put a breakpoint on exit `b exit` and run it `r`.
once wash calls exit, enter `bt` and paste the backtrace here.

philipperemy, 2017-03-08:
Here is what I have:

Thanks for the help!

```
sudo gdb --args wash -i wlan1mon
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from wash...done.
(gdb) b exit
Breakpoint 1 at 0x402430
(gdb) r
Starting program: /home/philippe/Desktop/reav/pixiewps/reaver-wps-fork-t6x/src/wash -i wlan1mon

Wash v1.5.3 WiFi Protected Setup Scan Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner
mod by t6_x<t6_x@hotmail.com>, DataHead, Soxrok2212, Wiire, AAnarchYY & rofl0r

BSSID              Ch  dBm  WPS  Lck  ESSID
--------------------------------------------------------------------------------

Breakpoint 1, __GI_exit (status=0) at exit.c:104
104	exit.c: No such file or directory.
```

philipperemy, 2017-03-08:
Just for information, I've checked in the `MakeFile` and the debug flag is here:
`CFLAGS=-DCONF_DIR='"$(CONFDIR)"' -Wall -g -O2 -Wno-unused-function`

rofl0r, 2017-03-08:
yeah, now you should paste the output of `bt` after you get exactly there

philipperemy, 2017-03-09:
```
(gdb) bt
#0  __GI_exit (status=0) at exit.c:104
#1  0x00007ffff74f2f4c in __libc_start_main (main=0x402500 <main>, argc=3, 
    argv=0x7fffffffe5a8, init=<optimized out>, fini=<optimized out>, 
    rtld_fini=<optimized out>, stack_end=0x7fffffffe598) at libc-start.c:321
#2  0x00000000004029bc in _start ()
```

philipperemy, 2017-03-09:
Just for info if I execute `p bt` I have:
```
104	exit.c: No such file or directory.
(gdb) p bt
No symbol "bt" in current context.
```
Thanks for the help!

rofl0r, 2017-03-09:
ok that at least tells us at return is called from main(), and not exit().
in that case it's probably `next_packet()` that returns NULL.
**why** is the question.
you could try to set a bp on next_packet: `b next_packet`, then run `r`, then several times `n` until the function exits and a couple lines more.

philipperemy, 2017-03-09:
```
Starting program: /usr/local/bin/wash -i wlan1mon

Wash v1.5.3 WiFi Protected Setup Scan Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner
mod by t6_x<t6_x@hotmail.com>, DataHead, Soxrok2212, Wiire, AAnarchYY & rofl0r

BSSID              Ch  dBm  WPS  Lck  ESSID
--------------------------------------------------------------------------------

Breakpoint 1, next_packet (header=header@entry=0x7fffffffe0b0) at 80211.c:38
38	{
(gdb) n
47	    while((packet = pcap_next(get_handle(), header)) != NULL)
(gdb) 
56	        if(get_validate_fcs())
(gdb) 
76	}
(gdb) 
monitor (bssid=bssid@entry=0x0, passive=passive@entry=0, source=source@entry=0, channel=channel@entry=0, mode=mode@entry=0) at wpsmon.c:344
344	        parse_wps_settings(packet, &header, bssid, passive, mode, source);
(gdb) 
346	        memset((void *) packet, 0, header.len);
(gdb) 
342	    while((packet = next_packet(&header)))
(gdb) 

Breakpoint 1, next_packet (header=header@entry=0x7fffffffe0b0) at 80211.c:38
38	{
(gdb) 
47	    while((packet = pcap_next(get_handle(), header)) != NULL)
(gdb) 
56	        if(get_validate_fcs())
(gdb) 
76	}
(gdb) 
monitor (bssid=bssid@entry=0x0, passive=passive@entry=0, source=source@entry=0, channel=channel@entry=0, mode=mode@entry=0) at wpsmon.c:344
344	        parse_wps_settings(packet, &header, bssid, passive, mode, source);
(gdb) 
346	        memset((void *) packet, 0, header.len);
(gdb) 
342	    while((packet = next_packet(&header)))
(gdb) 
```

```[...]```

```
(gdb) 
monitor (bssid=bssid@entry=0x0, passive=passive@entry=0, source=source@entry=0, channel=channel@entry=0, mode=mode@entry=0) at wpsmon.c:344
344	        parse_wps_settings(packet, &header, bssid, passive, mode, source);
(gdb) 
346	        memset((void *) packet, 0, header.len);
(gdb) 
342	    while((packet = next_packet(&header)))
(gdb) 

Breakpoint 1, next_packet (header=header@entry=0x7fffffffe0b0) at 80211.c:38
38	{
(gdb) 
47	    while((packet = pcap_next(get_handle(), header)) != NULL)
(gdb) 
56	        if(get_validate_fcs())
(gdb) 
76	}
(gdb) 
monitor (bssid=bssid@entry=0x0, passive=passive@entry=0, source=source@entry=0, channel=channel@entry=0, mode=mode@entry=0) at wpsmon.c:344
344	        parse_wps_settings(packet, &header, bssid, passive, mode, source);
(gdb) 
346	        memset((void *) packet, 0, header.len);
(gdb) 
342	    while((packet = next_packet(&header)))
(gdb) 

Breakpoint 1, next_packet (header=header@entry=0x7fffffffe0b0) at 80211.c:38
38	{
(gdb) 
47	    while((packet = pcap_next(get_handle(), header)) != NULL)
(gdb) 
56	        if(get_validate_fcs())
(gdb) 
76	}
(gdb) 
monitor (bssid=bssid@entry=0x0, passive=passive@entry=0, source=source@entry=0, channel=channel@entry=0, mode=mode@entry=0) at wpsmon.c:344
344	        parse_wps_settings(packet, &header, bssid, passive, mode, source);
(gdb) 
346	        memset((void *) packet, 0, header.len);
(gdb) 
342	    while((packet = next_packet(&header)))
(gdb) 

Breakpoint 1, next_packet (header=header@entry=0x7fffffffe0b0) at 80211.c:38
38	{
(gdb) 
47	    while((packet = pcap_next(get_handle(), header)) != NULL)
(gdb) 
76	}
(gdb) 
monitor (bssid=bssid@entry=0x0, passive=passive@entry=0, source=source@entry=0, channel=channel@entry=0, mode=mode@entry=0) at wpsmon.c:351
351	}
(gdb) 
main (argc=<optimized out>, argv=<optimized out>) at wpsmon.c:284
284	        printf("\n");
(gdb) 

240	    for(i=argc-1; i>0; i--)
(gdb) 
287	    ret_val = EXIT_SUCCESS;
(gdb) 
290	    globule_deinit();
(gdb) 
291	    if(bssid) free(bssid);
(gdb) 
292	    if(out_file) free(out_file);
(gdb) 
293	    if(wpsmon.fp) fclose(wpsmon.fp);
(gdb) 
295	}
(gdb) 
__libc_start_main (main=0x402500 <main>, argc=3, argv=0x7fffffffe4f8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffe4e8) at libc-start.c:321
321	libc-start.c: No such file or directory.
```

rofl0r, 2017-03-10:
ok for some reason pcap_next() fails. maybe one of the other applications interfere ?
i see this in one of your pastes:
```
Found 4 processes that could cause trouble.
If airodump-ng, aireplay-ng or airtun-ng stops working after
a short period of time, you may want to run 'airmon-ng check kill'

  PID Name
  837 avahi-daemon
  846 avahi-daemon
  868 NetworkManager
  992 wpa_supplicant
```
maybe you should kill those processes first.

kcdtv, 2017-03-13:
Could you copy paste the stdout of `dpkg -l | grep pcap`?
Thanks

notaz, 2017-06-12:
This happens on some systems when pcap_open_live() timeout is 0 (see src/init.c). I recommend adding a timeout there, at least 10000 for 10s, or so.

philipperemy, 2017-06-13:
Sorry for the lack of update. I gave up and ran successfully with Kali.
@notaz your answer seems great! Can we check if it was the case?

kcdtv, 2017-06-22:
notice that ubuntu 14.04 is a **very tricky version** as it is kind of  hybrid system with both  init/system.d.  
It also add issues with mode monitor and network manager interfering too much 

> Could you copy paste the stdout of dpkg -l | grep pcap?

Would be nice 😉  
i suggest you to update your Ubuntu system at least at 16.04 which has LTS. 
I can ensure you that it wil work with all 15.X, 16.X and latest 17.4  
Otherwise try to create an expetion for your inetrface in Network-Manager.conf file or use check kil to disable all conflictive process.

rofl0r, 2017-06-22:
@notaz can you open a PR with your proposal and a detailed commit message?

notaz, 2017-06-23:
Unfortunately it's not that easy, adding timeouts block on pcap_next() for that long, which sometimes timeout some communications.

philipperemy, 2017-06-24:
So maybe we could explicitly request an updated version of ubuntu in the documentation. Something newer than 14.04.

soxrok2212, 2017-09-13:
This issue was opened on an old version. If it still persists on 1.6.1, open a new bug report. Thank you.

closed 2017-09-13 by soxrok2212