t6x/reaver-wps-fork-t6x #130: [MacOS] Error with makefile

xKillerByte, 2017-03-10:
I ran ./configure in the reaverwps directory, the script said:  `config.status: creating Makefile`.
Then I ran make command and it printed the following error: 
`./iwlib.h:54:10: fatal error: 'linux/types.h' file not found
#include <linux/types.h>                /* for "caddr_t" et al          */
         ^
1 error generated.
make[1]: *** [iwlib.so] Error 1
make: *** [libiw] Error 2
`
Currently I am running MacOS.
Im looking forward to set it up.
Thank you.

soxrok2212, 2017-03-10:
Reaver is a Linux program. It is not designed to run on any other OS for varirous reasons.

closed 2017-03-10 by soxrok2212

rofl0r, 2017-03-10:
i don't know which source code you have, but i can't find that line in my copy of this fork's master branch.