t6x/reaver-wps-fork-t6x #131: Reaver '-a' option no longer available

hy-l, 2017-03-18:
I noticed that the '-a' option is removed recently,so i must type 'y' and enter to make it load the previous session each time after i run reaver.it's ok when run it in foreground,but it is a problem when run as a daemon with stdout redirected. Because all inputs are ignored and reaver will create a new configure file and overwrite the previous one.Maybe you guys could add that option again later?

kcdtv, 2017-03-19:
Yes it is not here anymore..  I don't remember when -who - why... Thanks for reporting.

rofl0r, 2017-03-20:
it was removed to be able to get rid of the sqlite dependency.

hy-l, 2017-03-20:
I don't know about programming,but -a let reaver auto load cfg files and use the proper advanced options,this really needs libsqlite3? Also I found the 'reaver.db' file in etc/reaver isn't used anymore,it contains ap info in sql database format, what is that file used for?
PS: Sorry for my poor English,i speak Chinese as usual.you may need bear with that.thanks... ;P

kcdtv, 2017-03-28:
> Also I found the 'reaver.db' file in etc/reaver isn't used anymore,it contains ap info in sql database format, what is that file used for?

Maybe it was just for this option? I am not a programmer neither... what is clear is that it has nothing to do with saved session (That what i thought the first time i saw this reaver.db file)

rofl0r, 2017-03-28:
i think we should add an option `-y` i.e. 'answer yes to all questions'. that part of the functionality was interconnected to some of the session loading stuff which i stripped off altogether to get rid of the 2 or 3 users of the sqlite db.
https://github.com/t6x/reaver-wps-fork-t6x/commit/64fc57960840e995a64b1cffc573336c86008b07

hy-l, 2017-03-29:
well, maybe you are right. but an option for auto detecting settings should be somehow useful, too. 
a few days ago, i did a test on a router with known pin code, but reaver ran into an endless loop and kept sending/receiving M# massages and didn't show the password. then i tried some other options,finally fixed it with -n. i just think, perhaps 'auto detection'  is needed at that point? How do you say?

rofl0r, 2017-03-29:
we're working on fixing those issues. when everything works as it should, there's no need for such hacks.

kcdtv, 2017-06-22:
> finally fixed it with -n.

That is now normally enabled by default...
... Would you mind to try again with 1.6b to check it? 
Like this you are our beta tester for this  :D 
Do not use -n, just send the PIN with -p option and tell us if it worked.

About a new flag "**-y**" it wouldn't be the same as former "**-a**" option. 
what it would do is answer "yes" to the question "A previous brute force session has been detected, do you want to use the former session? y/n" (something like that)  
It wouldn't launch the last command used.
adding it (a -y flag) is not bad idea, even-though in an automation perspective it is not necessary:
If you (or the script) launch reaver indicating the session with -s , it doesn't ask you to confirm as it is clear that you want to use the previous session.

hy-l, 2017-06-25:
okay, it worked. thank you. :)

kimocoder, 2017-06-25:
confirmed here too :)

hy-l, 2017-06-26:
alright, this issue should be closed now, I think. ;P

closed 2017-06-26 by hy-l