t6x/reaver-wps-fork-t6x #132: Add new feature for sending PIN as arbitrary string

binarymaster, 2017-03-28:
Some router firmwares contain "wrong" PIN codes, such as "222222" (some D-Link DIR-300NRU routers) or just an empty string "" (some Huawei routers), see this article:
https://hg658c.wordpress.com/2015/06/20/obtaining-the-wifi-password-in-a-few-seconds-using-wps/

Let reaver allow to send the PIN as arbitrary string with new command-line argument:
`reaver -i wlan0mon -c %channel% -b %bssid% -vv --pin-string=""`

and short variant:
`reaver -i wlan0mon -c %channel% -b %bssid% -vv -pS ""`

Reaver already have similar -X "exhaustive" option, but it does not allow to specify an arbitrary string.

kcdtv, 2017-03-28:
Hi! :smile_cat: 
We had a similar case in spain back in 2012 with ASL-26555 (PIN does not respect checksum rule) and there are some modified version of reaver and wpscrack around for that  
Bully allows this (that was introduced in the orignal version) .
About the null string... i have some  doubts and this is not realy clearly presented... and the patch with ` +            wps_registrar_add_pin(wps->wps->registrar, NULL, "lalalalalala" , 0 , 0);`
is effectively kind of dirty...  
Thanks you very much for your suggestion, we keep this ticket open to see/decide what to do.

binarymaster, 2017-03-28:
Thanks! Of course, that patch in the article is not clear enough to apply, and it is for 1.5.2 version only.

According to the actual source in master branch, it should be something like that:
```c
/src/pins.c

		if(get_arbitrary()) // check whether arbitrary string PIN argument is specified
		{
			/* Copy 8 characters from the given pin string value */
			snprintf(pin, pin_len, "%s", pin_str_value);
		}
		else
		{
			if(get_exhaustive())
			{
				/* Generate an 8-digit pin from the given key index values */
				snprintf(pin, pin_len, "%s%s", p1_value, p2_value);
			}
			else
			{
				/* Generate a 7-digit pin from the given key index values */
				snprintf(key, pin_len-1, "%s%s", p1_value, p2_value);

				/* Generate and append the pin checksum digit */
				snprintf(pin, pin_len, "%s%d", key, wps_pin_checksum(atoi(key)));
			}
		}
```

binarymaster, 2017-03-28:
Another variant:

```c
/src/pins.c

 		/* Add the new pin */
		int add_result;
		if(get_arbitrary()) // check whether arbitrary string PIN argument is specified
		{
			add_result = wps_registrar_add_pin(wps->wps->registrar, NULL, (const u8 *) pin_string, strlen(pin_string), 0);
		}
		else
		{
			add_result = wps_registrar_add_pin(wps->wps->registrar, NULL, (const u8 *) pin, PIN_SIZE, 0);
		}
		if(add_result != 0)
		{
			free(pin);
			pin = NULL;
		}
```

soxrok2212, 2017-03-28:
Not quite sure I understand the problem being referenced, the pin in the GUI does not follow the checksum? If this is the case, then the device isn't WiFi Certified OR the user manually set a custom pin.

Regardless, let's see what @rofl0r has to say before we merge.

binarymaster, 2017-03-28:
> the pin in the GUI does not follow the checksum?

According to the article, some devices can work in PBC mode and accept the empty string pin, without digits and checksum, and without pressing the WPS button, which is important.

rofl0r, 2017-03-28:
i commented in the PR.

kcdtv, 2017-03-29:
> If this is the case, then the device isn't WiFi Certified OR the user manually set a custom pin.

Or the Manufaturer (or the ISP)  did not respect the recommendation or were to lazy to codify the checksum rule... That what happened with Amper asl25666 in spain.
The trick Ii was doing in this cases (brute force on 4 digits in second half) was to modify line 74 from **defs.h** from reaver 1.4 and replace `#define P2_SIZE                 1000` by `#define P2_SIZE                 10000`
Than edit the array from key.c to have all the second half with four digits. 
I know, i know, this is crap... but maybe it helps. 

> According to the article, some devices can work in PBC mode and accept the empty string pin, without digits and checksum, and without pressing the WPS button, which is important.

  Very few  ... Yes, that could happen in a kind of "grey configuration state" with WPS enabled in PIN mode without PIN configured and some chipset-devices. I don't discard it. A bit buggy.  
According to what we see the PIN given back is 12345670 which is the "fall back" PIN in case of misconfiguration   
 Just to say it not linked directly to PBC. 
WPS is enabled "generally" (WPS state in the tagg) but there is no PIN configured for the PIN mode. 
The interface shown in the next images  is very similar to this one (from Comtrend and a spanish ISP)
 1- disabled:
[image]
 2 - enabled
[image]

  Very crappy and limited WPS interface... you cannot change the PIN (in the case above there is a PIN configured and it is breakable), you cannot disable the PIN mode without disabling WPS globally. .
  So the trick could eventually work on this kind of device (with a very bad firmware... this is not a good device) in a state where PIN is not defined. 
  I wish the document would be a bit more "consistent".

binarymaster, 2017-06-25:
Implemented in https://github.com/t6x/reaver-wps-fork-t6x/compare/40862b853ec85afa7087a103079a6ee2fc6542b2...493f72caed77b6badc045f1a0396747b382964e2

closed 2017-06-25 by binarymaster

kimocoder, 2017-06-25:
Great work sir, actually all off you are awesome 👍

slmafiq, 2017-12-29:
How i can start new version on reaver 1.6.3 in my bcmon samsung galaxy s 2? I want to crack Zte with -p "" options.  Thanks!