t6x/reaver-wps-fork-t6x #135: Compile for LEDE?

mile18, 2017-04-01:
Hi there,
Could you please compile version for LEDE, becouse ar71xx doesn't work anymore becouse it's changed to mips_24k.
I've tried and it's not working properly. Thank you in advance.

kcdtv, 2017-04-01:
I don't understand your question: What does it have to do with reaver if your chipset isn't supported by your firmware? ...  This is not clear, please be more specific

> I've tried and it's not working properly. 

That doesn't help neither... please be more specific and add terminal stdout to show what "isn't working properly"

mile18, 2017-04-02:
OpenWrt compile package with ar71xx tag, LEDE compiles it as mips_24k, so when you are trying to install OpenWrt package to LEDE, an error will accur saying wrong package arhitecture. Compiling is sucesfully done, but when reaver is started and tryes to get the pin with pixiewps, trace/breakpoint trap error ends process. On OpenWrt, same hardware it works with pre-compiled ipk. You can see error here: http://imgur.com/a/GWx1F

rofl0r, 2017-04-03:
probably they use a different floating point convention. one may use softfloat, the other not oslt. anyway, this is a toolchain issue and nothing we can help you with. you can try to make your own correct toolchain[1] and build it from source, but we don't do binary distributions here.
you should report your issue to LEDE/openWRT.

[1] https://github.com/GregorR/musl-cross

closed 2017-04-04 by kcdtv

kcdtv, 2017-04-04:
> we don't do binary distributions here.

:+1: 
Hey r0f! ...  Cya later today if u can to speak a little? I'll be around in the afternoon-evening.