t6x/reaver-wps-fork-t6x #136: Feature request: change MAC

Lexus89, 2017-05-11:
Excuse me if i am wrong but as far as i know this reaver does not have a mac changer option such as https://github.com/gabrielrcouto/reaver-wps It would be really nice to have this functionality in this fork as well.

peterpt, 2017-05-30:
you can use macchanger to do that .
- create your monitor mode
- put it off
- use macchanger to change real mac address
- enable again the monitor mode

kcdtv, 2017-06-22:
There are some issues with machanging automation... for example: If the user is connected with another wifi interface,  network manager won't let the mac change, you will change it but if you check with airodump-ng you will see that the real mac is used (even of network manager doesn't use the interface that you dedicate to montior mode)  
We can try to see that but if there are many checks to do and basic services to disable, better let the user do it manually. 
Let's say we have to totally shut down network service and/or network manager to have a real and effective macchange: Some (i would say many) users won't understand why they cannot connect again after stopping using reaver or won't know  how to restart the disables services 
If they do it by hand, they'll know what they have done. 
I think the interface management (enabling monitor mode, changing mac, stopping or not network manager) shouldn't be done by reaver: They are tools for that and not all users do it the same way depending on the context.

soxrok2212, 2017-09-13:
The general practice for using this tool is to test your own network, so it should not matter what MAC address you are using.

closed 2017-09-13 by soxrok2212