t6x/reaver-wps-fork-t6x #141: WARNING: Failed to associate with 00:18:E7:FD:F6:F6 (ESSID: internet1832)

CarlUpright, 2017-06-19:
I got the latest Kali Linux, and i got 2 wireless card, one internal and a usb TL-WN722N (ver. 1.10) (thats the one im using). Laptop is a Lenovo W520.```
root@kali:~# reaver -i wlan1mon -b 00:18:E7:FD:F6:F6 -vvv -S

Reaver v1.5.3 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
mod by t6_x <t6_x@hotmail.com> & DataHead & Soxrok2212 & Wiire & AAnarchYY & KokoSoft

[+] Waiting for beacon from 00:18:E7:FD:F6:F6
[+] Switching wlan1mon to channel 11
[!] WARNING: Failed to associate with 00:18:E7:FD:F6:F6 (ESSID: internet1832)
[!] WARNING: Failed to associate with 00:18:E7:FD:F6:F6 (ESSID: internet1832)
[!] WARNING: Failed to associate with 00:18:E7:FD:F6:F6 (ESSID: internet1832)
[!] WARNING: Failed to associate with 00:18:E7:FD:F6:F6 (ESSID: internet1832)
[!] WARNING: Failed to associate with 00:18:E7:FD:F6:F6 (ESSID: internet1832)
[!] WARNING: Failed to associate with 00:18:E7:FD:F6:F6 (ESSID: internet1832)
^C
[+] Nothing done, nothing to save.

```
i tried a few options, and i always get the same message. I did run the good old backtrack, and it works, with the same laptop on same bssid. 

Thats my network. im sitting next to the router.

kimocoder, 2017-06-19:
Well i get that too sometimes.. But that is pretty  rarely. I got lots of adapters, TL-WN722n too & Kali of course. Try setting channel manually (by adding - c <channel>), also, change "-S" with "-K 1" (and up from 1 to 4) (using the pixiedust attack) and report back. I may try to reproduce if you still fail.

CarlUpright, 2017-06-19:
I got the same result using -c and -K (1 to 4).

kimocoder, 2017-06-19:
Ok, i get that one (very) few kinds of routers too, but I did not investigate the problem further because it' rarely happens, but please feel free to share your findings in the future. May i ask which kind of router this is? Exact (to match chipset) please and I'll add it to my notes.

CarlUpright, 2017-06-19:
Router is D-Link DIR-628. 
but that happened to me before on my other laptop when i tried Kali 2016.1. I was using the internal card and that was a different router. I simply went back to Backtrack and everything was working fine.

I still got that older laptop, ill try more stuff tomorrow, but i might just downgrade reaver to the last working version. 
[image]

rofl0r, 2017-06-19:
can you try if this branch here works ? https://github.com/rofl0r/reaver-wps-fork-t6x/tree/clean_plus_pixie that's the cleaned up version i work on which is based on last official reaver with a lot of improvements backported, but hopefully not the bugs :)

CarlUpright, 2017-06-19:
Yes rofl0r, i can do that. 

I'll make a new kali 2017.1 usb drive in the next hour and boot from it to not mix my stuff to much. Should i install your branch right on top or uninstall some stuff first?

I'm pretty sure Kali is the problem or that some background process used by reaver should be downgraded. Reaver works great in backtrack and in xiaopan 0.4.7.2 on the same laptop.

Some people on another forum said they had to downgrade to "libpcap0.81.4.02 and reaver 1.4-2" to make it work in kali. I'm fairly new to linux (mostly following "how to" ) and i don't know how to downgrade stuff.

rofl0r, 2017-06-19:
you dont need to uninstall anything, you can just clone my repo and then `git checkout clean_plus_pixie` to switch branches, then `cd src ; ./configure ; make` and then run reaver/wash from inside there (`./reaver`, instead of `reaver`, the latter uses the version you have installed)

CarlUpright, 2017-06-19:
```
root@kali:~# git clone https://github.com/rofl0r/reaver-wps-fork-t6x.git
Cloning into 'reaver-wps-fork-t6x'...
remote: Counting objects: 1659, done.
remote: Total 1659 (delta 0), reused 0 (delta 0), pack-reused 1659
Receiving objects: 100% (1659/1659), 1.66 MiB | 1.25 MiB/s, done.
Resolving deltas: 100% (1018/1018), done.
root@kali:~# git checkout clean_plus_pixie
fatal: Not a git repository (or any of the parent directories): .git
```
sorry, i'm still learning and i don't know how to do that

rofl0r, 2017-06-20:
`cd reaver-wps-fork-t6x` before git checkout :)

kcdtv, 2017-06-20:
To determine if Reaver as a specific association problem it makes sense to try first  with -A argument (do not associate) and launch in parallel an association with aireplay-ng (-1) 
  I can associate to my PA with reaver and this chipset in Kali
Association itself is not a problem but the chipset was working much better in the distributions a cople of year ago, it is kind of dirty right now..  But it can associate... But it is bearly working, far from being  as good as it did in the past. 
it looks like a lack of control when sending packet and can comes from change in mailine drivers, 
with reaver as it is now
```
`[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
```
with reaver "cleaned+pixie".... the same (association works but very hard to lanch a single PIN
```
kcdtv@kalimuX0:~/Escritorio/REAVER/clean_plus_pixie/reaver-wps-fork-t6x/src$ sudo ./reaver  -i wlan2mon -b 18:17:25:31:F8:74   -vvv -c 1 

Reaver v1.4 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Switching wlan2mon to channel 1
[+] Waiting for beacon from 18:17:25:31:F8:74
[+] Associated with 18:17:25:31:F8:74 (ESSID: ONO31F874)
[+] Trying pin 12345670
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received M1 message
[+] Sending WSC NACK
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x03), re-trying last pin
[+] Trying pin 12345670
```
we'll first put on trail the new version than will try to do something about it although i have a very strong feeling that there is a kind of "mode monitor regression"  in ath9k_htc kernel drivers as i observed latly very strange behaviours in several distribution. .

closed 2017-09-13 by soxrok2212

UJJWAL-GARG, 2017-12-18:
Failed to associate in reaver is because of three main problems :----
1. Wi-Fi adapter is not able to hack into access point. (sometimes a monitor mode adapter also have this problem) so try with another adapter..
2. Wi-Fi Network not vulnerable (sometimes an AP is not vulnerable i.e like "sony bravia" does not have wps vulnerability)
3. Your version of reaver or another tool (try with other versions i.e old and new both)

rofl0r, 2017-12-18:
@UJJWAL-GARG your response is not really helpful.
you list 3 valid reasons why reaver could fail in general, but its not related to "failed to associate". that said, i'm locking this topic to not further confuse the OP.