t6x/reaver-wps-fork-t6x #145: compile on non-linux currently broken

rofl0r, 2017-06-23:
commit e671b44cbd314431bd706de57d8a6b5b871fbb60 fixed big endian issues, but does so using a linux specific header `<asm/byteorder.h>`.
this breaks build on non-linux platform.
mac os X is not supported anyway due to injection problems on that platform, but it would be neat if it could be used on freebsd too.
this is very low priority though.
so if someone from the freebsd community is interested in getting this fixed, we would accept patches. the approach i have in mind is:

- provide a header modeled after byteorder.h, which provides the endian conversion macros with slightly different names (for example without `__` prefix, which is reserved).
also equivalent int types like __le32. the header can internally use gcc's `__BYTE_ORDER__` macro to determine the right endianness. my approach which works well so far is the following:

```c
#ifdef __linux
#include <endian.h>
#else
#include <machine/endian.h>
#endif

#if (!defined(__BYTE_ORDER)) && (!defined(BYTE_ORDER))
#error "no __BYTE_ORDER macro found!"
#endif

#ifndef __BYTE_ORDER
#define __BYTE_ORDER BYTE_ORDER
#endif

#if (!defined(__LITTLE_ENDIAN)) && (!defined(LITTLE_ENDIAN))
#error "no __LITTLE_ENDIAN macro found!"
#endif

#ifndef __LITTLE_ENDIAN
#define __LITTLE_ENDIAN LITTLE_ENDIAN
#define __BIG_ENDIAN BIG_ENDIAN
#endif

```
it could be improved by testing for gcc `__BYTE_ORDER__` macro before doing any includes, because all modern gcc's (and probably clang too) define it now.

- this header will then be used from inside the tree unconditionally, instead of using linux' byteorder.h.

DJDan, 2017-10-01:
Let me know when its fixed. Thanks

Martinfx, 2017-11-13:
For endian on  freebsd you could 
`#include <sys/endian.h>`

Martinfx, 2017-11-13:
For freebsd you could use include for endians, and for standart types  sys/types.h
```
#include <sys/types.h>
#include <sys/endian.h>
```

Why do you use non  standart types __be and __le ?

rofl0r, 2017-11-13:
ask the openwrt guy that did the change. since I knew that openwrt was using that patch and it was working (i.e. tested), i just merged theirs instead of doing my own, and i dont even have big endian hardware to test.

rofl0r, 2017-11-13:
also see my project rofl0r/endianess.h where i'm in the process of adressing this issue.

closed 2017-11-26 by rofl0r

rofl0r, 2017-12-02:
@DJDan can you test on mac and open a new issue with detailed build logs if the compilation on mac still fails ? thanks!

Martinfx, 2017-12-02:
@rofl0r  Do you will make new release ?

rofl0r, 2017-12-03:
@Martinfx first i need you to test if it actually works
we had it compiling on mac since a while, but there's a ton of issues i'm addressing atm
and i expect it will be similar on freebsd

Martinfx, 2017-12-03:
@rofl0r On FreeBSD program run but have only [+] Waiting for beacon .... maybe i have bad AP.

rofl0r, 2017-12-03:
did you put your adapter into monitor mode ?

Martinfx, 2017-12-03:
@rofl0r  Yes, but i will tomorrow again, because i had maybe bad parameters.

commit 664a4b32 (2017-11-26): fix unportable use of asm/byteorder.h

closes #145