t6x/reaver-wps-fork-t6x #147: POLL: location of reaver's wpc directory (/etc/reaver)

rofl0r, 2017-06-27:
i found it weird since i first used reaver that it would install its reaver.db and when used, write .wpc files into /etc/reaver... since /etc is meant to be a directory for static configuration data that's not supposed to change, and many embedded distros even mount it on read-only storage.
imo /var/lib/reaver is a much better suited place ([and i actually use reaver this way since almost day one](https://github.com/sabotage-linux/sabotage/commit/f9e384b2673286446c2a142c072a72389a1cc541)).

according to https://en.wikipedia.org/wiki/Filesystem_Hierarchy_Standard :

> /etc
> Host-specific system-wide configuration files

> /var
> Variable files—files whose content is expected to continually change during normal operation of the system—such as logs, spool files, and temporary e-mail files.

feel free to discuss your opinion.

soxrok2212, 2017-06-27:
Move to /var!

kimocoder, 2017-06-27:
Agreed with @soxrok2212 -- go for /var

kcdtv, 2017-06-28:
I don't care... As long as there is a dedicated directory. :smiley_cat:

closed 2017-06-29 by rofl0r

Techwolf, 2017-10-04:
I just upgraded to 1.6.2 from 1.5.2. Turns out that on a gentoo system, /var/lib/lib/reaver is the path used. Note the double /lib path there.

rofl0r, 2017-10-05:
what's in the ebuild ?

commit 9b4a9622 (2017-06-29): change installation directory of .wpc storage to /var/lib/reaver

it was previously stored to /etc/reaver, which didnt really make sense.

closes #147