t6x/reaver-wps-fork-t6x #149: Wash doesn't detect some PA in 5Ghz frequencies

kcdtv, 2017-06-30:
The bug is present since the first version of wash and has been around since 2012.
You can see in the following picture that **wash** is not able to see my PA even if i am two meters aways from my routeur.

[image]

Funny thing is that there is/was a bug with airodump-ng and airepaly-ng in 5Ghz frequencies on my PA. 
And **jpmv27 ** found a solution:

>  I am running into a problem where "aireplay-ng -9" is not seeing some APs in 5 GHz band.
> These APs are visible when running airodump-ng.
> 
> Looking at the code of grab_essid, the first thing it does is look for a "**DSSS Parameter Set**" IE (IE ID, AKA tagtype, 3), and bails if it doesn't find it. So, any AP that doesn't include it in Beacon or Probe Response will be ignored.
> 
> However, by my reading of the standard, **that IE is not required when using OFDM in the 5 GHz band**. So this seems like a bug.

from: [https://github.com/aircrack-ng/aircrack-ng/issues/120](url)
I am pretty sure that wash does the same mistake. 

I forgot: there is also a little issue when scanning 5Ghz frequencies (wash -5): It always parses channel 1 (2.4 Ghz). That's not abig deal, it is more problematic to do not see the PA that can be two meter away from you.
Notice that i can attack my PA and get the WPA key with reaver without problems despite the fact that wash does not see it.

rofl0r, 2017-06-30:
maybe it's related to EXT bit handling in radiotap header too. if so, it only happens with the drivers affected by the "0 dbm" issue.

kcdtv, 2017-06-30:
There might be a little something more... 
The bug is only in 5Ghz (my PA is dual band and i can perfectly see the 2.4Ghz but not the 5Ghz) 
But yes, definitely,  this chipsets did have the "0dBm" bug with wash before Notaz committed his fixes.
And airodump-ng was giving -1dBm with my PA in 5Ghz band two release ago. 
That was fixed in latest aircrack-ng revision.

rofl0r, 2017-10-21:
can you try if it's fixed with db6cc396682b44e3533220278b69aa36c9906d47 ?
that commit either broke 5 GHz entirely ;) or fixed it too...

kimocoder, 2017-10-21:
Is it time for tagging a new release soon? You've been active lately I see:) great work

rofl0r, 2017-11-03:
fixed by https://github.com/t6x/reaver-wps-fork-t6x/commit/d14981de4fc37a3b8ecd08c6cc0ca9753b85a753

closed 2017-11-03 by rofl0r