t6x/reaver-wps-fork-t6x #151: No -K option on install.

xxRockOnxx, 2017-07-05:
On my other machine the `-K` option exists and is even in the example but on my Ubuntu 17.04 it doesn't have that.

Tried removing directory and cloning again but no luck. Why is this? It's like the regular `reaver` only

kcdtv, 2017-07-05:
If you check the  program description in first page it is explained:
The feature is temporary removed in order to code it "cleanly" (not as it was)
To do a pixie wps attack you can use `-Z` or  `-K`
Once the PIN is found with opixedust you just have to send it with the option -p. 
Cheers

xxRockOnxx, 2017-07-06:
Uhm, again? Not clear. I was asking `-K` was not present but you are suggesting

> To do a pixie wps attack you can use `-Z` or `-K`

soxrok2212, 2017-07-06:
You probably installed reaver from the Ubuntu repo... likely is Reaver 1.4. Install from source here and try again.

kcdtv, 2017-07-06:
>  I was asking -K was not present but you are suggesting

I answered you and I told you that it is explained in first page: 
![seleccion_706](https://user-images.githubusercontent.com/7338484/27931449-96173686-629a-11e7-9be6-6d1205ea4044.png

rofl0r, 2017-07-06:
i suspect he runs "reaver" without arguments and doesn't see "-K" in the help text/usage output, since it is indeed missing! i forgot to add it when i implemented the pixiedust attack in f78f896ad9135d3795f5e03aa555002503b0c85c.
proposed fix:
```diff
diff --git a/src/wpscrack.c b/src/wpscrack.c
index e21b19a..b228c45 100644
--- a/src/wpscrack.c
+++ b/src/wpscrack.c
@@ -174,6 +174,8 @@ int usage(char *prog_name)
         fprintf(stderr, "\t-E, --eap-terminate             Terminate each WPS session with an EAP FAIL packet\n");
         fprintf(stderr, "\t-n, --nack                      Target AP always sends a NACK [Auto]\n");
        fprintf(stderr, "\t-w, --win7                      Mimic a Windows 7 registrar [False]\n");
+       fprintf(stderr, "\t-K, --pixie-dust                run pixiedust attack\n");
+       fprintf(stderr, "\t-Z,                             run pixiedust attack\n");
 
         fprintf(stderr, "\nExample:\n\t%s -i mon0 -b 00:90:4C:C1:AC:21 -vv\n\n", prog_name);

```

xxRockOnxx, 2017-07-06:
@rofl0r so it's there but not just in the help output?

@soxrok2212 I didn't install from the repo. I cloned and and installed this one.

rofl0r, 2017-07-06:
@xxRockOnxx yes. is that what you meant when you talked about that it's missing?

xxRockOnxx, 2017-07-06:
Yes. Thank you for clearing it up. Should close this :)

closed 2017-07-06 by xxRockOnxx