t6x/reaver-wps-fork-t6x #154: Raspberry pi 3, failed to associate

anoo2niem, 2017-07-10:
Hello, 

Im running this fork on my laptop with kali installed and everything work flawless with the awus036nha adapter.

I recently bought an raspberry pi 3 model b, i installed kali linux on it & this fork with all its dependecies.
When i run reaver with the same adapter i use on my laptop, it constantly get the following output:
`[!] WARNING: Failed to associate with XX:XX:XX:XX:XX:XX`
I've done some research and came along that it could be an issue with newer libpcap versions.
with libpcap0.8-dev_1.4.0-2 it should work fine, so i looked for 64bit versions.
I only found a amd64 package but the package architecture does not match my system (armfh).

Any advice to make i work on my architecture? 

Thanks in advance!

rofl0r, 2017-07-10:
i doubt the pcap version is the problem, but you can try to compile it from source.

anoo2niem, 2017-07-11:
you're right, i did it and the problem remains..

kcdtv, 2017-07-11:
There are some issues with ath9k_htc, we spoke about it here [https://github.com/t6x/reaver-wps-fork-t6x/issues/141](url)

PoSHMagiC0de, 2017-07-11:
The Raspberry Pi 3 is not the only one having issues.  My ParrotOS just updated to 3.7 which runs on Debian 10 I believe.  I am getting the same issue you are getting now.  Before I was fine.  Of course during that update I do not know if reaver was updated also to 1.6b.
I tried the 1.6.1 pulled from here but get the exact same issues.  Bunch of Received identity request, Sending identity request followed by maybe some M1 and M2s, WSC NACK and ten failed code of 0x03 sometimes random failed to associate and timeouts.  Was working great before a few days ago.  Using an Alfa card.

anoo2niem, 2017-07-11:
Exactly as u say, i did the association with aireplay-ng and added -A in the reaver command, toghetter with --no-nacks. A bunch of sending/received identity requests, some m1 and m2, WSC NACK and then always 0x03. But thats as far as it gets. Did u try it with bully? Because i got the same with bully on raspberry pi 3 (kali). O i doubt its a problem with reaver since i got issues with bully also... Thats why i thought its an issue with a library like libpcap, but i tested it with the same libpcap i have on my laptop with same version of reaver, still with same output...

PoSHMagiC0de, 2017-07-11:
I tried it with Bully and am seeing the same thing, well not the exact same messages because the apps are different but the error type seems to be the same.  Timeouts, no associations sometimes, etc.

anoo2niem, 2017-07-11:
Exactly, it must be an issue with newer debian distributions. I,ve tried it with libpcap0.8-dev_1.4.0-2 as mentioned in some fora. Without luck still same output. I've also read that u had to put yr wlan interface in monitor mode before running reaver on newer kali distributions, but for me that also didn't work.

anoo2niem, 2017-07-11:
I've didn't saw the comment of kcdtv, i guess indeed there must be an issue with the ath9k_htc interfaces on newer debian distributions. i remember i did apt-get install atheros-firmware. Would downgrading this firmware work?

rofl0r, 2017-07-12:
@PoSHMagiC0de 

> Was working great before a few days ago. Using an Alfa card.

what was changed on your system during the time when it worked and when it didn't work anymore ?

PoSHMagiC0de, 2017-07-13:
@rofl0r 

> what was changed on your system during the time when it worked and when it didn't work anymore ?

Linux distro I am running (ParrotOS) did a major update from 3.6 to 3.7 which, I believe, moved my base distro from debian 9 to 10.

Recently the ParrotOS repo added the 1.6.1 reaver update a day after its 3.7 update so tried it again hoping it was a fix (even though I already tried to pull and build from git the 1.6.1 version with same results) and got same issue.

rofl0r, 2017-07-13:
@PoSHMagiC0de so if i understand correctly, 2 things (of importance) changed:

1) the version of your kernel (and so the wifi drivers)
2) the version of reaver

to make certain there's no regression in reaver, could you try to if using reaver 1.4 makes any difference ? thanks! btw, it might help to set the m57 timeout to a higher value (defaults to 0.2sec, `-T 3` could work considerably better)

soxrok2212, 2017-07-13:
Also note that it was Debian 9 that was just released, not Debian 10. It very well could be a problem with the drivers. Previously on kernel 3.16? I experienced less issues than I did on 4.4 with my 036H. Reaver still works on 4.4 for me, however.

PoSHMagiC0de, 2017-07-13:
Sorry, wrong debian.  My kernel is reading 4.11 now.  Cannot remember previous kernel.
Wireless card I am trying this on is an AR9271 in an Alfa.  Driver it is currently using is "ath9k_htc".
Above it is mentioned about the new version of that driver having issues.
Reaver I was on before was 1.6b and it worked.  After Parrot upgrade I still was on 1.6b and began having issues.  Built and tried the 1.6.1 from the git repo but still same issue.  Didn't install, just ran from source.  A day later 1.6.1 was in Parrot's repo and updated but issue is still present.
Tried T3 but still same issue.  It is acting like the access point is too far away when I am right next to it.  Soon it will get stuck in the EAPOL loop and proceed no further.

PoSHMagiC0de, 2017-07-14:
Just to add, my issue is not related to my Pi but laptops and desktops.  My USB stick that was updated to newest Parrot around same time has same issue and same version Kernel.
Now, My Pi2 works with the -T 3 option.  Difference is I have not updated that one yet.  It is running on Kernel 4.4.9.  So, might be something that is happening in the newest updates on some OSes since even some are reporting it after updating Kali.

rofl0r, 2017-07-15:
at least, kernel 4.4.x is a long-time-support kernel, so it should be possible in some distributions to select it. and you should probably stick to it as long as you can... unless we can find out what caused the issue. that's not impossible to do, at first one could try which major kernel version introduced it (4.8 ? 4.9 ? ...) and then one can start to git bisect the changes between the last known good and the first known bad commit; or maybe that information alone is already sufficient to identify the culprit by looking at the commits, in case there weren't many. but yeah, someone who knows how to compile his own kernel needs to stand up and do it.

PoSHMagiC0de, 2017-07-15:
Some good news.  I was able to get reaver to operate on the current kernel I am on which is 4.11.  I didn't try this before because I seen a previous post say they tried it with no success.
I pretty much with the -T 3 option added in the -A option and then used aireplay fakeauth as the authenticator.  I had to do this before one time a year or so ago on Kali for a period until the issue just cleared up after an apt upgrade.  Anyway, I can reave now if I use aireplay as the authenticator.

On a side note: Did aireplay fakeauth always have the ability to detect disassociations and reassociate before the delay time?  I never seen that before but been awhile since I used that option.  Pretty cool.  I thought the constant short looping of associations would suck but with that option and a high delay time it is not that bad,

soxrok2212, 2017-09-13:
The Reaver session should stop working if you are deauthenticated. The drivers on kernel 4 suck, I've had nothing but problems. Fortunately, this is not a problem with Reaver, but you will either need to find a different driver to run or open a bug to get the fixed on the new kernels.

closed 2017-09-13 by soxrok2212