t6x/reaver-wps-fork-t6x #157: How to fix Wash Error?

Deathlive, 2017-08-10:
after ./configure , Wrote make and.
https://prnt.sc/g6rrrc

How to fix?

rofl0r, 2017-08-10:
that's not a checkout from git since we dont use libsqlite3 anymore.
try current master or the latest release https://github.com/t6x/reaver-wps-fork-t6x/releases

Deathlive, 2017-08-11:
I install new release, but the error did not disappear.
http://prntscr.com/g73y8o

rofl0r, 2017-08-11:
for the record, his builds fails because the linker can't find the ualarm function.

which OS / libc are you using ? ualarm is part of GLIBC and musl libc.

Deathlive, 2017-08-11:
@rofl0r i use linux emulator → Termux on Android.
In my, musl is built into Android?

rofl0r, 2017-08-11:
no, android uses a libc called bionic. dunno if we want to support that, but for the moment you can try to put this at the very beginning of src/wpsmon.c:

```
#define _GNU_SOURCE
#include <unistd.h>
#include <sys/time.h>

unsigned ualarm(unsigned value, unsigned interval)
{
        struct itimerval it = {
                .it_interval.tv_usec = interval,
                .it_value.tv_usec = value
        };
        setitimer(ITIMER_REAL, &it, &it);
        return it.it_value.tv_sec*1000000 + it.it_value.tv_usec;
}

```
that's musl's implementation of ualarm(), if it compiles you should be good.

Deathlive, 2017-08-12:
@rofl0r Thanks, has turned out.

closed 2017-08-12 by Deathlive