t6x/reaver-wps-fork-t6x #159: Can't compile on gentoo from Pentoo overlay

junked, 2017-08-16:
https://pastebin.com/feVW1SY6

kcdtv, 2017-08-17:
You are not compiling anything,,,   
Read the 1st FAQ: it will solve your issue (local repository actualization) 
 [https://github.com/t6x/reaver-wps-fork-t6x/wiki/FAQ](url)

junked, 2017-08-18:
Need to modify ebuild with this steps? Cause i can't manually do this in portage distfiles directory.

kcdtv, 2017-08-18:
I do not really understand your questions... 😺 
But i can see your problem: 
` ERROR: net-wireless/reaver-wps-fork-t6x-1.5.2_p20161210-r2::pentoo failed (unpack phase):`
 The answer is to use **this** repository in order to have **reaver 1.6.1.**  which the latest stable release
 We have nothing to do with packets inside other repositories; that's a matter you should consult with the maintainers from this overlay.

dmio, 2017-09-11:
junked, some refs/sha's were wiped out. You can see this in your error message. So, old ebuilds are not usable anymore.
You can use this ebuild.
https://github.com/dmio/gentoo-overlay/blob/master/net-wireless/reaver-wps-fork-t6x/reaver-wps-fork-t6x-1.6.1.ebuild

closed 2017-09-26 by kcdtv

Techwolf, 2017-10-09:
Just some info for those that find this issue via a search. I do have a working ebuild for gentoo users on my gentoo overlay. https://github.com/Techwolf/techwolf-overlay