t6x/reaver-wps-fork-t6x #160: libcap not found

David90000, 2017-08-17:
Hello, i can't install reaver because of pcap.

root@David:~/reaver-wps-fork-t6x/src# ./configure
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... myes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for pcap_open_live in -lpcap... no
error: pcap library not found!

i try this : root@David:~# sudo apt-get install libsqlite3-0 libpcap0.8
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances
Lecture des informations d'état... Fait
libsqlite3-0 is already the newest version (3.14.1-1).
libpcap0.8 is already the newest version (1.7.4-2kali1).
0 mis à jour, 0 nouvellement installés, 0 à enlever et 0 non mis à jour.

but nothing change. Any idea ?

rofl0r, 2017-08-17:
you need the devel versions of these packages i.e. libpcap-dev

closed 2017-08-17 by rofl0r

David90000, 2017-08-17:
how to do ? :-D 

Thanks you for helping.

rofl0r, 2017-08-17:
`sudo apt-get install libpcap-dev` i'd guess

David90000, 2017-08-17:
packet not found :-(

rofl0r, 2017-08-17:
well i don't use debian/ubuntu so i can't tell you exactly. maybe try `aptitude search libpcap` so it shows you a list of possible package names. you want one that ends in -dev or -devel

rofl0r, 2017-08-17:
btw the README mentions how to install on debian-based systems, and it looks like package should indeed be `libpcap-dev`. maybe you made a typo and forgot the first "p", like in your issue title...

David90000, 2017-08-17:
i already try apt-get -y install build-essential libpcap-dev, but can't find the packet.

kcdtv, 2017-08-17:
The packet does exist and it is called **libpcap-dev** 
```
kcdtv:~$ apt search libcap
Ordenando... Hecho
Buscar en todo el texto... Hecho
libcap-dev/now 1:2.25-1 amd64 [instalado, local]
  POSIX 1003.1e capabilities (development)
```
 Your Kali system is not updated and maintained correctly.You are running kali 2016.1 with outdated libraries.First update your system correctly (kernel included).

David90000, 2017-08-17:
update in progress, thank you.