t6x/reaver-wps-fork-t6x #161: Request: Update Big_endian branch to version 1.6b

adde88, 2017-08-18:
Hi there. and thanks for the awesome work! Especially the latest version bump!
It would be awesome if the Big_endian branch could be updated to version 1.6b.
So that we could compile the newest Reaver for OpenWRT. 👍

kimocoder, 2017-08-18:
@kcdtv what do you say? is it on the future todo list? im home from holidays & may take a look at it too :)

takk for forespørselen, jeg tar en kikk på det nå i helga :)

adde88, 2017-08-18:
Thank you very much for such a quick response!
If there's anything i can do to help, let me know :)

PineNappler, 2017-08-18:
This would very useful for me and others i know. Would be great to have! 👍

kimocoder, 2017-08-18:
well, feel free to start porting code from 1.6b repo to the big_indian one if the knowledge of howto is in place. dunno yet if it needs modifications, we'll see

rofl0r, 2017-08-18:
huh ? why would you want that ? you can just use the latest release and it'll work on openwrt with big and little endian.

adde88, 2017-08-18:
@rofl0r 
Attempting to compile the latest commit with the Openwrt-SDK gives the following error-messages:
```
{standard input}: Assembler messages:
{standard input}:20: Error: unrecognized opcode `wsbh $2,$4'
{standard input}:780: Error: unrecognized opcode `wsbh $2,$2'
{standard input}:781: Error: unrecognized opcode `rotr $2,$2,16'
{standard input}:806: Error: unrecognized opcode `wsbh $4,$4'
{standard input}:807: Error: unrecognized opcode `rotr $4,$4,16'
{standard input}:889: Error: unrecognized opcode `wsbh $3,$2'
{standard input}:890: Error: unrecognized opcode `rotr $3,$3,16'
<builtin>: recipe for target '80211.o' failed
```

A full error-message can be found here:
https://pastebin.com/5P94AsVE

adde88, 2017-08-18:
This can now be closed, i managed to get it compiling. :)

closed 2017-08-18 by adde88