t6x/reaver-wps-fork-t6x #163: Reaver installation issue !!

Ghostfro, 2017-08-30:
Hi guys,
i just downloaded reaver and got a problem while installing it on the command " make " after ./configure
this is what i got : 

root@kali:~/reaver-wps-fork-t6x-master/src# make
(cd lwe && make BUILD_STATIC=y libiw.a)
make[1]: Entering directory '/root/reaver-wps-fork-t6x-master/src/lwe'
make[1]: 'libiw.a' is up to date.
make[1]: Leaving directory '/root/reaver-wps-fork-t6x-master/src/lwe'
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -Wall -g -O2 -Wno-unused-function -Ilibwps -I. wpsmon.c globule.o init.o misc.o 80211.o iface.o builder.o crc.o libwps/libwps.o wps/wps_attr_build.o wps/wps_attr_parse.o wps/wps_attr_process.o wps/wps.o wps/wps_common.o wps/wps_dev_attr.o wps/wps_enrollee.o wps/wps_registrar.o wps/wps_ufd.o utils/libutils.a crypto/libcrypto.a tls/libtls.a lwe/libiw.a -lm -lpcap  -o wash
wpsmon.c: In function ‘parse_wps_settings’:
wpsmon.c:358:26: warning: unused variable ‘encryption’ [-Wunused-variable]
     enum encryption_type encryption = NONE;
                          ^~~~~~~~~~
/usr/bin/ld: i386 architecture of input file `globule.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `init.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `misc.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `80211.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `iface.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `builder.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `crc.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `libwps/libwps.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `wps/wps_attr_build.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `wps/wps_attr_parse.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `wps/wps_attr_process.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `wps/wps.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `wps/wps_common.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `wps/wps_dev_attr.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `wps/wps_enrollee.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `wps/wps_registrar.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `wps/wps_ufd.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `utils/base64.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `utils/common.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `utils/uuid.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `utils/wpa_debug.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `utils/wpabuf.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `utils/os_unix.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `utils/eloop.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `crypto/aes-cbc.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `crypto/aes-internal-dec.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `crypto/aes-internal-enc.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `crypto/dh_group5.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `crypto/dh_groups.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `crypto/sha1-internal.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `crypto/sha256.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `crypto/sha256-internal.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `crypto/crypto_internal-modexp.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `crypto/aes-internal.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `tls/bignum.o' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `lwe/libiw.a(iwlib.so)' is incompatible with i386:x86-64 output
collect2: error: ld returned 1 exit status
Makefile:37: recipe for target 'wash' failed
make: *** [wash] Error 1

[image]

rofl0r, 2017-08-30:
you're somehow mixing up compilers for 32bit and 64bit. did you add -m32 to your CFLAGS, or do you have some broken environment variables set up?

Ghostfro, 2017-08-31:
Hi rofl0r
Actually i already installed the latest version of reaver from github, and when i tested it on a pixiedust attack with -K 1 it failed to find the pin, i know its a normal thing for some routers that are unvulnerable, but in an older version of reaver that i was using, even after it fails, it start the attack again automaticly with the --force or -f command, and then i just have to wait 30 min to get the pin. But in the latest version it dosent work like that as they removed --force or -f, thats why i downloaded a slightly older version of reaver to make it work. But i got this error while installing it.

Ghostfro, 2017-08-31:
Im also a little new to pen testing and all that linux stuff, and i didnt really understand your question : 
 did you add -m32 to your CFLAGS, or do you have some broken environment variables set up?
I explained everything i did in the last comment, if you need some more info just tell me

rofl0r, 2017-08-31:
> pixiedust attack with -K 1 it failed to find the pin

the latest version outputs the entire pixidust command, and its output. if pixiewps says that the AP is potentially vulnerable, it's often sufficient to wait for a minute, then paste the entire command into the shell again and it works (as far as i understood this would be due to a difference of your system time and the AP's system time).

soxrok2212, 2017-09-13:
No response for 2 weeks, assuming you've fixed the issue.

closed 2017-09-13 by soxrok2212