t6x/reaver-wps-fork-t6x #164: [!] WPS transaction failed (code: 0x04), re-trying last pin

shibumi, 2017-09-22:
Hello,
Do you have any idea why I get:
```
[!] WPS transaction failed (code: 0x04), re-trying last pin
```
with every wifi device I try? 

I use AWUS036H with chipset rtl8187. Everything else works fine. 
I can go into monitor mode, inject packets, do fakeauth attacks etc. 

Only reaver with WPS transactions fails. When I specify the right WPS pin via `-p` option I get the following result:

```
[+] 90.91% complete @ 2017-09-23 01:32:52 (0 seconds/pin)
```

Isn't that kind a strange? I have 90.91% completed but 0 seconds / pin and the WPS transaction failed.
In my `dmesg` log I see btw this here:

```
[28237.698748] device wlan0mon entered promiscuous mode
[28245.574900] device wlan0mon left promiscuous mode
[28256.658792] device wlan0mon entered promiscuous mode
[28259.950367] device wlan0mon left promiscuous mode
[28382.220202] device wlan0mon entered promiscuous mode
[28382.380457] device wlan0mon left promiscuous mode
[28383.446854] device wlan0mon entered promiscuous mode
[28383.863655] device wlan0mon left promiscuous mode
[28384.923594] device wlan0mon entered promiscuous mode
[28385.520325] device wlan0mon left promiscuous mode
[28575.852230] device wlan0mon entered promiscuous mode
[28576.125832] device wlan0mon left promiscuous mode
[28577.178835] device wlan0mon entered promiscuous mode
[28577.519107] device wlan0mon left promiscuous mode
[28578.575554] device wlan0mon entered promiscuous mode
[28578.852310] device wlan0mon left promiscuous mode
[28579.915546] device wlan0mon entered promiscuous mode
[28580.395732] device wlan0mon left promiscuous mode
[28581.458876] device wlan0mon entered promiscuous mode
[28581.806469] device wlan0mon left promiscuous mode
[28582.862201] device wlan0mon entered promiscuous mode
[28583.146124] device wlan0mon left promiscuous mode
[28584.205509] device wlan0mon entered promiscuous mode
[28584.592760] device wlan0mon left promiscuous mode
[28585.642065] device wlan0mon entered promiscuous mode
[28585.918875] device wlan0mon left promiscuous mode
```

and while doing `airmon-ng start wlan0` I get the following:
```
		(mac80211 monitor mode vif enabled for [phy2]wlan0 on [phy2]wlan0mon)
		(mac80211 station mode vif disabled for [phy2]wlan0)
```

I get the stuff on my kali as well on my arch linux.  Both with reaver version: 1.6.1

Any idea how to fix that issue?

rofl0r, 2017-09-23:
the constant changing of promiscious mode suggests to me that there's some application interfering in the background. shut down all process using your wlan interface, such as wpa_supplicant, etc.
btw you should use `airmon-zc` to put card into monitor mode, and it will alert you about interfering processes.

shibumi, 2017-09-25:
Hello,
Thanks for your comment. Well I checked for other active network software with `airmon-ng check` and it said that there is no other software active.

But that didn't seem to be the problem. When I attacked my router via reaver I realized that it does work when I add the `-w`-parameter. But only a specific distance. When I leave the room for example it stopped working and I get a lot of failed WPS transactions. But one day later it worked on the same distance. That's pretty non-deterministic.

rofl0r, 2017-09-25:
it would work deterministically with a cable. but you can't really influence the WIFI traffic of your neighbours interfering with your experiments.

closed 2017-09-25 by rofl0r