t6x/reaver-wps-fork-t6x #166: Problem armhf compiled -K function of reaver

nlona, 2017-09-25:
I downloaded latest release of reaver & pixiewps foto my raspberry pi2 & compiled&installed but once rebooted all' -K options are ignored, it behaved like reaver w/o pixiewps although they are both installed. Try also to un'installazione/reboot/make clean/apt-get clean/reboot/configure/make & make install/reboot. It did not work.

soxrok2212, 2017-09-25:
What do you mean 'all -K options are ignored'? Is it still recognized as a valid command?

nlona, 2017-09-25:
Neither -K 1 nor -K 4! You meant that there are no more needs of figures After K?

nlona, 2017-09-25:
It is recognized as a normale older reaver command no list of how many tries have been submitted when you restart a session e.g.

nlona, 2017-09-25:
Should I replace -K 1 with -K & -K 4 with -Z?

rofl0r, 2017-09-25:
use only `-K`.

nlona, 2017-09-26:
reaver -i wlan0mon -vv -b E0:B9:E5:6C:82:0
9 -c 1 -d 0 -N -S -t 7  -l 240                               

Reaver v1.6.1 

[+] Switching wlan0mon to channel 1
[+] Waiting for beacon from E0:B9:E5:XX:XX:XX
[+] Associated with E0:B9:E5:XX:XX:XX (ESSID: XXXXX
-9)
[+] Trying pin "12345670"
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received M3 message
[+] Sending M4 message
[+] Received WSC NACK
[+] Sending WSC NACK
[+] Trying pin "00005678"
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received M3 message
[+] Sending M4 message
[+] Received WSC NACK
[+] Sending WSC NACK
[+] Trying pin "01235678"

Could you suggest me how to get pin attempt Number & total?

nlona, 2017-09-26:
Switching wlan0mon to channel 1
[?] Restore previous session for E0:B9:E5:6C:82:09? [n/Y]
[+] Restored previous session
[+] Waiting for beacon from E0:B9:E5:6C:82:09
[+] Associated with E0:B9:E5:6C:82:09 (ESSID: FASTWEB-1-6C8209)
[+] Trying pin "11115670"
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Received M3 message
[+] Sending M4 message
[+] Received WSC NACK
[+] Sending WSC NACK
[!] WARNING: Detected AP rate limiting, waiting 60 seconds befor
 re-checking

nlona, 2017-09-26:
Could you please erase my bssid's last 6 exadec data?

nlona, 2017-09-26:
A

kcdtv, 2017-09-26:
Hello there
Your tittle is quite confusing.
-K  option (do a pixie dust attack + retrieve the WPA key of PIN is found) is left behind for the moment. When you use -K it is like using -Z: reaver will perform a pixie dust attack but will not send the PIN. 

> Could you suggest me how to get pin attempt number & total & progress percentage?

You tried 3 first half PIN so you have a bit less than 0,3% done. 
 Open the ***.wpc** file called `E0B9E5XXXXXX.wpc` and check the headers: you will know exactly how many first and second half PIN you tried.

closed 2017-09-26 by kcdtv

nlona, 2017-09-26:
Just a lil problem, as U can see there is no E0B9E5XXXXXX.wpc file in the appropriate reaver folder
root@kali:~# cat /etc/reaver/
00OXXXXXXXX.wpc  EC0XXXXXXXX.wpc
60E327XXXXXX.wpc  reaver.db

rofl0r, 2017-09-26:
look in /usr/local/etc, you probably didnt set the right directories when running configure

kcdtv, 2017-09-27:
I don't think that nlona will find any new session in `/etc/reaver` (or `/usr/local/etc/reaver`) but in `/var/lib/reaver` (or `/usr/local/var/lib/reaver`) 
Rember: Sessions can't be in etc directory because a little TUX baby  would die every minutes if we would have kept the directory that  reaver used since day one to store the sessions....:smiley_cat:

rofl0r, 2017-09-28:
hahhahaha lmao