t6x/reaver-wps-fork-t6x #169: Reaver stuck at "Waiting for beacon"

Hritik14, 2017-10-08:
Reaver v1.6.2
OS: Arch.
Journal logs.
```Oct 08 18:10:11 Journey pam_usb[26016]: Access granted.
Oct 08 18:10:11 Journey sudo[26016]:   hritik : TTY=pts/3 ; PWD=/home/hritik ; USER=root ; COMMAND=/usr/bin/reaver -i wlp2s0mon -b MAC:ADDRESS -vv
Oct 08 18:10:11 Journey sudo[26016]: pam_unix(sudo:session): session opened for user root by (uid=0)
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: Queue 1 is active on fifo 5 and stuck for 10000 ms. SW [0, 14] HW [0, 14] FH TRB=0x00
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: Microcode SW error detected.  Restarting 0x2000000.
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: Start IWL Error Log Dump:
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: Status: 0x00000200, count: 6
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: Loaded firmware version: 29.541020.0
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x00000084 | NMI_INTERRUPT_UNKNOWN       
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x200002B0 | trm_hw_status0
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x00000000 | trm_hw_status1
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x00043D58 | branchlink2
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x0004B016 | interruptlink1
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x00007354 | interruptlink2
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x00000000 | data1
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x00000080 | data2
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x03030000 | data3
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x00000000 | beacon time
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x0DBE2AAF | tsf low
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x00000000 | tsf hi
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x00000000 | time gp1
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x0DBE2AAF | time gp2
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x00000001 | uCode revision type
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x0000001D | uCode version major
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x0008415C | uCode version minor
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x00000210 | hw version
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x00C89200 | board version
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x00220400 | hcmd
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x27E62000 | isr0
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x01000000 | isr1
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x00000002 | isr2
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x00401080 | isr3
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x00000000 | isr4
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x00220400 | last cmd Id
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x00000000 | wait_event
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x00004288 | l2p_control
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x00010020 | l2p_duration
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x00000000 | l2p_mhvalid
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x00002000 | l2p_addr_match
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x00000007 | lmpm_pmg_sel
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x15061956 | timestamp
Oct 08 18:10:30 Journey kernel: iwlwifi 0000:02:00.0: 0x0034E0F8 | flow_handler
Oct 08 18:10:30 Journey kernel: ieee80211 phy1: Hardware restart was requested
Oct 08 18:11:50 Journey sudo[26016]: pam_unix(sudo:session): session closed for user root
```

rofl0r, 2017-10-08:
try latest git. i have no idea what your log there is trying to demonstrate.

Hritik14, 2017-10-18:
Updated. Now it's stuck at "Trying pin 123456789..."

rofl0r, 2017-10-18:
run with -vvv

rofl0r, 2017-10-18:
a quick guess is that association does not work. you can also try to use `-A ` option and use airreplay-ng to do the association in a second terminal.

Hritik14, 2017-10-18:
Ran with -vvv and with airodump-ng running, now it's
WARNING:Receive timeout occurred.

PS: I changed the MAC.

rofl0r, 2017-10-18:
looks like you're too far away from the AP. maybe time to go wardriving ? :-O

oh, i forgot: you're supposed to use this program only on your own routers LOL

Hritik14, 2017-10-18:
I can confirm that the AP is not too far.

kcdtv, 2017-10-19:
That doesn't mean anything. 
We need to see the RXQ and PWR level in airodump-ng terminal. 
You can eventually rise the timeout delay
Which wifi chip are you using?

Hritik14, 2017-11-13:
` BSSID              PWR  Beacons    #Data, #/s  CH  MB   ENC  CIPHER AUTH ESSID
 EC:08:6B:F2:B6:6A  -53       26        0    0   4  54e. WPA2 CCMP   PSK  TP-LINK_B66A        
Now it's stuck on Trying pin
`Reaver v1.6.2 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Waiting for beacon from EC:08:6B:F2:B6:6A
[+] Switching wlp2s0mon to channel 1
[+] Switching wlp2s0mon to channel 2
[+] Switching wlp2s0mon to channel 3
[+] Switching wlp2s0mon to channel 4
[+] Received beacon from EC:08:6B:F2:B6:6A
[+] Associated with EC:08:6B:F2:B6:6A (ESSID: TP-LINK_B66A)
[+] Trying pin "12345670"`

I'm using 
Network controller: Intel Corporation Dual Band Wireless-AC 3165 Plus Bluetooth (rev 99)
Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 10)

kcdtv, 2017-11-16:
Thanks for giving more information.

>  Intel Corporation Dual Band Wireless-AC 3165 

I am not familiar with this chipset. 
It could be due to your chipset.
I say it "could" because we lack information about what you do (we don't even see the command line you used)
It is possible to try to fix it but you first need to do as explained here:  [https://github.com/t6x/reaver-wps-fork-t6x/issues/175#issuecomment-341671218](url

Hritik14, 2017-11-16:
@kcdtv  So I'd need another Wireless card ? I'm afraid I don't have one.

kcdtv, 2017-11-30:
Hi there! 
Sorry for not answering, i didn't see your post... anyway, the answer is yes: You need a second wireless interface to debug.

Hritik14, 2017-12-05:
@kcdtv Sorry but I don't seem to understand very well. If I have to buy another wireless card to debug and if that works then why won't I be using that one for everything ?
Well, I have another PC. Will that help ? 
Can you guide a bit to debug or maybe some search terms, articles ?

kcdtv, 2017-12-06:
First i can't assure you that it will work. 
It could be a reaver issue that can be fixed, it could be something else. I won't recommend you to buy a wireless card for that. 
And a second card (that works properly) is required to sniff the traffic in order to be able to undertsand what could be wrong with the buggy interface
If you have a second laptop with a supported interface than that would work. 
First check with your second PC if you can crack your PA with the correct PIN. 
If you can crack your PA with this second PC we will see for debugging.

kcdtv, 2017-12-08:
check this fix: #175

kcdtv, 2018-02-23:
More than two months passed since last message... It seems that no feedback will come, let's close this issue.

closed 2018-02-23 by kcdtv