t6x/reaver-wps-fork-t6x #171: Reaver stuck at error (code 0x02) after cracking for some hours

nikeby, 2017-10-10:
Hello there,

I am using a live version in a USB of Kali Rolling, the most recent version that was available, with a AWUS036H. When I try to crack my router (Linksys E2500), the following can happen: I can crack the WPS pin in a couple of hours or I start to get an error that doesn't allow me to keep cracking. In the past I cracked it many times, but the last time I tried, the crack progressed up to 5% and the error started. Tried many fixes, but none worked, and I thought cracking it was out of question. Then, last week (was with a lot of free time), I decided to play with reaver again. And, for my surprise, I was able to almost crack it again, with error showing up at 91.67% progress. I am thinking that might be some sort of security measure the router is triggering. This was last Friday, when the error happened. Waited until today, but the error persisted. So I want to know if anyone there have an ideia of a possible solution to this issue, besides waiting months or updating reaver branch (dont't know which one solved my problem).

Here's a very similar issue, just timeouts, no matter the tool I use. The only difference is that it only happens with my router, which seems like I exhausted it.

[https://forums.kali.org/showthread.php?35973-Reaver-Bully-problem-Kali-problem-backports-problems&highlight=mdk3](url)

Thank you in advance, any help will be appreciated.

Ah, I don't think it'll help, but I took a pic when the last pin was tested sucessfully:

[image]

**Reaver command:**
`reaver -i wlan0mon -b 48:F8:B3:xx:xx:xx -e xxxx -c 11 -d 4 -r 5:15 -A -N -S -w -x 360 -vvv`

**Reaver:**
```
Reaver v1.6.2 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Switching wlan0mon to channel 11
[?] Restore previous session for 48:F8:B3:xx:xx:xx? [n/Y] y
[+] Restored previous session
[+] Waiting for beacon from 48:F8:B3:xx:xx:xx
[+] Received beacon from 48:F8:B3:xx:xx:xx
[+] Associated with 48:F8:B3:xx:xx:xx (ESSID: xxxx)
[+] Trying pin "09670754"
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[!] WARNING: Receive timeout occurred
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x02), re-trying last pin
[+] Trying pin "09670754"
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[!] WARNING: Receive timeout occurred
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x02), re-trying last pin
[+] Trying pin "09670754"
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[!] WARNING: Receive timeout occurred
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x02), re-trying last pin
```

**Aireplay command:**
`aireplay-ng -1 60 -b 48:F8:B3:xx:xx:xx -e xxxx -o 1 -q 10 wlan0mon`

**Aireplay:**
```
No source MAC (-h) specified. Using the device MAC (00:C0:CA:xx:xx:xx)
16:40:28  Waiting for beacon frame (ESSID: xxxx) on channel 11
Found BSSID "48:F8:B3:xx:xx:xx" to given ESSID "xxxx".

16:40:28  Sending Authentication Request (Open System) [ACK]
16:40:28  Authentication successful
16:40:28  Sending Association Request [ACK]
16:40:28  Association successful :-) (AID: 1)

16:40:38  Sending keep-alive packet [ACK]
16:40:48  Sending keep-alive packet [ACK]
16:40:58  Sending keep-alive packet [ACK]
16:41:08  Sending keep-alive packet [ACK]
16:41:18  Sending keep-alive packet [ACK]
16:41:28  Sending Authentication Request (Open System)

16:41:30  Sending Authentication Request (Open System) [ACK]
16:41:30  Authentication successful
16:41:30  Sending Association Request [ACK]
16:41:30  Association successful :-) (AID: 1)
```

kcdtv, 2017-11-06:
> I am thinking that might be some sort of security measure the router is triggering.

The only security measure that offcially exit is the lock system but your AP doesn't lock at out.  
Maybe somehow this is too much connection attempt for your router sometimes?

minanagehsalalma, 2017-12-07:
@nikeby  i think it's wps pbc wifi not pin @kcdtv

kcdtv, 2018-02-23:
Nope, none of them. Anyway it is a long time since this issue was opened and there is no more feedback so let's close it.

closed 2018-02-23 by kcdtv