t6x/reaver-wps-fork-t6x #172: Reaver Problem : WPS transaction failed (code: 0x04), re-trying last pin. Please HELP !!

Ghostfro, 2017-10-11:
Hi everyone,
im trying to get the pin of an AP, but i have this problem, 
Reaver WPS Transaction Failed (code: 0x04 ), retrying last pin.
and it repeats the first pin : "12345670"
so it's stuck at 0%
The PWR is good  "-55" , it's a WPA2 protected AP, i already tryed Pixiewps and the attack fails so i wanted to get the pin with only reaver.
the injection is 100% on this ap when i tested it with aireplay, and the wireless card is totaly working with other AP's so it's not a wireless card problem " it's an Alfa AWUS036H " 
i searched so much in google, still didnt found any solution so far.
The AP is a Technicholor as u can see 
can anyone help me ? thx :)

wifislax64 ~ # airmon-ng start wlan0

Found 3 processes that could cause trouble.
If airodump-ng, aireplay-ng or airtun-ng stops working after
a short period of time, you may want to kill (some of) them!

PID     Name
1855    NetworkManager
2327    wpa_supplicant
2357    dhclient

Interface       Chipset         Driver

wlan0           Realtek RTL8187L        rtl8187 - [phy0]
                                (monitor mode enabled on mon0)

wifislax64 ~ # airmon-ng check kill

Found 3 processes that could cause trouble.
If airodump-ng, aireplay-ng or airtun-ng stops working after
a short period of time, you may want to kill (some of) them!

PID     Name
1855    NetworkManager
2327    wpa_supplicant
2357    dhclient
Killing all those processes...
wifislax64 ~ # 

 CH 10 ][ Elapsed: 6 s ][ 2017-10-10 22:48                                         
                                                                                                                          
 BSSID              PWR  Beacons    #Data, #/s  CH  MB   ENC  CIPHER AUTH ESSID
                                                                                                                          
 C4:EA:1D:1F:5A:67  -55        7        0    0  11  54e  WPA2 CCMP   PSK  TNCAP1F5A67                                     
 FE:98:66:99:E6:79  -59        9        0    0   1  54e  WPA2 CCMP   PSK  SALIHA                                          
 18:D6:C7:7F:AC:44  -63       11        0    0   9  54e. WPA2 CCMP   PSK  <length:  0>                                    
 00:19:70:38:FF:6F  -69        2        0    0   1  54e. WEP  WEP         ADSL5272                                        
                                                                                                                          
 BSSID              STATION            PWR   Rate    Lost    Frames  Probe                                                
                                                                                                                          
 (not associated)   50:F0:D3:46:11:FC  -67    0 - 1      0        2  TNCAP1F5A67
 
 
 
 wifislax64 ~ # reaver -i mon0 -b C4:EA:1D:1F:5A:67 -c 11 -vvv 

Reaver v1.5.4 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
t6_x <t6_x@hotmail.com> & DataHead & Soxrok2212 & Wiire & AAnarchYY & KokoSoft fork
MOD by vk496 for www.seguridadwireless.net

[+] Switching mon0 to channel 11
[+] Waiting for beacon from C4:EA:1D:1F:5A:67
[+] Associated with C4:EA:1D:1F:5A:67 (ESSID: TNCAP1F5A67)
[+] Starting Cracking Session. Pin count: 0, Max pin attempts: 11000
[+] Trying pin "12345670"
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[... 119 lines cut ...]
[P] AuthKey: d2:56:80:1d:53:bf:ce:14:11:df:f4:4b:c7:37:a5:3b:a0:9b:2f:3c:8b:bb:a7:87:6d:30:e0:cb:32:62:28:d8
[+] Sending M2 message
[+] Received WSC NACK (reason: 0x000F)
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x04), re-trying last pin
[+] Trying pin "12345670"
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[P] E-Nonce: 02:6b:ac:80:62:e4:ea:6d:90:30:82:c3:91:69:cf:b9
[P] PKE: 2a:97:0e:6f:c6:97:93:73:e0:1b:6e:53:78:9e:65:d9:17:03:3f:29:94:7d:f9:cd:71:83:0f:97:61:37:a0:94:7b:ff:0a:11:d3:af:fb:9c:31:a8:96:ce:d1:6b:a6:c7:61:6a:b9:a6:47:26:cb:a8:fd:1e:fa:be:de:75:bd:26:07:8e:e3:95:31:6f:5c:0c:cf:14:58:24:47:e2:d3:53:49:93:9d:68:92:58:2a:a9:20:b3:58:56:ad:57:d5:d1:63:85:8b:22:66:e4:31:08:c7:94:2d:17:d4:84:46:91:fa:0b:f4:76:12:b3:b8:19:a8:d0:4d:9a:23:52:44:e9:88:57:04:55:fc:bd:b5:50:de:c8:29:5c:66:41:21:88:2c:a0:0a:6f:d6:19:24:7c:04:63:67:3d:7c:0c:9c:e9:74:01:3c:db:6c:7a:d6:d0:3a:d4:cf:90:23:19:7d:4e:2e:9a:aa:9e:14:e3:3b:3e:59:34:4b:4b:af:af:2a:6b
[P] WPS Manufacturer: Technicolor
[P] WPS Model Name: MediaAccess TG
[P] WPS Model Number: 799vn v2
[P] Access Point Serial Number: 1528CS3J9
[+] Received M1 message
[P] R-Nonce: 1a:af:77:fc:d2:58:92:4a:ba:6f:f3:e9:61:a3:c6:ed
[P] PKR: 53:9c:79:b1:7d:2d:6a:dc:38:59:42:f9:e4:7c:e2:d5:88:2d:dd:b2:2d:b8:17:b1:62:25:39:c4:05:76:04:37:0f:db:07:ca:84:d0:11:1b:3a:88:e6:d4:ba:0f:96:45:6b:76:0e:13:75:5c:5d:11:5d:ab:be:a8:71:54:68:78:a8:c3:6c:6d:ad:17:55:4d:f1:1a:c7:e5:80:3c:c3:da:a3:1a:b0:6f:8e:2e:46:41:54:63:e0:df:ba:f8:5a:93:90:3f:c6:f4:3f:a1:38:cd:73:4e:1b:b7:ab:82:63:5f:47:2c:23:4f:ea:20:ac:bf:c7:7e:f5:45:80:a7:3b:2a:8d:21:34:01:12:0c:fe:94:e9:57:55:3f:9a:f0:66:30:d3:a8:e1:54:5e:cd:c6:0b:d8:53:64:bb:3c:36:f6:bd:bc:c1:0c:a8:cd:03:7f:f2:bf:63:2e:8b:8c:6a:64:0b:5d:e1:12:77:d5:aa:4d:b3:c3:0b:a9:4e:6d:b2:b2:da
[P] AuthKey: e3:5d:1e:84:d0:0d:c4:98:bc:23:d1:cb:32:5d:4e:8e:e6:e1:39:86:a0:14:48:27:94:23:48:f0:1d:fe:e1:21
[+] Sending M2 message
[+] Received WSC NACK (reason: 0x000F)
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x04), re-trying last pin
^C
[+] Nothing done, nothing to save.

soxrok2212, 2017-10-11:
Is WPS pin mode enabled in the web UI?

soxrok2212, 2017-10-11:
The admin panel on the router.

Ghostfro, 2017-10-11:
yes

Ghostfro, 2017-10-11:
and it shows also on the wash command

soxrok2212, 2017-10-11:
Also, you're using a very old version of Reaver. Please update to the newest. Try manually specifying the PIN first and see if you recover the key that way.

Ghostfro, 2017-10-11:
I used every version of reaver got the same problem, even the lastest one ! and my goal is to get the pin of this AP, i dont know it so i cant try it manualy with "-p"

soxrok2212, 2017-10-11:
This tool is meant to test your own AP that you already know the credentials for. We do not support breaking into other people's networks here.

closed 2017-10-11 by soxrok2212

Sayedaghaj, 2023-03-04:
> This tool is meant to test your own AP that you already know the credentials for. We do not support breaking into other people's networks here. That being said, you will not be able to supply proper information to even fix an issue if there is one, since you don't own the AP.

FUCK YOU BULLLY 
DICK  @!

Sayedaghaj, 2023-03-04:
> This tool is meant to test your own AP that you already know the credentials for. We do not support breaking into other people's networks here. That being said, you will not be able to supply proper information to even fix an issue if there is one, since you don't own the AP.

FUCK YOU BULLY
DICK

soxrok2212, 2023-03-04:
> > This tool is meant to test your own AP that you already know the credentials for. We do not support breaking into other people's networks here. That being said, you will not be able to supply proper information to even fix an issue if there is one, since you don't own the AP.
> 
> FUCK YOU BULLY DICK

I was going to delete this but I hope if you have a future employer(s), they see this and reconsider.