t6x/reaver-wps-fork-t6x #173: Failed to Associate with BSSID

TeslaKepler, 2017-10-20:
Hi!
I'm trying to use reaver, however I receive thise error : [Warning]: failed to associate with BSSID.
The wps mode is on.
I'm pretty newbie, but I tried few things.
reaver -i wlan0mon -b BSSID -S -N -vv
reaver -i wlan0mon -b BSSID -K 1 -vv
reaver -i wlan0mon -b BSSID -S -N -A -vv

I tried aireplay-ng, but I'm not sure if I did it correctly. I did something like that : -watch -n30 aireplay-ng -1 0 wlan0mon -a BSIDD
I'm wondering if the issue could be my wifi card, intel 4965 AG with driver iwl4965.
The RSSI is -58 I'm not sure if it's good enough.

rofl0r, 2017-10-20:
read this comment about how to use reaver with aireplay-ng doing the association:
https://github.com/t6x/reaver-wps-fork-t6x/issues/171#issue-264368222

also -K 1 is no longer needed/supported, make sure to grab latest git version of reaver

TeslaKepler, 2017-10-22:
Hi rofl0r,

I followed the guide and I still have the same issue.

rofl0r, 2017-10-22:
with the latest version from git ? then you would at least get different output.

TeslaKepler, 2017-10-23:
I updated kali linux, I'm not sure if I got the latest version.

Yeah I tried alot of different output.

rofl0r, 2017-10-23:
getting the latest version means compiling from source

TeslaKepler, 2017-10-24:
I'll try it and let you know.

TeslaKepler, 2017-11-02:
Alright, I did the update. I'm using reaver 1.6.2 from github.

However, I got a new error : receive time out occured.

I tried few different command and I still have the save error.

The rssi is around -90.

Aireplay-ng doesn't work as well, authentication failed.

soxrok2212, 2017-11-02:
Move closer and try again.

TeslaKepler, 2017-11-02:
Is -90 is good or bad ?

what rssi is good ?

soxrok2212, 2017-11-02:
-90 is horrible. It depends on the chip you use and its sensitivity. For my AWUS036H, -65 and lower is good, for my WN722N, -70 and lower is good.

TeslaKepler, 2017-11-02:
Alright, I wasn't sure if higher the number is, better is the connection.