t6x/reaver-wps-fork-t6x #174: Stuck at 90.90%

idanlo, 2017-10-24:
root@kali:~# reaver -b AA:BB:CC:DD:EE:FF -c 11 --dh-small -N -L -i wlan0mon 

Reaver v1.6.1 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[?] Restore previous session for 20:76:00:14:A1:90? [n/Y] y
[+] Restored previous session
[+] Waiting for beacon from 20:76:00:14:A1:90
[+] Associated with 20:76:00:14:A1:90 (ESSID: neta2)
[+] 90.90% complete @ 2017-10-24 10:12:31 (2 seconds/pin)
[+] 90.90% complete @ 2017-10-24 10:12:40 (1 seconds/pin)
[+] 90.90% complete @ 2017-10-24 10:12:50 (1 seconds/pin)

This just goes on and on.. I've seen online many people have this problem but no solution found yet..

rofl0r, 2017-10-24:
Reaver v1.6.1 -> too old
use latest git, and use -vv

idanlo, 2017-10-25:
How do I get the latest git?

idanlo, 2017-10-25:
After cloning the git I need to do the other setup mentioned in the main page?

rofl0r, 2017-10-25:
you need to compile it, according to the instructions

kcdtv, 2017-12-12:
As the thread is invalid and idanlo didn't show up since more than one month this issue is closed.

closed 2017-12-12 by kcdtv