t6x/reaver-wps-fork-t6x #175: timout errors

oberon2007, 2017-11-03:
I'm finding reports about this problem in many places on the web, but none of those people apear to have found a solution to this. So I am running into this same issue and can't seem to be able to fix it.
This happens for me with two different devices - Intel Wireless 8265/8275 and broadcom BCM43228, I have tried multiple networks with pwr well over 80 and all sorts of command line options, extended timeout times and whatnot - no dice.
```
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
```

rofl0r, 2017-11-03:
which version of reaver ? try latest git.

oberon2007, 2017-11-03:
I'm already using https://github.com/t6x/reaver-wps-fork-t6x/commit/762093d62981363aba9f2c56148680249b5ed8a5

rofl0r, 2017-11-03:
ok. if you wanna help debug the issue, here's how

1) turn on monitor mode
2) start wireshark in another terminal and listen on that interface
3) turn off all running network manager and wpa_supplicant instances
4) use this script to try to send a single pin (or pixiedust mode) to the AP https://0x0.st/ssl0.py
5) if it works, go to wireshark, and save the capture under some filename
6) start wireshark again
7) do the same in reaver
8) save .cap
9) export the EAPOL start packet from the good capture (the one made with the python script) and the bad capture (the one made with reaver)
10) post them here so we can find out what's different

oberon2007, 2017-11-04:
I'll be able to check this in a few days. Back home I have a setup with two interfaces. 😉

Eimert, 2017-12-05:
@rofl0r I did what you requested, would you be so kind to debug reaver using these wireshark captures?
[2017-12-04_reaver_wireshark_capture_for_debugging.zip](https://github.com/t6x/reaver-wps-fork-t6x/files/1532656/2017-12-04_reaver_wireshark_capture_for_debugging.zip)

**Inside the zip:**
- A capture of a successful WPS connection using wpa_cli with wps_reg. The ssl0.py script didn't work for me. _2017-12-04_wlan1mon_wps_tp-link_wpa_cli_success.pcapng_ 
- The wpa_supplicant config and log: _2017-12-04_wlan0_wps_tp-link_wpa_cli_setup.txt_
- A capture of reaver trying to connect using the same pin that should work. _2017-12-04_wlan1mon_wps_tp-link_reaver_fail.pcapng_

OS: Kali 2017.2
Reaver build on 2017-12-04 from master branch commit 7d4153b 
libpcap0.8 1.8.1-5 amd4
kernel 4.12-0

rofl0r, 2017-12-05:
what issue did you have with the python script ? your .txt file misses all the debug output (you should have run wpa_supplicant with -d)

rofl0r, 2017-12-05:
ok, it seems the pcap's have the answer

rofl0r, 2017-12-05:
@Eimert can you try this patch: http://sprunge.us/hKPP
apply it, make clean, make - test

rofl0r, 2017-12-05:
if that didn't work, try this patch (but git pull first) : http://sprunge.us/hWaD

Eimert, 2017-12-06:
Tried both patches and even combined them. It does not work (yet).
```
root@lieke:~/reaver-wps-fork-t6x/src# reaver -i wlan1mon -b F4:EC:38:DD:93:D8 -c2 -p63066364 -vvv

Reaver v1.6.3 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Switching wlan1mon to channel 2
[+] Waiting for beacon from F4:EC:38:DD:93:D8
[+] Received beacon from F4:EC:38:DD:93:D8
[+] Vendor: AtherosC
WPS: A new PIN configured (timeout=0)
WPS: UUID - hexdump(len=16): [NULL]
WPS: PIN - hexdump_ascii(len=8):
     36 33 30 36 36 33 36 34                           63066364        
WPS: Selected registrar information changed
WPS: Internal Registrar selected (pbc=0)
WPS: sel_reg_union
WPS: set_ie
WPS: cb_set_sel_reg
WPS: Enter wps_cg_set_sel_reg
WPS: Leave wps_cg_set_sel_reg early
WPS: return from wps_selected_registrar_changed
[+] Trying pin "63066364"
[!] WARNING: Failed to associate with F4:EC:38:DD:93:D8 (ESSID: TP-LINK_DD93D8)
[!] WARNING: Failed to associate with F4:EC:38:DD:93:D8 (ESSID: TP-LINK_DD93D8)
^C
[+] Nothing done, nothing to save.

```
Here is another capture (using reaver patch in http://sprunge.us/hWaD): [2017-12-06_wlan1mon_wps_tp-link_reaver.zip](https://github.com/t6x/reaver-wps-fork-t6x/files/1536997/2017-12-06_wlan1mon_wps_tp-link_reaver.zip)

rofl0r, 2017-12-06:
try this on top of current git

http://sprunge.us/CJbh

Eimert, 2017-12-07:
**Big success**: by using http://sprunge.us/CJbh and by running reaver with -N (--no-nacks) 👍 .
```
root@lieke:~/reaver-wps-fork-t6x/src# reaver -i wlan1mon -b F4:EC:38:DD:93:D8 -c9 -p63066364 -vvv -N
(... output omitted ...)
[+] Received M7 message
WPS: Building Message WSC_NACK
WPS:  * Version
WPS:  * Message Type (14)
WPS:  * Enrollee Nonce
WPS:  * Registrar Nonce
WPS:  * Configuration Error (0)
[+] Sending WSC NACK
WPS: Building Message WSC_NACK
WPS:  * Version
WPS:  * Message Type (14)
WPS:  * Enrollee Nonce
WPS:  * Registrar Nonce
WPS:  * Configuration Error (0)
[+] Sending WSC NACK
[+] Pin cracked in 54 seconds
[+] WPS PIN: '63066364'
[+] WPA PSK: 'cocacola'
[+] AP SSID: 'TP-LINK_DD93D8'
[+] Nothing done, nothing to save.
```
Could you merge http://sprunge.us/CJbh into the master branch?

kcdtv, 2017-12-08:
\O/ Well done guys! :smiley_cat:

rofl0r, 2017-12-10:
i merged the fix, hopefully it doesnt break association with other routers that previously worked. i'm leaving this open, since it appears the issue encountered by @oberon2007 happened waiting for EAPOL response, while @Eimert issue was related to rejected association.

kimocoder, 2017-12-10:
Both me and a friend also struggled with timeout error, while ut worked before, so we'd happy to test with the new pr :) will take a dive soon, thanks

rofl0r, 2017-12-10:
> struggled with timeout error, while ut worked before

oh ? please more specifics. without details about in which version it works, and in which it doesnt (using the exact same AP and hardware and location), plus detailed info at which step (as can be seen with -vvv) this statement doesn't add any value to the conversation.

kcdtv, 2018-02-23:
Shouldn't we consider this issue as solved? As kimocoder did not reported any new problems it looks like there is no more troubles.

rofl0r, 2018-02-23:
i have a suspicion that the issue with no response after could be related to the lack of "active monitor" mode, as described in the README of my [wpakey](https://github.com/rofl0r/wpakey) project... so i guess it'd be good if @oberon2007 or @kimocoder can test with latest git and provide some more details if the issue persists.

oberon2007, 2018-02-23:
hm. unfortunately I still have the same problem with latest commit (48b1381)
```
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
```

rofl0r, 2018-02-23:
can you try with an ath9k_htc adapter (like tplink 722n) which has active monitor on by default ?

Chessy1991, 2018-09-01:
Hi, this isn't over for me yet... i didn't try the patch that worked for @Eimert ... tell me how to do that guys?!

rofl0r, 2018-09-01:
check with wireshark whether the AP reacts to the EAPOL START packets, if not, it's the fault of the AP, not reaver.

Zarados, 2018-09-04:
Hi, I find some interesting thing about this question.
I have two wifi dongle EDIMAX EW-7318USG and TP-Link TL-WN722N.
TL-WN722N timeout occurred after sending EAPOL Start Request every time.
EW-7318USG works until the AP lock WPS.
TL-WN722N can't work on it
(reaver -i <interface> -b <Mac address> -vv -N)
I'm not pro. Maybe someone @rofl0r  can find what's different between TL-WN722N and EW-7318USG.
I'm sure it not the question about reaver.
I hope this comment will help you.

kimocoder, 2018-09-04:
The TL-WN722N got different chipsets (v1, v2, v3) do this should be specified.

Zarados, 2018-09-04:
@kimocoder how should i check chipsets when i work? 
Thank you for your reply.

Zarados, 2018-09-04:
I see a number (v1.10) after serial number.

rofl0r, 2018-09-04:
what's the chip used by `EW-7318USG` ? i did a quick search but didn't find the answer.
if you @Zarados don't know it either you could tell us what driver `airmon-ng` reports and what `lsusb` tells about it.

the TL-WN722N v1.10 uses ATH9K_HTC which supports active monitor mode, so it's interesting that it fails here...

Zarados, 2018-09-06:
Hi, @rofl0r EW-7318USG used Ralink Technology, Corp. RT2501/RT2573 Wireless Adapter.
the TL-WN722N v1.10 uses ATH9K_HTC still work to monitor mode.
I used it to receive reaver packets to Wireshark.

I have the other question about reaver.
I can't find PIN number of brute force list from packets.
[https://imgur.com/a/mT08nBL](url)
What is device password ID 0x1012 ?

Sincerely

rofl0r, 2018-09-06:
looks to me like a magic number that tells the router that a PIN is being sent.
@Zarados could you provide 2 pcap: one with edimax adapter where we get response to EAPOL, and one with TPLINK where we don't get response ? this could be helpful trying to figure out what goes wrong.

Chessy1991, 2018-09-19:
> **Big success**: by using http://sprunge.us/CJbh and by running reaver with -N (--no-nacks) +1 .
> 
> ```
> root@lieke:~/reaver-wps-fork-t6x/src# reaver -i wlan1mon -b F4:EC:38:DD:93:D8 -c9 -p63066364 -vvv -N
> (... output omitted ...)
> [+] Received M7 message
> WPS: Building Message WSC_NACK
> WPS:  * Version
> WPS:  * Message Type (14)
> WPS:  * Enrollee Nonce
> WPS:  * Registrar Nonce
> WPS:  * Configuration Error (0)
> [+] Sending WSC NACK
> WPS: Building Message WSC_NACK
> WPS:  * Version
> WPS:  * Message Type (14)
> WPS:  * Enrollee Nonce
> WPS:  * Registrar Nonce
> WPS:  * Configuration Error (0)
> [+] Sending WSC NACK
> [+] Pin cracked in 54 seconds
> [+] WPS PIN: '63066364'
> [+] WPA PSK: 'cocacola'
> [+] AP SSID: 'TP-LINK_DD93D8'
> [+] Nothing done, nothing to save.
> ```
> 
> Could you merge http://sprunge.us/CJbh into the master branch?

how to use the script @Eimert ?

rofl0r, 2018-09-20:
@Chessy1991 that patch that you call "script" is a code change that's been already merged into reaver. so if you download the latest sourcecode here from git, you have it included.

rofl0r, 2018-09-22:
so you may check https://github.com/t6x/reaver-wps-fork-t6x/wiki/Troubleshooting

Chessy1991, 2018-09-22:
thanks craig!

rofl0r, 2018-09-22:
no problem, melanie!

W-quan, 2019-03-31:
I got the latest commit, still got timeout. somebody help me?

rofl0r, 2019-03-31:
@W-quan please follow the troubleshooting guide https://github.com/t6x/reaver-wps-fork-t6x/wiki/Troubleshooting

W-quan, 2019-04-01:
@rofl0r I had try it, but still doestn't work.  And i already had run at three PC.   

About 2 years ago, there is no preblem using `reaver 1.4`.

W-quan, 2019-04-01:
How to filter the output of Wireshark.  Can somebody tell me the rule?

rofl0r, 2019-04-01:
please report what happened along each step of the troubleshooting guide.
as for filtering wireshark, look for the first EAPOL paket (you can sort by protocol, then scroll down until "e" starts, its sorted alphabetically), note the frame number, then sort by framenumber and read from there on.

W-quan, 2019-04-03:
@rofl0r  In step 4,  i can see the AP had reply. 

Here is the cap file.

[cap.pcapng.zip](https://github.com/t6x/reaver-wps-fork-t6x/files/3039613/cap.pcapng.zip)

rofl0r, 2019-04-03:
you can see in the radiotap header of the beacon frames of the AP that signal quality (SSI) is -89 dbm that's really bad. you need to get a better signal, e.g. by using a directional antenna. i also don't see a reply to your EAPOL start packages when i put this filter `wlan.sa == 8c:a6:df:xx:xx:xx || wlan.da == 8c:a6:df:xx:xx:xx`.
the troubleshooting guide says you should try associating with aircrack-ng. did that work? i suppose no, with such a bad signal.
btw you might want to remove the pcap file now, it could contain sensible data.

W-quan, 2019-04-04:
I see reaver had associating to AP, so i didn't associating with aircrack-ng.

What about this:

The signal quality (SSI) is -73dbm, i think it is no bad.

[reaver+wire.zip](https://github.com/t6x/reaver-wps-fork-t6x/files/3043650/reaver%2Bwire.zip)

rofl0r, 2019-04-04:
what i see here in the pcap is that the router sends deauthentication packets after your EAPOL packets. this should be visible in the reaver output too, but in the attached text file it is not. once you get a deauth packet, you need to associate again. reaver doesn't do this automatically.

W-quan, 2019-04-06:
> what i see here in the pcap is that the router sends deauthentication packets after your EAPOL packets. this should be visible in the reaver output too, 

It's not visible in the reaver output. 

I try associated with airplay-ng, but still not work. 

[log.zip](https://github.com/t6x/reaver-wps-fork-t6x/files/3050195/log.zip)

rofl0r, 2019-04-06:
are you using latest git master ? the notification about deauth packets was recently added.

W-quan, 2019-04-07:
yeah, i had see the deauth packets, and alway get deauth packets. what's wrong about this case?

rofl0r, 2019-04-07:
i have no definite answer on that. i've experienced routers (albeit only in situations with relatively bad signal) that sent deauth packets sometimes, but not always, but i've also experienced routers that always send deauth packets. i suspect the former is due to error or congestion control mechanism by the router, the latter is potentially a "protection" against WPS hackers. I don't know how popular latest reaver is, but it is certainly possible to fingerprint it with enough effort. so the router could detect that somebody is using reaver, then send deauth packet. however imo its much more likely that the ISP shipping the router simply made WPS unusable for anybody to prevent attacks. very often the programmers of these embedded devices are really incompetent, so maybe they didn't find the proper way to disable WPS, so they just added some lines of code to make it unusable.

AnApprentince, 2020-06-07:
> if that didn't work, try this patch (but git pull first) : http://sprunge.us/hWaD

Very very dumb question i presume, but how do i apply the patch?

rofl0r, 2020-07-07:
> Very very dumb question i presume, but how do i apply the patch?

why not ask google ? regardless, the patch is 3 years old and most likely doesn't apply/help you.

samyak1971, 2022-08-11:
> 

Please explain kind sir

rofl0r, 2022-08-11:
closing due to lack of feedback about the original issue.

closed 2022-08-11 by rofl0r