t6x/reaver-wps-fork-t6x #179: compile on mac currently broken

DJDan, 2017-12-03:
Reference: #145
github.com/t6x/reaver-wps-fork-t6x/issues/145

./configure
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for pcap_open_live in -lpcap... yes
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /usr/local/bin/ggrep
checking for egrep... /usr/local/bin/ggrep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for stdlib.h... (cached) yes
checking for stdint.h... (cached) yes
checking for string.h... (cached) yes
checking pcap.h usability... yes
checking pcap.h presence... yes
checking for pcap.h... yes
configure: creating ./config.status
config.status: creating config.mak

====

make
gcc -DCONF_DIR='"/usr/local/var/lib/reaver"' -Wall -g -O2 -Wno-unused-variable -Wno-unused-const-variable -Wno-unused-function -Wno-pointer-sign -Ilibwps -I. -Ilwe -DCONFIG_IPV6   -c -o iface.o iface.c
iface.c:79:19: error: use of undeclared identifier 'SIOCGIFHWADDR'
                        if(ioctl(sock, SIOCGIFHWADDR, &ifr) == 0)
                                       ^
iface.c:81:36: error: no member named 'ifr_hwaddr' in 'struct ifreq'
                                set_mac((unsigned char *) &ifr.ifr_hwaddr.sa_data);
                                                           ~~~ ^
2 errors generated.
make: *** [iface.o] Error 1

rofl0r, 2017-12-03:
should be fixed with today's fixes, ending with https://github.com/t6x/reaver-wps-fork-t6x/commit/1381ee2d9b544cfdcb25a3bc77b9b0242668fe4a

closed 2017-12-03 by rofl0r

rofl0r, 2017-12-28:
@DJDan i made some changes that could have broken build on mac again, it'd be nice if you could test if it still compiles.

rofl0r, 2018-01-24:
@DJDan ping. i pushed some changes regarding FCS which may have broken mac. would you be so kind as to test latest git if it still works ?