t6x/reaver-wps-fork-t6x #180: Stuck after receiving beacon

iamvinny, 2017-12-04:
It always gets stuck after receiving beacon, I used to crack this router with no problems, but now it won't crack. I'm very close to the router and I know the pin as well. 

```
./reaver -i wlan0mon -b 34:57:60:0F:E4:FC --pin="10416600" -vvv 

Reaver v1.6.3 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Waiting for beacon from 34:57:60:0F:E4:FC
[+] Switching wlan0mon to channel 11
[+] Received beacon from 34:57:60:0F:E4:FC
[+] Vendor: RalinkTe
```

My WiFi adaptor:

```
PHY	Interface	Driver		Chipset

phy3	wlan0mon	ath9k_htc	Atheros Communications, Inc. AR9271 802.11n
```

Any tip to get it working would be greatly appreciated.

soxrok2212, 2017-12-04:
Is WPS locked? Is WPS configured (and not just enabled) in the web GUI? Have you tried manually specifying the channel? Are you using the latest version (updated a few hours ago as of this post)? What wireless card are you using?

rofl0r, 2017-12-04:
please also try different versions: latest git, 1.6.3 release, 1.6.2, ... so we can tell whether it is a regression or not

kcdtv, 2017-12-05:
>  What wireless card are you using?

Himaviny uses some ath9K_htc USB device, probably TP-Link wn722n or alfa awuso36nha)
This chipsets was working like a charm in the past, i believe that not so much now. 
I suspect some kind of driver-kernel digression. 
I have one somewhere: I and bob  can help for debugging.

AlexHilgertBRA, 2017-12-05:
Could you show the wash output?
I mean... when I try reaver in a network with wps DISABLED it happens the exactly same thing.
See ya.

wiire-a, 2017-12-14:
I had the same issue with git and 1.6.3, Ralink based AP.

Except for me it worked everytime for the first run, on 2 different occasions. Then it got stuck.

Unfortunately I didn't record wash output.

Card: AWUS036H (rtl8187)
OS: Ubuntu 16.04

vido89, 2018-01-13:
I also have stuck situation from time to tome and need CTRL+C, dont remember which chip-set was targeted and I use rtl3070 based card

rofl0r, 2018-01-14:
it seems the AP doesnt let you associate. maybe due to distance, or mac filtering ? try if association works with aireplay-ng

amazer-bil, 2018-01-15:
@rofl0r hey bro, i have an ap that doesn't let me to associate with when i use the aireplay-ng 
hope you can help me 
thanks !

rofl0r, 2018-01-15:
yo bro i'm not your bro and you're probably too far away

amazer-bil, 2018-01-15:
@rofl0r what about ' hey man ' !
no i'm not far away, actually i have a good signal like  ' -48 ' ,and still won't associate when i use aireplay-ng !

rofl0r, 2018-01-15:
then maybe it uses a mac address filter - you could try to see who's connected and fake a mac of one of the connected clients...

amazer-bil, 2018-01-15:
@rofl0r can you explain to me more about how to do it !
sorry for wasting your time :) !

rofl0r, 2018-01-16:
- use airodump do see mac addrs of connected clients
- use ifconfig wlanX hw ether XX:XX:XX:XX:XX to fake mac of one of clients
- use google for details

soxrok2212, 2018-01-29:
Have the suggestions been tested or can we close this issue due to no response for 2 weeks?

kcdtv, 2018-02-23:
it seems that there won't be more feedback about that... Let's close it.

closed 2018-02-23 by kcdtv

udaykiran192, 2022-03-02:
Hey, hi. even I am facing same problem please help me out. I use reaver-1.6.6 version. What might be the problem. I found no one answered. I also tried installing older version but it shows error when use make commend to compil it. Please help me. My case is same as vido 89. After fake Mac what should I do, please don't mind, I am a bigger

udaykiran192, 2022-03-02:
> Hey, hi. even I am facing same problem please help me out. I use reaver-1.6.6 version. What might be the problem. I found no one answered. I also tried installing older version but it shows error [...]

maxprog, 2022-04-14:
Hi, I know solution how solve this problem. Please contract with me directly by email