t6x/reaver-wps-fork-t6x #182: Android Build

EvilAirborn, 2017-12-08:
Need A Reaver-Forked Build For Android. is their anyway to make it run in android shell? i tried to compile in arm,but as i run it , it says no file or directory

kcdtv, 2017-12-08:
And you think that we can guess what happens with so few "information" ? 
explain waht you do step by step, describe precisly your system, give us the full stdout with the full error message etc...

EvilAirborn, 2017-12-10:
Actually we need a reaver that Support pixiewps for Android, We need This for a Android GUI Hacking Tools Project Named "hijacker" , I think you already know about this project in GitHub,.  We need a reaver build that Exicute pixiewps and support nexutil hooking, I tried to build reaver with LDFLAGS=-static..it runs on Android terminal but not let nexutil hook "libfakeioctl.so".....

rofl0r, 2017-12-10:
well if "we" need this, then i suppose "we" roll back the sleeves and start working. i'm sure "we" can work out the issues. just as a reminder: static linking is incompatible with LD_PRELOAD hooks.

EvilAirborn, 2017-12-10:
I know that, static flags will won't let librarys to hook with it,
But our hijacker app project have been made various success.... We are just stuck with reaver, if anyone made reaver pixie fork for Android,it will be a great kind and we may complete our project :)

soxrok2212, 2017-12-10:
There is an older port of pixiewps for Android: https://github.com/aanarchyy/pixiewps-android

EvilAirborn, 2017-12-10:
I didn't asked for pixiewps port..... I already made the lastest pixiewps build for Android.... But I need the reaver-wps-fork-t6x ported to Android

soxrok2212, 2017-12-10:
Sorry your previous comment was unclear. Honestly, Android support seems kind of wasteful and would be at the very bottom of the to-do list, considering you'd need OTG cables, a rooted phone, and an external power supply depending on your wireless card. On top of that, it's far easier to just use a Raspberry Pi or similar and then SSH in with a phone or tablet or whatever. Not sure of @roflor's thoughts on this and I'm not sure who "we" are, but there is really no need for this in my opinion.

EvilAirborn, 2017-12-11:
Did you read my second comment :)

EvilAirborn, 2017-12-11:
"Me and One" have an project on GitHub named hijacker. We just need a reaver-fork-t6x  to complete our GUI project, it already Support Capturing handshakes plus craking,mdk3, reaver non pixiewps,aireplay & most importantly GUI airodump , this is a full GUI hacking tools with all features, we just need Android port of reaver to complete the project

kcdtv, 2017-12-11:
> Android support seems kind of wasteful and would be at the very bottom of the to-do list, considering you'd need OTG cables, a rooted phone, and an external power supply depending on your wireless card.

I do not have or use Android and I do not like this OS as it is not compliant at all with GNU/hackers philosophy. So for me i won't even put it on any "to do" list. 
Saying so this repository is open to collaboration and if android users work on that and get it done... Cool!  

>  We need This for a Android GUI Hacking Tools Project Named "hijacker" , I think you already know about this project in GitHub,

 I did not know your project and checked it and I think that it is a shame that you do not give any information about the tools used by your wrapper.... When you use GPL softwares you are supposed to give credits to the authors and some kind of link to the project page. I friendly invite you to take care in respecting the beautiful GPL license that you used for your own project. :smiley_cat:

rofl0r, 2017-12-11:
so no one of the team has an interest in getting an android binary build done for you. looks like you're on your own. however, as android is basically linux with an odd hackjob libc on top, there should be nothing major to prevent you from getting it compiled. that being said, i'm closing this since it already wasted enough of our time. if you have any **specific** issue or question other than "plz halp !!!11", you may open another issue.

closed 2017-12-11 by rofl0r

EvilAirborn, 2017-12-11:
Thanks For An Nice Insult, I Have been Done Building a Reaver Hard Fork With Some Hard Library Forks that works on Android, no need to help, I am not a child to post a issue with ''plz halp !!!" :)