t6x/reaver-wps-fork-t6x #184: WARNING: Receive timeout occurred

vido89, 2018-01-02:
Im using latest reaver but I keep getting timeout error

```
$ sudo ./reaver -i wlx3c3300200748 -b 04:8D:38:XX:XX:XX -p 72187517 -vv

Reaver v1.6.3 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Waiting for beacon from 04:8D:38:XX:XX:XX
[+] Switching wlx3c3300200748 to channel 6
[+] Received beacon from 04:8D:38:XX:XX:XX
[+] Vendor: RealtekS
[+] Trying pin "72187517"
[+] Sending authentication request
[!] WARNING: Receive timeout occurred
[+] Sending authentication request
[!] WARNING: Receive timeout occurred
[+] Sending authentication request
[!] WARNING: Receive timeout occurred
[+] Sending authentication request
[!] WARNING: Receive timeout occurred
[+] Sending authentication request
[!] WARNING: Receive timeout occurred
[+] Sending authentication request
```
What could be wrong ? Im in rage, and I placed XX to cover MAC address ?

soxrok2212, 2018-01-02:
Is your card in monitor mode?

vido89, 2018-01-02:
Hi @soxrok2212 yes it is, I tried with mon0 also, same deal on three different routers 
```
iwconfig 
enp2s0    no wireless extensions.

lo        no wireless extensions.

vmnet1    no wireless extensions.

mon0      IEEE 802.11bgn  Mode:Monitor  Frequency:2.462 GHz  Tx-Power=20 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          
wlx3c3300200748  IEEE 802.11bgn  Mode:Monitor  Frequency:2.462 GHz  Tx-Power=20 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          
vmnet8    no wireless extensions.
```

soxrok2212, 2018-01-02:
What is the card?

vido89, 2018-01-02:
Its Ra Link 3070

rofl0r, 2018-01-02:
its clear from the log what happens:
authentication packets are sent, but no response is received.
possible reasons:
- injection doesnt work
- AP doesn't receive your packets (too far away/too much congestion)
- AP uses mac filtering

soxrok2212, 2018-01-02:
Also note that in some cases (very rare), WPS is enabled but not configured and may cause errors. Might also be useful to see if you can associate with aireplay-ng.

vido89, 2018-01-02:
It seams that does not support injection any more, strange it was working with BT and kali some time a go. I will try backports, do you guys have any sugestion how to roll back drivers to some old version. The drivers are issue.

rofl0r, 2018-01-02:
what's the version of your kernel (run `uname -a`) ?

vido89, 2018-01-02:
```
$ uname -a
Linux G31M-ES2L 4.4.0-104-generic #127-Ubuntu SMP Mon Dec 11 12:16:42 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
```

vido89, 2018-01-02:
I think that this is new feature in rtl3070 drivers for monitor mode, does not support sending packets

rofl0r, 2018-01-02:
that bug was fixed in kernel 4.5, see https://github.com/sabotage-linux/sabotage/blob/master/KEEP/linux-4.3-ralink-monitor.patch

vido89, 2018-01-02:
Con you do anything to apply those changes to my kernel or to just wait for it. tnx

rofl0r, 2018-01-02:
i dont think the LTS distro you use will ever switch to a newer kernel. try booting from a kali live cd, or read up on how to install a newer kernel on ubuntu. good luck

vido89, 2018-01-02:
I see ok tnx for your help,I wish you happy new year and good helth cheers

rofl0r, 2018-01-02:
thanks, same to you.

closed 2018-01-02 by rofl0r

rofl0r, 2018-01-02:
@vido89 could you please also close your double-post https://forum.aircrack-ng.org/index.php/topic,1719.msg5768.html#msg5768 and point at the resolution here? thanks.

vido89, 2018-01-02:
Sure :)

vido89, 2018-01-03:
Something is not right with rtl3070, I tried it on kali but reaver isnt working, here is [reaver.log](https://github.com/t6x/reaver-wps-fork-t6x/files/1601538/reaver.log)
Also I have OpenWrt/LEDE box with ar9285 and it works just fine, its same distance to router, the drivers for rtl 3070 are not correct I assume ?

vido89, 2018-01-03:
And here is log from LEDE box 
[AR9285.log](https://github.com/t6x/reaver-wps-fork-t6x/files/1601592/AR9285.log)

rofl0r, 2018-01-03:
you need -N

vido89, 2018-01-03:
Yup that worked, tnx a lot

ligmaDX800, 2019-03-23:
What about this one, output is same except that different arguments were used.
@rofl0r

rofl0r, 2019-03-23:
please open your own, new issue, after you tried everything pointed out here and in the wiki article https://github.com/t6x/reaver-wps-fork-t6x/wiki/Troubleshooting