t6x/reaver-wps-fork-t6x #185: Segmentation fault and (ESSID: (null)) on LEDE/OpenWrt

vido89, 2018-01-04:
Im using latest stable LEDE Reboot 17.01.4 r3560-79f57e422d and reaver 1.4.2 and 1.6.3.2

With both versions Im getting Segmentation fault 

```
# LD_LIBRARY_PATH=. ./reaver1.4.2  -i wlan0-1 -b 04:8D:38:AD:8B:9E -v -c 6 -m 00:00:00:00:0
0:09

Reaver v1.4 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[?] Restore previous session for 04:8D:38:AD:8B:9E? [n/Y] y
[+] Restored previous session
[+] Waiting for beacon from 04:8D:38:AD:8B:9E
[+] Associated with 04:8D:38:AD:8B:9E (ESSID: Goran)
[+] Trying pin 00025676
[!] WARNING: Failed to associate with 04:8D:38:AD:8B:9E (ESSID: Goran)
Segmentation fault
```
And if I use -m something Im getting (ESSID: (null))

```
# LD_LIBRARY_PATH=. ./reaver1.4.2  -i wlan0-1 -b E4:6F:13:67:B4:AE -v -c 1 -m 00:00:00:00:0
0:09

Reaver v1.4 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[?] Restore previous session for E4:6F:13:67:B4:AE? [n/Y] y
[+] Restored previous session
[+] Waiting for beacon from E4:6F:13:67:B4:AE
[!] WARNING: Failed to associate with E4:6F:13:67:B4:AE (ESSID: (null))
[!] WARNING: Failed to associate with E4:6F:13:67:B4:AE (ESSID: (null))
[!] WARNING: Failed to associate with E4:6F:13:67:B4:AE (ESSID: (null))
[!] WARNING: Failed to associate with E4:6F:13:67:B4:AE (ESSID: (null))
[!] WARNING: Failed to associate with E4:6F:13:67:B4:AE (ESSID: (null))
[!] WARNING: Failed to associate with E4:6F:13:67:B4:AE (ESSID: (null))
^C
[+] Session saved.
```
and

```
# LD_LIBRARY_PATH=. ./reaver -i wlan0-1 -b E4:6F:13:67:B4:AE -v -c 1 -m 00:00:00:00:00:09

Reaver v1.6.3 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[?] Restore previous session for E4:6F:13:67:B4:AE? [n/Y] y
[+] Restored previous session
[+] Waiting for beacon from E4:6F:13:67:B4:AE
[+] Received beacon from E4:6F:13:67:B4:AE
[+] Trying pin "22225672"
[+] Associated with E4:6F:13:67:B4:AE (ESSID: (null))
Segmentation fault
```

rofl0r, 2018-01-04:
thanks for your report.
please try the following:
- clone latest version from this repo
- `cd src`
- `CFLAGS="-O0 -g3" ./configure && make`
- `gdb --args ./reaver -i wlan0-1 -b E4:6F:13:67:B4:AE -v -c 1 -K 1`
then you get a gdb prompt. type `r` to start the prog, and when it crashes type `bt` and paste the backtrace here. please also provide info about the AP. (essid)

rofl0r, 2018-01-04:
i forgot to ask, does the issue also happen if you do **not** restore the session ? i.e. if you answer with "no" rather than yes

vido89, 2018-01-04:
If I answer no
```
# LD_LIBRARY_PATH=. ./reaver -i wlan0 -b E4:6F:13:67:B4:AE -v -c 1 -N -K 1

Reaver v1.6.3 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[?] Restore previous session for E4:6F:13:67:B4:AE? [n/Y] n
[+] Waiting for beacon from E4:6F:13:67:B4:AE
[+] Received beacon from E4:6F:13:67:B4:AE
[+] Vendor: RealtekS
[+] Trying pin "12345670"
[+] Associated with E4:6F:13:67:B4:AE (ESSID: Petar)
[+] Trying pin "12345670"
[+] Associated with E4:6F:13:67:B4:AE (ESSID: Petar)
[+] Trying pin "12345670"
[+] Associated with E4:6F:13:67:B4:AE (ESSID: Petar)
executing pixiewps -e d0141b15656e96b85fcead2e8e76330d2b1ac1576bb026e7a328c0e1baf8cf91664371174c08ee12ec92b0519c54879f21255be5a8770e1fa1880470ef423c90e34d7847a6fcb4924563d1af1db0c481ead9852c519bf1dd429c163951cf69181b132aea2a3684caf35bc54aca1b20c88bb3b7339ff7d56e09139d77f0ac58079097938251dbbe75e86715cc6b7c0ca945fa8dd8d661beb73b414032798dadee32b5dd61bf105f18d89217760b75c5d966a5a490472ceba9e3b4224f3d89fb2b -s da51c89d6bc258a3e0c484a72f43ec947159ef5c263acb29af396630003dbfd3 -z f12fa29ac4581d690a8683d6e9cb631d1e4c600d83b1f85416d19c5ab41adb8c -a fd904445c4572eb39f73a9424cdf514780dc8fef71611462939b6b56c43f1422 -n 65265cc736d81e186f18ae3d0ec352f1 -r 749d3360e1ad61d72950e5b58cd731ffbab43882d05fa00e7d303150a3dd4e70fc394e60291cdcd624361c30d6b541118d86fec389c8f7c9d3a89055d6bf53154b895da540ac926b4e7e280c5a35e3950a8b843f8470ae847fe64aae9dfe7dded41cf2a90231f581f9aacd893db7358efc8f9

f10b76ac5ffe8c2a0c10c139ed338085a5e44910e023e0f741b1a1604022b090b770f5ddfe4effd908e14cf9181964e7992229bca353a68285219a75137d8130956e878956f42f3673daa72ea36
```
and waits there, I waited 1 min but nothing

Im pretty useless when it come to cflags and what not so I need you to help me on this. To compile reaver I used LEDE SDK so I need to put "CFLAGS to make file, but not know where ?
Is it under CONFIGURE_ARGS += ? Here is my [Makefile.txt](https://github.com/t6x/reaver-wps-fork-t6x/files/1605041/Makefile.txt)

rofl0r, 2018-01-05:
when running this pixiewps command manually, i get instantly this result:
```
./pixiewps -e d0141b15656e96b85fcead2e8e76330d2b1a28c0e1baf8cf91664371174c08ee12ec92b0519c54879f21255be5a8770e1fa1880470ef423c90e34d7847a6fcb4924563d1af1db0c481ead9852c519bf1dd429c163951cf69181b132aea2a3684caf35bc54aca1b20c88bb3b7339ff7d56e09139d77f0ac58079097938251dbbe75e86715cc6b7c0ca945fa8dd8d661beb73b414032798dadee32b5dd61bf105f18d89217760b75c5d966a5a490472ceba9e3b4224f3d89fb2b -s da51c89d6bc258a3e0c484a72f43ec947159ef5c263acb29af396630003dbfd3 -z f12fa29ac4581d690a8683d6e9cb631d1e4c600d83b1f85416d19c5ab41adb8c -a fd904445c4572eb39f73a9424cdf514780dc8fef71611462939b6b56c43f1422 -n 65265cc736d81e186f18ae3d0ec352f1 -r 749d3360e1ad61d72950e5b58cd731ffbab43882d05fa00e7d303150a3dd4e70fc394e60291cdcd624361c30d6b541118d86fec389c8f7c9d3a89055d6bf53154b895da540ac926b4e7e280c5a35e3950a8b843f8470ae847fe64aae9dfe7dded41cf2a90231f581f9aacd893db7358efc8f9f10b76ac5ffe8c2a0c10c139ed338085a5e44910e023e0f741b1a1604022b090b770f5ddfe4effd908e14cf9181964e7992229bca353a68285219a75137d8130956e878956f42f3673daa72ea36

 Pixiewps 1.4

 [?] Mode:     3 (RTL819x)
 [*] Seed N1:  1515107539 (Thu Jan  4 23:12:19 2018 UTC)
 [*] Seed ES1: 1515107543 (Thu Jan  4 23:12:23 2018 UTC)
 [*] Seed ES2: 1515107543 (Thu Jan  4 23:12:23 2018 UTC)
 [*] PSK1:     9f7aea0b45ba2ef21286484e5b4ead8d
 [*] PSK2:     0d775d39825260c802e680082429929b
 [*] ES1:      5cf3c3d00b09b6680ae156c7330af36a
 [*] ES2:      5cf3c3d00b09b6680ae156c7330af36a
 [+] WPS pin:  97905820

 [*] Time taken: 0 s 384 ms

```
since it works when you answer "n", i suspect your savefile is broken.

if you still want to try the CFLAGS, maybe try it on your ubuntu desktop. i have no idea how to do it with LEDE.

vido89, 2018-01-06:
Ok right tnx. I did download latest raver and place dir `src, tools` in side build_dir, replacing original but while compiling I get an error
```

make[4]: Entering directory '/home/slobodan/Desktop/LEDE/lede/build_dir/target-mips_24kc_musl/reaver-1.6.3/src'
mips-openwrt-linux-musl-gcc -DCONF_DIR='""'  -Wno-unused-variable -Wno-unused-function -Wno-pointer-sign -Ilibwps -I. -Ilwe -DCONFIG_IPV6   -c -o argsparser.o argsparser.c
cc1: note: someone does not honour COPTS correctly, passed 0 times
argsparser.c:40:20: fatal error: config.h: No such file or directory
compilation terminated.
<builtin>: recipe for target 'argsparser.o' failed
make[4]: *** [argsparser.o] Error 1
make[4]: Leaving directory '/home/slobodan/Desktop/LEDE/lede/build_dir/target-mips_24kc_musl/reaver-1.6.3/src'
Makefile:58: recipe for target '/home/slobodan/Desktop/LEDE/lede/build_dir/target-mips_24kc_musl/reaver-1.6.3/.built' failed
make[3]: *** [/home/slobodan/Desktop/LEDE/lede/build_dir/target-mips_24kc_musl/reaver-1.6.3/.built] Error 2
make[3]: Leaving directory '/home/slobodan/Desktop/LEDE/lede/feeds/packages/net/reaver'
package/Makefile:109: recipe for target 'package/feeds/packages/reaver/compile' failed
make[2]: *** [package/feeds/packages/reaver/compile] Error 2
make[2]: Leaving directory '/home/slobodan/Desktop/LEDE/lede'
package/Makefile:105: recipe for target '/home/slobodan/Desktop/LEDE/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/slobodan/Desktop/LEDE/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/slobodan/Desktop/LEDE/lede'
/home/slobodan/Desktop/LEDE/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
```
Do you have any idea what went wrong

rofl0r, 2018-01-06:
README is your friend

it says something about using ./configure

vido89, 2018-01-06:
No Im using LEDE SDK, I dont use ./configure at all, just selected reaver in menuconfig ? Or I missing something ?

rofl0r, 2018-01-06:
./configure creates config.h, i have no idea how LEDE works

vido89, 2018-01-06:
Ok I'v got somehere with 1.6.3-2
```

# LD_LIBRARY_PATH=. ./gdb --args ./reaver -i wlan0 -b 04:8D:38:A
D:8B:9E -v -c 1 -K 1 -N
GNU gdb (GDB) 8.0.1
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "mips-openwrt-linux".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./reaver...(no debugging symbols found)...done.
(gdb) r
Starting program: /tmp/G/reaver -i wlan0 -b 04:8D:38:AD:8B:9E -v -c 1 -K 1 -N
warning: Unable to find dynamic linker breakpoint function.
GDB will be unable to debug shared library initializers
and track explicitly loaded dynamic code.

Reaver v1.6.3 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

[+] Waiting for beacon from 04:8D:38:AD:8B:9E
[+] Received beacon from 04:8D:38:AD:8B:9E
[+] Trying pin "12345670"
[+] Associated with 04:8D:38:AD:8B:9E (ESSID: (null))
[+] Trying pin "12345670"
[+] Associated with 04:8D:38:AD:8B:9E (ESSID: (null))
[+] Trying pin "12345670"
[+] Associated with 04:8D:38:AD:8B:9E (ESSID: (null))

Program received signal SIGSEGV, Segmentation fault.
0x77fcc7b4 in ?? ()
(gdb)
```

vido89, 2018-01-06:
Can you make temporary release of current reaver, name it 1.6.3-something so I can download it with sdk ?

rofl0r, 2018-01-06:
sorry ? can't you just put a commit id (hash of latest commit) to download a github tarball ? that works exactly the same as a tag

vido89, 2018-01-06:
I'll try that :)

vido89, 2018-01-06:
I need similar link to this one https://github.com/t6x/reaver-wps-fork-t6x/releases/download/v1.6.3/reaver-1.6.3.tar.xz but I cant find a way to make one ? How did you make it avalible in form `releases/download/v1.6.3/reaver-1.6.3.tar.xz`

vido89, 2018-01-07:
Never mind I compiled latest git version, just need to find how to add those CFLAGS

rofl0r, 2018-01-07:
you pass them along when you run configure, as described in this post https://github.com/t6x/reaver-wps-fork-t6x/issues/185#issuecomment-355418142

vido89, 2018-01-07:
Here it is compiled with cflags

```
[+] Associated with 04:8D:38:AD:8B:9E (ESSID: (null))

Program received signal SIGSEGV, Segmentation fault.
0x77f22ee9 in ?? ()
(gdb) bt
#0  0x77f22ee9 in ?? ()
warning: GDB can't find the start of the function at 0x77f22ee9.

    GDB is unable to find the start of the function at 0x77f22ee9
and thus can't determine the size of that function's stack frame.
This means that GDB may be unable to access that stack frame, or
the frames below it.
    This problem is most likely caused by an invalid program counter or
stack pointer.
    However, if you think GDB should simply search farther back
from 0x77f22ee9 for code which looks like the beginning of a
function, you can increase the range of the search using the `set
heuristic-fence-post' command.
```
Dont now why it does not show more info

rofl0r, 2018-01-07:
that's not really helpful... for some reason important info is missing.
can you pastebin your .wpc session file ? (it should either be in current dir or in /var/lib/reaver)

vido89, 2018-01-07:
Sure https://pastebin.com/QTx19F67

rofl0r, 2018-01-07:
i can't see anything obviously wrong... so this probably has to wait until I or someone else has the opportunity to debug it on a proper PC.
reaver's code dealing with session state is generally very bad and requires an overhaul, however these days its hard to test since almost all APs can't be bruteforced anymore.
so debugging/developing improvements here would require some artificial setup.

adde88, 2018-01-09:
@vido89 
How are you compiling Reaver? I presume with a LEDE-SDK, right?
If so, you should be able to use the Makefiles for OpenWRT SDK, as they're quite similar.

Here's one for the latest version of Reaver: [link](https://raw.githubusercontent.com/adde88/reaver-1.6.1-openwrt/master/package/net/reaver/Makefile)

vido89, 2018-01-09:
@adde88 Yes I'v used LEDE-SDK to compile reaver, I have reaver Makefile its same as yours and I have compiled it with debugging options included, but I heaving trouble doing back trace when it crashes

adde88, 2018-01-11:
I'm not sure if it could be related, but the normal OpenWRT-SDK for CC is using **uClibc**.
Master for LEDE-SDK is using **musl**, could this be the cause?

rofl0r, 2018-01-11:
i'm using musl too, so that's not the cause. what's possible though is that his libc.so is built without debug info.

kcdtv, 2018-01-11:
Hi! Just to confirm that the *.wpc file used is correct/legitimate. You can safely discard it as a reason for the error.