t6x/reaver-wps-fork-t6x #187: -B command

zayaerme, 2018-01-10:
When I try to use it with command it says unknown command -B. It is updated. When I try to use --pin command it is same but with -p i can enter "" value but I dont know if i am doing it right. Should it be empty or I should add "" as a value ?

rofl0r, 2018-01-10:
if you want to send empty pin, then yes, use `-p ""`.
i don't know what `-B` is supposed to do, so i can't help you with that.

vido89, 2018-01-10:
@zayaerme you should use lower case "b" like so -b

kcdtv, 2018-01-11:
To send a legitimate 8 or 4 digits PIN just use -p 12345670 (replace 12345670 by your PIN)  
  To understand what is the "arbitrary sring" about please refer to our wiki: [Introducing a new way to crack WPS: Option p with an Arbitrary String](https://github.com/t6x/reaver-wps-fork-t6x/wiki/Introducing-a-new-way-to-crack-WPS:-Option--p-with-an-Arbitrary-String)  
  To answer your question about -B, as vido89 and r0flor said, there is no such option.
  I close this issue because...  there is no issue.  😺

closed 2018-01-11 by kcdtv