t6x/reaver-wps-fork-t6x #192: Wash dual band scan

OscarAkaElvis, 2018-01-22:
Hi, I noticed using "-5" argument on wash scanning make appear ONLY 5ghz access points. It could be nice to have a "dual band scan" like airodump has. I mean, scan to detect 2.4ghz and 5ghz networks all in the same scan.

This could be very useful because usually, on the first scan you don't know what you want (2.4ghz/5ghz) because you don't know yet what access points exist in range. Thank you.

kcdtv, 2018-01-22:
Not a bad idea... :smiley_cat:

adde88, 2018-01-28:
Yeah, that'd actually be really useful 😄

closed 2018-01-28 by rofl0r

rofl0r, 2018-01-29:
does it work ?

OscarAkaElvis, 2018-01-29:
I'm checking your commit. Which branch? master? how to use? just putting -2 and -5 on wash line command? I'm at work now and I can't test for now. I'll check this night.

Another thing I need is to know the version of reaver which will contain this feature... 1.6.4? 1.6.5? I need to know it in order to implement it on my script based on version. Thanks.

rofl0r, 2018-01-29:
> Which branch? master

yes

> just putting -2 and -5 on wash line command?

> the version of reaver which will contain this feature

since 1.6.4 is already released, yes, the next release containing it will be 1.6.5

rofl0r, 2018-01-29:
> in order to implement it on my script based on version

you could also just run `wash --help 2>&1 | grep 2ghz > /dev/null && supported=true`

OscarAkaElvis, 2018-01-29:
I tested it in a vm at work. Cloned master and compiled. I'm trying to testing it but argument -2 is not working. Always I use it, the help of wash is shown in the same way like when you put an illegal argument. The -2 option is shown on help text but it's not working.

Option -5 is working fine. But -2 can't be combined with -5 because as I said, always you use -2 the help is shown even using -2 only (it's understood the simplest command `wash -i wlan0mon -2`).

rofl0r, 2018-01-29:
silly me, i forgot to add the short option for getopt. can you try again (pull first) ?

OscarAkaElvis, 2018-01-29:
It's working now!! awesome. You rock dude! 🥇

commit bd6e3692 (2018-01-28): wash: make it possible to select both 2.4 and 5 ghz band

use -2 and -5 together

closes #192