t6x/reaver-wps-fork-t6x #193: Reaver and Wash give a blank stdout with rtl8814au since version 1.6.4

kcdtv, 2018-01-24:
I tried with the stable releases 1.6.1 - 1.6.2 and 1.6.3 and both reaver and wash work as they should.
With the latest commit they do not anymore.
With full verbose mode enabled reaver stays at "waiting for beacon" vs all targets
```
[+] Switching wlan1 to channel 6
[+] Waiting for beacon from B2:46:FCXXXXX
```
Wash does not show anything
Clearly something went wrong with this chip in the way to treat the beacons. 
I will pass by IRC this evening. Cheers.

rofl0r, 2018-01-24:
the RTL8812AU driver from aircrack-ng is buggy, it produces wrong FCS checksum, but it says in the radiotap header: 1) FCS checksum bit is set 2) BAD_FCS checksum bit is not set.
@kimocoder  ^
as a quick fix, this could be fixed in the driver by always setting the BAD_FCS bit, but it would of course be better to debug where the bad checksum comes from.

closed 2018-01-24 by rofl0r

rofl0r, 2018-01-24:
oh it seems the issue is known, and a fix too https://github.com/aircrack-ng/rtl8812au/issues/28

kimocoder, 2018-01-25:
It has been merged. Thanks for approving the fix.

commit 74254238 (2018-01-24): bring back --ignore-fcs options, this time for both reaver and wash

wash historically used -C for the --ignore-fcs option, but reaver uses
that letter already for other purposes. we now reintroduce the option
(removed in 31cd6da1c4ed55163443ea5da923b7fbb3e94f5d under wrong premises[0])
using the letter -F, so both reaver and wash can use the same.
note that reaver in the past did not even have a way to ignore FCS validation.

this fixes issues with the wrong checksum produced by the RTL8812AU
driver from https://github.com/aircrack-ng/rtl8812au .

note this only needed for drivers that are br [... cut]