t6x/reaver-wps-fork-t6x #194: Debian Alfa 36NH issue

jcuervo947, 2018-01-26:
Hey everyone, I cant seem to find any info searching for my issue. But, when I do a clean Kali or Parrotsec install. I run Wash, and AP's with WPS show up just fine. However, after I do the update to latest, both OS's wont detect WPS anymore at all. I have multiple computers and multiple wifi adapters albiet they are all the same model. I cant seem to find out exactly what updated file is doing this. I thought it was the libpcap08, so I downgraded back to 1.7 but that didnt fix it. I then tried the 1.8dev still nothing. Since reaver/wash/wifite are the only issues I have seen so far im asking in here rather than on parrotsec and kali forum. That being said I looked there and havent seen anyone with the same issue.  Im not running either OS in a VM.  Would anyone be able to point me in the right direction? Thanks in advance.

vido89, 2018-01-26:
What you saying that wash does not display wps enabled AP-s ?
What version of wash/reaver did you use when they worked ?
What version do you run now ?
Can you try to run it on let say ubuntu because it seams to be OS problem ?
Can you run uname -a and post it here ?

kcdtv, 2018-01-27:
I always test reaver from Kali which i use nearly everyday and i did not notice the issue. It is not a fresh install but it is updated recently.. Let's check that:
```
uname -a
Linux kalimuX0 4.14.0-kali3-amd64 #1 SMP Debian 4.14.12-2kali2 (2018-01-11) x86_64 GNU/Linux
```
Before updating everything works fine and that are the updates i have:

```
   bdfproxy dconf-cli dconf-gsettings-backend dconf-service dpkg dpkg-dev
   exploitdb fonts-wine gir1.2-atk-1.0 gir1.2-vte-2.91 gsfonts-x11
   gtk2-engines-pixbuf hddtemp kali-archive-keyring kali-desktop-common
   kali-desktop-live kali-linux kali-linux-full kmod krb5-locales libamd2
   libatk1.0-0 libatk1.0-data libbabl-0.1-0 libcamd2 libccolamd2 libcholmod3
   libcolamd2 libdconf1 libdpkg-perl libegl-mesa0 libegl1-mesa libgadu3
   libgail-common libgail18 libgbm1 libgl1-mesa-dev libgl1-mesa-dri
   libgl1-mesa-glx libglapi-mesa libglx-mesa0 libgssapi-krb5-2 libgtk2.0-0
   libgtk2.0-bin libgtk2.0-common libk5crypto3 libkmod2 libkrb5-3
   libkrb5support0 libopencolorio1v5 libosmesa6 libqt5concurrent5 libqt5core5a
   libqt5dbus5 libqt5designer5 libqt5gui5 libqt5help5 libqt5network5
   libqt5printsupport5 libqt5sql5 libqt5sql5-sqlite libqt5test5 libqt5widgets5
   libqt5xml5 libsuitesparseconfig5 libumfpack5 libvte-2.91-0
   libvte-2.91-common libwayland-egl1-mesa libwine libxatracker2
   linux-compiler-gcc-7-x86 linux-headers-4.14.0-kali3-amd64
   linux-headers-4.14.0-kali3-common linux-image-4.14.0-kali3-amd64
   linux-kbuild-4.14 linux-libc-dev mesa-common-dev mesa-va-drivers
   mesa-vdpau-drivers pixiewps pocl-opencl-icd qt5-gtk-platformtheme tzdata
   whois wine wine64
```
At first sight I do not see any library directly related to reaver 
I do the update and check reaver to tell you what happens.
. 

 
By the way, Do you use the packaged version from kali' repositories or do you use the latest revision directly downloaded from git hub? 
 I have some updating to do. Maybe the bug appears now.
to keep a trace this are the package i am gonna install and they include a kernel update

ericgingras, 2018-01-27:
I am also having problems with wifi drivers in latest kali. I had to go back to 2017.2

jcuervo947, 2018-01-27:
Thanks for getting back so quickly, I will test out Ubuntu when tomorrow.  I have tried it with both Reaver Wash's. The one that comes with the Kali 2017.3 – 21st November, 2017 – The third 2017 Kali Rolling release. Kernel 4.13, GNOME 3.26 and the one from this site. After a fresh install I run apt update &&  apt upgrade. When I run airmon-ng it shows the chipset is RT2870/RT3070 driver is rt2800usb.   An interesting observation i made, when I ran wifite while its scanning every once in a while a AP with WPS would show "yes" and a split second later it will change to "no" like its trying and something is over riding it.  and again all the APs have WPS enabled and I am several feet away from them.  Ill post a uname -a tomorrow.

closed 2018-01-27 by jcuervo947

kcdtv, 2018-01-28:
As the stdout of wash changed a little to display the manufacturer of the wifi chipset it is possible that wifite does not look for the information in the good place and always display "no". 

> Sorry to have wasted your time.

Don't worry...  For me it took 5 minutes tu update (which i would have done anyway) and 5 more to check, I didn't waste any time.     
Cheers