t6x/reaver-wps-fork-t6x #195: Makefile should install the man page

yurivict, 2018-01-28:


rofl0r, 2018-01-29:
which manpage ?

yurivict, 2018-01-29:
docs/reaver.1.gz

soxrok2212, 2018-01-29:
The manpage is severely outdated. Will need to be updated beforehand.

rofl0r, 2018-01-29:
i don't intend to maintain it. do you ? if not, let's just remove it.

soxrok2212, 2018-01-29:
It is not my priority right now. Perhaps @kcdtv wants to take on the task.

yurivict, 2018-01-29:
I can't maintain random manpages.
But it's certainly nice to have them.

rofl0r, 2018-01-29:
oh, you can speak ? :)

i agree it's nice, but in case of reaver its just a copy of output of ./reaver --help. any place where it is duplicated represents a maintainance burden, i'm already unhappy about having the output duplicated in the README.

rofl0r, 2018-01-29:
thinking of it, we may make it an empty manpage which says "run reaver --help to see possible options". that would be acceptable. but it is really helpful ?

yurivict, 2018-01-29:
No, it should be either meaningfully maintained, or removed.

yurivict, 2018-01-29:
As a comparison, jq https://github.com/stedolan/jq prints the brief usage explanation with ```--help```, and has a detailed documentation in its man page.
Or you can just maintain a man page and print the same text when it is called with ```--help```.

kcdtv, 2018-01-29:
Hi!
I personally like the fact to have the options in shell when I execute an incorrect command line; I would not change that. 
I don't have neither a lot of time to dedicate to that but I think that it is a good idea to have an up to date man page. 
So I propose myself to do it (slowly. there is no hurry for that, isn't it?)

kcdtv, 2018-01-29:
@ sox: And you are "assigned" for(to?) spelling and  grammar check...  :smiley_cat:

kcdtv, 2018-01-29:
I've checked a little and that is my proposal: 
I will update   [README.REAVER ]( reaver-wps-fork-t6x/docs/README.REAVER ) and [READEME.WASH]( reaver-wps-fork-t6x/docs/README.WASH ) which are perfects to be used as a man page for each tool. 
So we keep things as they are until the updating is done and than we introduce `man reaver` and `man wash` displaying the specific readme from the tool called. 
I do not see any advantage in creating a short and a long version of the atcual option stdout in shell
Doing `reaver --help` or `-h` works "by default" as it will display option description (option does not exist so it is like a mlaformed command line) 
Let's keep it simple and do not overthink it.  We just add a good man page as we agree that is a good idea.

rofl0r, 2018-01-29:
ok, so can it be removed for now ? btw: your links above are both 404

soxrok2212, 2018-01-29:
I don't think they are meant to be links. I'll check over grammar 👍

rofl0r, 2018-01-29:
oh, i didn't get this point of @kctdv earlier:

> So we keep things as they are until the updating is done and than we introduce man reaver and man wash displaying the specific readme from the tool called.

you can't just use a textfile or .md file as manpage, manpages need to be written in a special format called nroff. some projects generate the manpage at `make` time using a tool from text input, but i think that is overkill - i actually hate that behaviour since most of the time those tools have a shit-load of dependencies, and not having it installed will make your build fail. example is tinyproxy which requires `a2x` tool from `asciidoc`.

rofl0r, 2018-01-29:
i unpacked the manpage so one can actually take a look at its anatomy and content. https://github.com/t6x/reaver-wps-fork-t6x/blob/8ba3acbff2593616e5a05155bee7553a30c10c1f/docs/reaver.1

as one can see, this format is quite odd and maintaining it by hand requires extensive study of the nroff format.

kcdtv, 2018-01-29:
Thanks for your explanations. That is more work than what I thought with this very weird _mise-en- page_