t6x/reaver-wps-fork-t6x #198: pcap.h not found

Ghost0s, 2018-01-29:
after executing the make command i got this : 
cp lwe/wireless.21.h lwe/wireless.h
cc -DCONF_DIR='""'  -Wno-unused-variable -Wno-unused-function -Wno-pointer-sign -Ilibwps -I. -Ilwe -DCONFIG_IPV6   -c -o argsparser.o argsparser.c
In file included from globule.h:37:0,
                 from argsparser.c:37:
defs.h:43:10: fatal error: pcap.h: No such file or directory
 #include <pcap.h>
          ^~~~~~~~
compilation terminated.
<builtin>: recipe for target 'argsparser.o' failed
make: *** [argsparser.o] Error 1

rofl0r, 2018-01-30:
he probably also didn't run ./configure - which checks for pcap.h

Ghost0s, 2018-01-31:
@rofl0r Yes, i did !

Ghost0s, 2018-01-31:
@soxrok2212 Thank you m8 !